balena-io/etcher #876: Website suggests downloading wrong version of Etcher

Charelene, 2016-11-17:
- **Etcher version: v1.0.0-beta.16**
- **Operating system and architecture: Ubuntu 16.10 x86**
- **Do you see any meaningful error information on DevTools? nah, not really**
So the thing is, that clicking "Download for Linux" button downloads x64 AppImage instead of x86. It may confuse some people, I guess?

lurch, 2016-11-17:
This is related to http://stackoverflow.com/questions/1741933/detect-64-bit-or-32-bit-windows-from-user-agent-or-javascript - there's no reliable way to detect whether the host OS is 32-bit or 64-bit (javascript reports the browser arch rather than the host arch). So I guess the Etcher website is just defaultting to 64-bit as that's the most common option.

ping @jviotti and @craig-mulligan

WasabiFan, 2016-11-17:
It depends on the browser. Some browsers do include the architecture in their user agent string, but it's up to the browser implementation and user settings. I'd expect that browsers often report their own architecture instead of the OS' if they report it at all.

craig-mulligan, 2016-11-18:
Yea, it just defaults to 64. I guess providing a table of downloads would help here, though we still have to offer a 'default', perhaps the best option is to add a small description below the download button describing what it is defaulting to.

jviotti, 2017-01-25:
I'm closing this since there is no way to reliably detect the architecture. As a solution, we have a table containing all downloads for all operating systems and architectures in the website.

closed 2017-01-25 by jviotti