balena-io/etcher #877: Windows installer directory "error"

lurch, 2016-11-18:
- **Etcher version:** 1.0.0-beta16
- **Operating system and architecture:** Windows7 x64
- **Do you see any meaningful error information on DevTools?** n/a

I installed the x64 version of Etcher onto my 64-bit Windows 7 laptop tonight, and was very surprised to see the Etcher files ending up in `C:\Program Files (x86)\Etcher\`! According to [this](http://answers.microsoft.com/en-us/windows/forum/windows_7-files/programs-files-vs-program-files-x86/7d631676-e688-472a-a1bb-eefccfeca6d7), `Program Files (x86)` is only supposed to be used for 32-bit programs. Using [this](http://superuser.com/questions/358434/how-to-check-if-a-binary-is-32-or-64-bit-on-windows#889267) I verified that `C:\Program Files (x86)\Etcher\Etcher.exe` is indeed a 64-bit application.

So then as a test I uninstalled the x64 version of Etcher, and installed the x86 version. It again installed into `C:\Program Files (x86)\Etcher\`, but this time using the info from the earlier superuser.com link I was able to verify that `C:\Program Files (x86)\Etcher\Etcher.exe` was now a 32-bit application.

I.e. it's not a packaging problem, as `Etcher-1.0.0-beta.16-win32-x64.exe` contains the x64 application and `Etcher-1.0.0-beta.16-win32-x86.exe` contains the x86 application - it's merely an installation-directory problem.

jviotti, 2016-11-18:
Hm, there must be something going on with the current installer. At the moment, we just use a third-party module to take care of the installation directory, which means we don't have any insights/control about that its actually doing. 

I suggest to solve this once we transition to making the installer in our own build scripts instead of using third party stuff.

lurch, 2016-11-18:
> once we transition to making the installer in our own build scripts instead of using third party stuff.

Hmmm, is there an issue for that?

jviotti, 2016-11-19:
Not really, but is a requirement for the issue about optimizing the final package size, so I guess it's covered by that one.

closed 2017-06-13 by jviotti