balena-io/etcher #881: Button text-case question

lurch, 2016-11-18:
- **Etcher version:** 1.0.0-beta16

pinging @konmouz and @taahirisaacs 

I've just noticed that the "main buttons" in the Etcher UI are displayed in mixed-case i.e. 'Select image', 'Select drive' and 'Flash!', but the buttons on the dialogs are displayed in upper-case i.e. 'CONTINUE' on the drive-selection dialog and 'ENABLE UNSAFE MODE' on the enable-unsafe dialog. And when you've finished flashing, the main UI displays the buttons 'Use **same** image' and 'Use **different** image'.

So I guess it's self-consistent in that UI buttons are always mixed-case and dialog-buttons are always upper-case, but should the dialog-buttons be mixed-case too?

Feel free to close if everyone's happy with the current situation.

closed 2017-01-25 by jviotti

konmouz, 2017-01-25:
If we use 'Title Case' everywhere should be fine, good point @lurch.

jviotti, 2017-01-25:
@konmouz Everywhere including modal titles, and step captions (e.g: "SELECT IMAGE", etc)?

konmouz, 2017-01-25:
Yes @jviotti. It is not wrong what we have now, just to make it look a bit more consistent.