balena-io/etcher #884: Make meaning clear of "connect a drive" text when no drives are fount

WasabiFan, 2016-11-20:
- **Etcher version:** Beta 15, 16
- **Operating system and architecture:** Windows 10 64-bit
- **Do you see any meaningful error information on DevTools?** N/A

When you don't have a drive connected, Etcher looks like this:

[image]

Once you connect one, that changes to this:

[image]

Not only is the first one unclear -- I can't tell what I am actually supposed to do there -- but it barely changes once you have things connected. I think that the "no drives connected" state should be changed to make the lack of drives clear and to make the rectangle look less like a button.

WasabiFan, 2016-11-20:
Interesting... it looks like this was a regression since beta 7, when the rectangle was red while no drives were available. I don't think that the old behavior was optimal, but it was better than it is now.

jviotti, 2016-11-20:
Yeah, we should definitely get this fixed for once and for all. @taahirisaacs , @konmouz mind giving us a way forward here?

lurch, 2016-11-28:
It's also been niggling me slightly that the `Change` buttons (for both Image and Drive) look nothing like buttons; but I dunno if that's something that needs "fixing" or not...

konmouz, 2016-11-30:
Please check [here](https://github.com/resin-io/etcher/issues/888) regarding 'connect a drive' issue.

konmouz, 2017-02-27:
@Shou @jviotti, we still have the issue with 'connect a drive' on the latest version. This looks like a CTA while it is just a message. Ideally, this message appears within the drives' modal window (screenprint from Zeplin). If this is tricky, 'connect a drive' should have an orange background instead of blue.

[image]

jviotti, 2017-02-27:
Yeah, let's make the current dialog open by default if there is no drive and keep improving from there. /cc @Shou

Shou, 2017-02-28:
So to clarify we open the drive selector modal on startup and display the message there if there is no drive?

lurch, 2017-02-28:
No, I think automatically opening the drive-selector modal on startup would confuse / scare people.

jviotti, 2017-02-28:
OK, I'm really confused now :P As far as I understand, @konmouz proposal is to show the drive selector automatically once you select the image, even if there are no drives (e.g: we show an empty drive selector), thus completely covering the "Connect a drive" button, making it unnecessary.

konmouz, 2017-02-28:
Yes, my proposal is to automatically open the drives' modal except if there is only one drive, so it will be automatically selected. If you think this is too aggressive, instead of 'connect drive' we could have the actual button 'select drive' and user can open the modal manual. Again, if there is only one drive, it will be automatically selected.

Shou, 2017-02-28:
Maybe we can do both for convenience here—orange to show that there's no drive but also allow the user to open the dialog or something to that effect? I think not having an indicator right there on the main page is a tiny step backwards. My idea is that it should get the least in the way of the user, because sometimes they may want to see whether there's a drive before clicking it, but also allow users to open the drive selector dialog in preparation before they connect their drives.

Either way I'll push some changes to a branch any moment now.

Shou, 2017-02-28:
Here we go.

lurch, 2017-02-28:
Yeah, perhaps it's just me, but I think automatically opening any modal (except for the update-available notification) as soon as the application starts up, without any other user-interaction first, feels a bit aggressive. IMVHO. Especially as the current "flow" is to select an image first, not select a drive first.

.....

Ah! I now see that I missed the "once you select the image" part of @jviotti 's last comment ;-)  But I guess if we do open the drive-selection modal automatically as soon as an image has been selected, maybe it makes Etcher feel a bit like an "installation wizard" type program rather than a typical desktop application? \*shrug\*

closed 2017-03-01 by jviotti