balena-io/etcher #885: Detects sd card as system drive on 2-in-1 Convertable

BleedingEdgeTech, 2016-11-20:
- 1.0.0-beta. 16
- Windows 10 64-bit

I was chatting with @WasabiFan on resin-io/etcher
he sent me this skript

jviotti, 2016-11-20:
Can you try running the following VBScript code, and see what it prints?

```vbs
strComputer = "."

Set objWMIService = GetObject("winmgmts:\\" & strComputer & "\root\cimv2")
Set colDevices = objWMIService.ExecQuery _
    ("Select * From Win32_USBControllerDevice")

For Each objDevice in colDevices
    strDeviceName = objDevice.Dependent
    strQuotes = Chr(34)
    strDeviceName = Replace(strDeviceName, strQuotes, "")
    arrDeviceNames = Split(strDeviceName, "=")
    strDeviceName = arrDeviceNames(1)
    Set colUSBDevices = objWMIService.ExecQuery _
        ("Select * From Win32_PnPEntity Where DeviceID = '" & strDeviceName & "'")
    For Each objUSBDevice in colUSBDevices
        Wscript.Echo objUSBDevice.Description
    Next    
Next
```

You can save to a `.vbs` file and running with `cscript.exe example.vbs`.

BleedingEdgeTech, 2016-11-20:
Output:
USB-Root-Hub (xHCI)
USB-Verbundgerät
USB-Videogerät
Intel(R) Wireless Bluetooth(R)
Generischer USB-Hub
USB-Eingabegerät
Synaptics HID TouchPad
Synaptics HID Device
USB-Eingabegerät
HID-konformes, vom Hersteller definiertes Gerät
HID-Tastatur
HID-konformes Benutzersteuergerät
HID-kompatible Drahtlos-Empfängersteuerelemente

jviotti, 2016-11-23:
So the output is localized, which doesn't help much :/

Can you try the following script?

```vbs
On Error Resume Next 
arrComputers = Array(".")
For Each strComputer In arrComputers
    WScript.Echo
    WScript.Echo "===================================="
    WScript.Echo "Computer: "& strComputer
    WScript.Echo "===================================="

    Set objWMIService = GetObject("winmgmts:\\" _
        & strComputer& "\root\CIMV2") 
    Set colItems = objWMIService.ExecQuery( _
        "SELECT * FROM Win32_PnPEntity",,48) 
    For Each objItem in colItems 
        Wscript.Echo "-----------------------------------"
        Wscript.Echo "Win32_PnPEntity instance"
        Wscript.Echo "-----------------------------------"
        Wscript.Echo "Name: "& objItem.Name
        Wscript.Echo "Status: "& objItem.Status
        Wscript.Echo "Caption: "& objItem.Caption
        Wscript.Echo "ID: "& objItem.DeviceID
        Wscript.Echo "PNP Class: "& objItem.PNPClass
        Wscript.Echo "PNP ID: "& objItem.PNPDeviceID
    Next
Next
```

lurch, 2016-12-05:
Following on from #828 , should this be closed too?

jviotti, 2016-12-05:
@lurch I propose to keep this opened for a while. This issue is about internal SD-Card readers, and I'm pretty sure we have pretty good chances of "fixing" this edge case.

lurch, 2016-12-05:
I'm beginning to suspect that it might depend on how the SD-Card reader is internally connected (USB / PCIe / something-else), and/or what parameters the SD card reader reports to the OS, and/or which particular driver Windows uses for it?
But yeah, it's worth digging into, if we can :+1:

alexandrosm, 2017-03-02:
Hi, @BleedingEdgeTech did you ever manage to run the script? We're kind of stuck without your help.

lurch, 2018-02-26:
@jhermsmeier I'm closing this, given the lack of feedback from the OP and (more importantly) the significant changes to `drivelist` since this issue was opened.

closed 2018-02-26 by lurch