balena-io/etcher #888: Make behavior of protected system drives more obvious to user

WasabiFan, 2016-11-22:
This came out of a discussion on Gitter, but I wanted to materialize it here for tracking.

> @alexandrosm: @WasabiFan in general I wonder if instead of not showing system drives at all, instead Etcher should just not auto-select them, de-emphasise them on the selection menu, and if the user actively selects it, it gives a warning, but ultimately goes ahead.
@alexandrosm: in other words treat the system drive heuristic as a heuristic instead of as a law as it does now

> @WasabiFan: Yeah, I like that. Or maybe gray them out and then put the "unsafe mode" checkbox on the selection screen (something like "Allow selection of system drives").
This would make it much easier for users to understand why their drives aren't showing up.

In general, the idea would be to make the filtering out of "system" drives visible to the user, so that they can see what is being filtered out and why.

alexandrosm, 2016-11-22:
One more thing to highlight, I like the idea of moving things out of the
"garbage dump" settings page and into the specific screens they are
relevant on. If unsafe mode makes more sense in the drive selection, let's
move it in there.

WasabiFan, 2016-11-22:
Agreed, especially if in doing so that checkbox gets a better name! It would be great to see "settings" become something more like "preferences" (logging, telemetry, debugging, etc.).

jviotti, 2016-11-23:
Sounds cool. Lets put a "system drive" badge as we do for locked, or small devices, and disable them as well. Once that's done, we can move the checkbox there.

lurch, 2016-11-28:
Yeah, overlaying badges on different types of drives is also something that we discussed with @taahirisaacs and @konmouz at the [Summit](https://resin.io/blog/summit-2016/). I think we decided that removable drives (that we don't detect as "system") should always be displayed at the top of the list.

IMHO if we're going to be displaying "System" drives to the user (with 'just' a simple tickbox to enable writing to them) then perhaps we also ought to attempt to identify "boot" drives, and **never** allow those to be overwritten? And we could/should also filter "source" drives, as mentioned in #830 

So, there'd then be:
 * Removable drives - always okay to write
 * "System" drives - writable if the user enables "unsafe mode" (or whatever it gets renamed to)
 * "Boot" drives (e.g. where the current Operating System was loaded from - should we also try detecting swap / pagefile drives currently being used?) - we should prevent the user writing to here

And then any of the above drives can also be:
 * "Source" drives where the disk image is currently being read from - we should prevent the user writing to here
 * "Locked" drives (e.g. SD cards with the read-only switch enabled) - non-writable at the hardware level 
 * "Small" drives where the capacity is less than the currently-selected image - should we allow these to be written to in "unsafe mode"?

Any other categories? :)

WasabiFan, 2016-11-28:
I would recommend avoiding attempting to detect "boot drives". I suspect that Etcher wouldn't be able to do it reliably, and if it isn't reliable I don't think that it should be used at all. For example, on one of my PCs, I had my user folder stored on one drive while the system was on another; applications were stored on the same drive as my user folder. How would this be detected? I'd bet that edge cases like that would render this sort of "protection" unreliable at best and outright dangerous at worst. I'd favor just detecting generic "system" drives and putting protections in place there.

alexandrosm, 2016-11-28:
we discussed this today with @konmouz in person, and he has some sleek
designs, i suspect he might post them here for further comment

konmouz, 2016-11-29:
Yes, I will upload soon, I just want to recap with @taahirisaacs to sync the designs.

The idea is that there will be a pill label for system drives (drive selection modal). All drives, including system ones, are always selectable without enabling 'unsafe mode' (this option will be removed). However, if the user selects a system drive, there will be some sort of alert before continuing (also color change of button). In addition, after clicking 'Flash', user will get another alert modal to re-confirm that flashing a system drive is intended. So practically, it is quite impossible to flash a system drive accidentally.

As an advance setting, we could let users deactivate the final alert to speed up the process. However, I don't think that this will be a frequent use case so probably doesn't really worth it.

Finally, the boot drives concept sounds really good. If this can be implemented reliably it would be great.

jviotti, 2016-11-29:
> IMHO if we're going to be displaying "System" drives to the user (with 'just' a simple tickbox to enable writing to them) then perhaps we also ought to attempt to identify "boot" drives, and never allow those to be overwritten? And we could/should also filter "source" drives, as mentioned in #830

It sounds difficult to get right, but we could get almost there by detecting drives containing partitions mounted at `/`, `/boot`, `/home`, etc in the case of UNIX, and drives containing drive letters that match `%SYSTEMDRIVE%` and `%HOMEPATH%` in Windows.

lurch, 2016-11-29:
Yeah, that's what I was thinking. And we could also scan `/proc/cmdline` on Linux for `root=`. And possibly also check for `%PROGRAMFILES%` on Windows?

jviotti, 2016-11-29:
Sounds good. @lurch Do you mind opening a new issue in `drivelist` to track this?

lurch, 2016-11-29:
Done https://github.com/resin-io-modules/drivelist/issues/126

konmouz, 2016-11-30:
@taahirisaacs and I had a meeting earlier today, so we end up with the following Ui flow:

State after selecting an image:
- If only one removable drive is connected, it will be automatically selected
- Else, if no removable drive or more than one is connected, 'Browse Drives' CTA will be highlighted

[image]

Select drive modal:
- All drives are selectable, however, system drives have a clear label (hopefully boot drives as well)

[image]

Select drive modal:
- If a removable drive is selected, 'Continue' CTA will be highlighted in the usual blue color
- If a system drive is selected, 'Continue' CTA will be highlighted in a red color along with an 'alert' icon and a short confirmation message
- Regarding [double click selection](https://github.com/resin-io/etcher/issues/880#event-876991647), we can implement it only on removable drives

[image]
[image]

- If a system drive is selected, a small 'alert' icon is placed next to the drive name
- If the drive name or the icon is clicked, a small modal appears containing all the info relevant to the drive

[image]
[image]

Flash:
- If user clicks 'Flash' while a system drive is selected, a new confirmation modal appears displaying the selected drive and double-checking if flashing a system drive is an intended action
- User can confirm flashing or directly change the selected drive

[image]

lurch, 2016-11-30:
Looks very nice :grinning:

lurch, 2016-11-30:
...and WRT #884 , it's not explicitly spelled out above, but if we're always displaying system drives, then I guess the drive-list will **never** be empty (since there'll always be at least one system drive), and so the "connect a drive" button will never appear.

jviotti, 2016-11-30:
Looks very cool! What about sorting? Should we sort the available drives so that the system one always get below?

konmouz, 2016-11-30:
@lurch Yes, exactly. System drives are always displayed. Currently, the only case where Etcher selects a drive automatically is when only one removable drive is connected. 'Unsafe mode' has been [completely removed](https://github.com/resin-io/etcher/issues/919) (and therefore advanced settings as well).

@jviotti That would make sense since selecting a removable drive should be more often

WasabiFan, 2016-11-30:
> the "connect a drive" button will never appear.

This feels wrong. Isn't the idea that only removable drives are auto-selected?

alexandrosm, 2016-11-30:
I certainly thing system drives should go to the bottom, and potentially
should be grayed out and need a special checkbox/modal to be selected.

lurch, 2016-12-01:
@konmouz 's design already requires the user to explicitly confirm twice before system drives will be written to. Not sure an extra tickbox to 'enable' system drives to be selectable is really necessary? \*shrug\*

konmouz, 2016-12-01:
@WasabiFan, indeed only removable drives are auto-selected. If only a system drive is available, user will get the 'browse drivers' CTA and has to select it manually.

@alexandrosm, I agree with @lurch. We don't need an extra checkbox to confirm. Currently are notifying user in 4 ways. Red pill label, confirmation with alert icon, alert icon on main Ui page (next to drive) and final confirmation before flashing.

alexandrosm, 2016-12-01:
fair enough.

How about having system/boot drives at the bottom of the list and somehow
de-emphasised? (grayed out?)

alexandrosm, 2016-12-01:
The other thing to be careful about is the relevant text. Due to the fact
that we're still not 100% on this, we shouldn't say "This is a system
drive, are you sure you want to write to it" but something closer to
"Etcher has identified this as a system drive, are you sure you want to
write to it?". It's a small difference but important to communicate the
level of certainty.

konmouz, 2016-12-01:
Good point, I think we can rephrase. Regarding drives' sorting, I agree that removable disks should be on top and system/boot drives at the bottom. We can try de-emphasising them, the only concern is that they might look appear as unavailable.

lurch, 2016-12-01:
As discussed [earlier](https://github.com/resin-io/etcher/issues/888#issuecomment-263355732), I guess there'll effectively be three "levels" of drives:
* Drives always writable (Removable drives)
* Drives only writable to after user-confirmation (System drives)
* Drives never writable (Boot drives, Too-small drives, Locked/Read-only drives, Source drives)

IMHO they should be displayed in that order, with the bottom category never selectable, and with drives within each category sorted from smallest-to-largest in terms of capacity.

And I also agree with @alexandrosm 's distinction :+1:

Shou, 2016-12-01:
What about the behaviour of plugging a removable drive in while Etcher is running? Should that then become the selected drive, if there are only system drives available otherwise?

lurch, 2016-12-01:
Assuming the user hasn't already specifically selected a drive, then yes. (IIRC it's what the current behaviour does)

alexandrosm, 2016-12-01:
I like how the hive mind is working here :)

konmouz, 2016-12-02:
Example of list of drives, sorted as discussed (+ overflow).
[image]

Changed to @alexandrosm's suggestion. I am wondering -just to reduce words- if it makes sense to completely remove the question and keep only "Etcher has identified this as a system drive.", since we do make clear that this will erase your data in the last stage modal.
[image]

lurch, 2016-12-02:
> We can try de-emphasising them, the only concern is that they might look appear as unavailable.

Yeah, "SYSTEM DRIVE" and "BOOT DRIVE" now look equally un-selectable, even though one is and one isn't :-/

> if it makes sense to completely remove the question and keep only "Etcher has identified this as a system drive."

Yeah, that's a good point - clicking "Continue" here doesn't actually write to the drive, so it probably makes sense to only mention "are you sure you want to write to it?" when we're *actually* about to write to the drive. Or we could even make the text say "Warning: Etcher has identified this as a system drive." (would that still fit on a single line?)

konmouz, 2016-12-02:
We also provide the 'warning icon' so I think it is enough.

konmouz, 2016-12-02:
Actually it fits, just test it. So it is still an option.

WasabiFan, 2016-12-02:
How does the warning look if there are enough drives to force scrolling (i.e. They've got the bottom of the list control)?

jviotti, 2016-12-02:
> I am wondering -just to reduce words- if it makes sense to completely remove the question and keep only "Etcher has identified this as a system drive.", since we do make clear that this will erase your data in the last stage modal.

I'm all for reducing the amount of modals we use!

konmouz, 2016-12-02:
[image]

WasabiFan, 2016-12-02:
Looks good! I think the disabled "boot drive" should be greyed out though.

konmouz, 2016-12-02:
So you can not select your Boot Drive in any case, is that right? In this case we should also remove the greyed out 'tick' on the right.

WasabiFan, 2016-12-02:
Yep, I believe that's right.

alexandrosm, 2016-12-02:
@konmouz -- the interaction as designed implies there will only be one
drive selected, which will change with multi-write. Any ideas for how to
alter this to not insert the single-drive assumption?

lurch, 2016-12-03:
@konmouz Nice idea on removing the 'tick' for boot / locked drives :+1: 

For multi-writes I guess we could change the text to something like "Warning: Etcher has identified some of the drives you've selected as system drives." ?

alexandrosm, 2016-12-03:
yeah, something of that type.

konmouz, 2016-12-05:
@alexandrosm, regarding multi-writes, we might need to alter the modal or the whole UI. The drives space is two small for this purpose. Image selecting 10 drive, it is very difficult to have an overview of your selections.

alexandrosm, 2016-12-05:
We should seriously consider what that new modal would look like and try to
see if we can make our modifications go in that direction, as I'd hate to
have to re-do a bunch of stuff when we get there.

konmouz, 2016-12-05:
Sure, I'll prepare some sketches.

konmouz, 2016-12-05:
[image]

lurch, 2016-12-05:
While Etcher only supports writing to one drive I wonder if it makes sense to keep the current scrollable-list UI, and when Etcher switches to supporting multiple drives perhaps we could switch to using a 2D grid-layout (instead of a 1D list) so that all the drives are visible at once, without having to scroll?

alexandrosm, 2016-12-05:
on a somewhat-related note, doesn't it feel like the green tick should be
on the left of the drive name etc?

alexandrosm, 2016-12-05:
and maybe "continue" should be "select these 2 drives"?

lurch, 2016-12-06:
Something else I've been meaning to suggest / ask for ages, but keep forgetting to:
Does it "make sense" that while the drive-selection modal is open, changes made in the modal are **instantly** reflected in the parent Etcher window visible behind the modal (e.g. the "Flash" button changing state)? Would it be better if the parent Etcher window only updated once the "Continue" button had been clicked on? (this is similar to https://github.com/resin-io/etcher/issues/729#issuecomment-256057600 )

konmouz, 2016-12-06:
Agree with both @alexandrosm's and @lurch's points.

konmouz, 2016-12-06:
Update:
- Tick on the LHS
- 'Select drives (amount of drives)' CTA and 'Available (or detected) drives' as modal window name
- I tried to de-emphasize the system drives (I think they still look selectable, especially compared to the boot drive that has no tick available and also is greyed out)
- If drive's directory is too long, we can use a tooltip

[image]

WasabiFan, 2016-12-06:
My initial feeling on this one is that the nuanced grey shades won't be clearly distinct to all users (not just normal ones, but visually impaired ones too and people with low-contrast/brightness displays). It took me a bit to figure out what was going on (although to be fair I'm so tired I lost my phone in my own hand).

konmouz, 2016-12-06:
I see your point. In addition, since we have a bold visual warning when selecting a system drive, I am not sure if it really worth trying to de-emphasize them.

jviotti, 2016-12-06:
@konmouz 

> Just to be clear, is the multiple drive selection a feature for the current version or for v.1?
(For v.1 I am exploring a modal window free approach)

This will come after v1. Is a "nice to have" for v1, but I doubt we'll ship with it, since there are still some technical details to sort out.

lurch, 2016-12-06:
> Just to be clear, is the multiple drive selection a feature for the current version or for v.1?

Current version of Etcher only supports selecting a single drive at a time. When we _do_ have multiple drive selection, for the case where the user has only ticked one drive, would the button say "Select drive" or "Select drives (1)" ?

Hmm, that leads on to another interesting question: When Etcher supports writing to multiple drives at once, will that feature *always* be available (it might be confusing for newbies to accidentally select multiple drives, instead of automatically unselecting the previously-selected drive?), or would it only be available after enabling it in the settings screen?

Agreed the the labelling on the boot drive in the last image is possibly too-light of a grey.

@konmouz Only a small feature, but could you re-order the drives in that last image so that the Removable Drives are ordered smallest (8.0GB) to biggest (120.8GB) ?

konmouz, 2016-12-06:
`Current version of Etcher only supports selecting a single drive at a time. When we do have multiple drive selection, for the case where the user has only ticked one drive, would the button say "Select drive" or "Select drives (1)" ?`

"Select drives (1)" would be better, since it always implies multi-selection

`Hmm, that leads on to another interesting question: When Etcher supports writing to multiple drives at once, will that feature always be available (it might be confusing for newbies to accidentally select multiple drives, instead of automatically unselecting the previously-selected drive?), or would it only be available after enabling it in the settings screen?`

We can use the same functionality with Gdrive's list view.

`Only a small feature, but could you re-order the drives in that last image so that the Removable Drives are ordered smallest (8.0GB) to biggest (120.8GB) ?`

No problem

lurch, 2016-12-06:
> > When Etcher supports writing to multiple drives at once, will that feature always be available, or would it only be available after enabling it in the settings screen?

> We can use the same functionality with Gdrive's list view.

Not sure what you mean there (not being a Gdrive user) - could you post some screenshots? :)

konmouz, 2016-12-06:
Check this video:
https://www.youtube.com/watch?v=ZxAZfs4dqLs

jviotti, 2016-12-06:
> Only a small feature, but could you re-order the drives in that last image so that the Removable Drives are ordered smallest (8.0GB) to biggest (120.8GB) ?

What is the advantage of re-ordering by size?

lurch, 2016-12-06:
Ah-ha! The classic Shift-select / Ctrl-select technique (which nearly all file-managers use, not just limited to GDrive).

I'm not sure how well that would work when selecting multiple items out of a scrollable list though? Might be too easy to click-select (and therefore unselecting previously-selected drives that are scrolled out of view) instead of shift-selecting? I'd been expecting that when Etcher supports multi-drive-selection, it'd work like a traditional tickbox-list, i.e. click a drive to select it, click it again to unselect it.

Whereas at the moment (with just single-drive selection) it works like a traditional radiobox-list.

I may be barking up the wrong tree, but I think newbies would expect Etcher to work in a radiobox-list fashion (as it does now), and it's only advanced users who'll want to use multi-drive-selection, and they'll probably want it to work in a tickbox-list fashion (rather than shift-selecting), which is why I was suggesting there might need to be a settings-button to change the modes?

Sorry for rambling...

lurch, 2016-12-06:
> What is the advantage of re-ordering by size?

a) Always nice to have a persistent sort-mode, rather than having the drives appearing in a random order each time you start the application
b) I was thinking that a user is more likely to want to flash an image onto their 8GB SD card, than it is they'll want to flash an image to their 500GB removable harddrive? Therefore it probably makes sense to move the most-likely option (which I'm assuming is smallest drive) to the top of the list?

[14 more replies]

closed 2018-10-15 by jviotti