balena-io/etcher #892: Problem with the .gz file size ?

AlainGourves, 2016-11-23:
- **Etcher version: 1.0.0-beta16**
- **Operating system and architecture: OSX 10.11.6**

I make backups of SD cards with this script in the Terminal : 

```
diskutil unmountDisk /dev/disk1
sudo dd bs=1m if=/dev/rdisk1 | gzip -c > ~/Desktop/pi.img.gz
```
When I try to flash these `.gz` images, Etcher seems to get stuck at 99% of the process, ETA remans at 2 or 3 seconds, the transfer rate very slowly decreases, until I get the error :
"Not enough space on the drive. Please insert larger one and try again".
The same error appears with the uncompressed image file, but without the final slow-down.

However, on the command line, with :

```
gzip -dc ~/Desktop/pi.img.gz | sudo dd bs=1m of=/dev/rdisk1
```
there's no error and the resulting SD card works well. Am I doing something wrong or is there a problem with Etcher ?

AlainGourves, 2016-11-24:
I've tried with `.bz2`and `.xz` files and got the same error. There's nothing much meaningfull in the DevTools console, but many messages and errors related to USB and `.fseventsd` (as far as I can understand) in OSX's console.

Could my problem be related the presence of invisible files on the SD card which would modify the avalaible free space ?

Here are the messages of the last 2 minutes : 

```
24/11/2016 12:27:02,000 kernel[0]: USB device 05E3075126431000 - will be reset!
24/11/2016 12:27:03,000 kernel[0]: USB device 05E3075126431000 - fConsecutiveResetCount = 1.
24/11/2016 12:27:03,000 kernel[0]: disk1s1: media is not present.
24/11/2016 12:27:03,000 kernel[0]: disk1s1: media is not present.
24/11/2016 12:27:03,000 kernel[0]: disk1s1: media is not present.
24/11/2016 12:27:03,000 kernel[0]: disk1s1: media is not present.
24/11/2016 12:27:03,000 kernel[0]: disk1s1: media is not present.
24/11/2016 12:27:03,000 kernel[0]: msdosfs_fat_uninit_vol: error 6 from msdosfs_fat_cache_flush
24/11/2016 12:27:03,420 sharingd[401]: 12:27:03.419 : SDConnectedBrowser::_CFURLGetVolumePropertyFlags returned Error Domain=NSCocoaErrorDomain Code=4 "Le fichier « boot » n’existe pas." UserInfo={NSURL=file:///Volumes/boot/, NSFilePath=/Volumes/boot}
24/11/2016 12:27:03,420 sharingd[401]: 12:27:03.419 : SDConnectedBrowser::getResourceValue Error Domain=NSCocoaErrorDomain Code=4 "Le fichier « boot » n’existe pas." UserInfo={NSURL=file:///Volumes/boot/, NSFilePath=/Volumes/boot}
24/11/2016 12:27:03,517 fseventsd[50]: disk logger: failed to open output file /Volumes/boot/.fseventsd/fc007533b6276d84 (No such file or directory). mount point /Volumes/boot/.fseventsd
24/11/2016 12:27:03,518 fseventsd[50]: disk logger: failed to open output file /Volumes/boot/.fseventsd/fc007533b6276d84 (No such file or directory). mount point /Volumes/boot/.fseventsd
24/11/2016 12:27:03,518 fseventsd[50]: failed to unlink old log file /Volumes/boot/.fseventsd/fc007533b6276d84 (No such file or directory)
24/11/2016 12:27:03,518 fseventsd[50]: failed to unlink old log file /Volumes/boot/.fseventsd/fc007533b6276d84 (No such file or directory)
24/11/2016 12:27:03,518 fseventsd[50]: unmounting: failed to remove log dir /Volumes/boot/.fseventsd (No such file or directory)
24/11/2016 12:27:08,000 kernel[0]: USBMSC Identifier (non-unique): 0x5e3 0x751 0x1404
24/11/2016 12:27:08,000 kernel[0]: USB device 05E3075126431000 - will be reset!
24/11/2016 12:27:09,000 kernel[0]: USBMSC Identifier (non-unique): 0x5e3 0x751 0x1404
24/11/2016 12:27:09,000 kernel[0]: USB device 05E3075126431000 - fConsecutiveResetCount = 1.
24/11/2016 12:27:10,000 kernel[0]: USB device 05E3075126431000 - will be reset!
24/11/2016 12:27:10,000 kernel[0]: 837908.222100 IOUSBHostDevice@26431000: IOUSBHostDevice::start: failed to get device descriptor
24/11/2016 12:27:10,000 kernel[0]: 837908.222305 AppleUSB20HubPort@26431000: AppleUSB20HubPort::resetAndCreateDevice: failed to start device
24/11/2016 12:27:10,000 kernel[0]: 837908.222310 AppleUSB20HubPort@26431000: AppleUSB20HubPort::resetAndCreateDevice: failed to create device (0xe00002e9), disabling port
24/11/2016 12:27:10,000 kernel[0]: 837908.222600 USB Storage@26431000: AppleUSBDevice::start: failed to open _deviceClient
24/11/2016 12:27:10,000 kernel[0]: 837908.222604 AppleUSBLegacyRoot@: AppleUSBLegacyRoot::usbServiceCallGated: failed to attach or start legacy service
24/11/2016 12:27:10,000 kernel[0]: USB Storage: not registry member at registerService()
24/11/2016 12:27:11,000 kernel[0]: USBMSC Identifier (non-unique): 0x5e3 0x751 0x1404
24/11/2016 12:27:11,000 kernel[0]: USB device 05E3075126431000 - fConsecutiveResetCount = 2.
24/11/2016 12:27:11,000 kernel[0]: USB device 05E3075126431000 - will be reset!
24/11/2016 12:27:13,000 kernel[0]: USBMSC Identifier (non-unique): 0x5e3 0x751 0x1404
24/11/2016 12:27:13,000 kernel[0]: USB device 05E3075126431000 - fConsecutiveResetCount = 3.
24/11/2016 12:27:13,000 kernel[0]: USB device 05E3075126431000 - fConsecutiveResetCount = 4.
24/11/2016 12:27:13,000 kernel[0]: USBMSC Identifier (non-unique): 0x5e3 0x751 0x1404
24/11/2016 12:27:15,000 kernel[0]: USB device 05E3075126431000 - will be reset!
24/11/2016 12:27:15,000 kernel[0]: 837913.397716 IOUSBHostDevice@26431000: IOUSBHostDevice::start: failed to get device descriptor
24/11/2016 12:27:15,000 kernel[0]: 837913.398065 AppleUSB20HubPort@26431000: AppleUSB20HubPort::resetAndCreateDevice: failed to start device
24/11/2016 12:27:15,000 kernel[0]: 837913.398073 AppleUSB20HubPort@26431000: AppleUSB20HubPort::resetAndCreateDevice: failed to create device (0xe00002e9), disabling port
24/11/2016 12:27:16,000 kernel[0]: USB device 05E3075126431000 - fConsecutiveResetCount = 1.
24/11/2016 12:27:17,000 kernel[0]: USBMSC Identifier (non-unique): 0x5e3 0x751 0x1404
24/11/2016 12:27:17,000 kernel[0]: 837915.819720 IOUSBHostDevice@26431000: IOUSBHostDevice::start: failed to get device descriptor
24/11/2016 12:27:17,000 kernel[0]: 837915.819986 AppleUSB20HubPort@26431000: AppleUSB20HubPort::resetAndCreateDevice: failed to start device
24/11/2016 12:27:17,000 kernel[0]: 837915.819990 AppleUSB20HubPort@26431000: AppleUSB20HubPort::resetAndCreateDevice: failed to create device (0xe00002e9), disabling port
24/11/2016 12:27:17,000 kernel[0]: USB device 05E3075126431000 - fConsecutiveResetCount = 1.
24/11/2016 12:27:18,000 kernel[0]: USBMSC Identifier (non-unique): 0x5e3 0x751 0x1404
[... cut ...]
24/11/2016 12:28:31,968 fseventsd[50]: disk logger: failed to open output file /Volumes/boot/.fseventsd/fc007533b6277542 (No such file or directory). mount point /Volumes/boot/.fseventsd
24/11/2016 12:28:31,968 fseventsd[50]: failed to unlink old log file /Volumes/boot/.fseventsd/fc007533b6277542 (No such file or directory)
24/11/2016 12:28:31,968 fseventsd[50]: failed to unlink old log file /Volumes/boot/.fseventsd/fc007533b6277542 (No such file or directory)
24/11/2016 12:28:31,968 fseventsd[50]: unmounting: failed to remove log dir /Volumes/boot/.fseventsd (No such file or directory)
```

jviotti, 2016-11-24:
Hi @AlainGourves ,

Can you link us to the exact image you're using, so we can try to reproduce here? Also, does the problem still occur if you uncompress the file and flash the `.img` directly?

lurch, 2016-11-28:
See also #638

AlainGourves, 2016-11-30:
Hi, sorry for the delay !
Here’s the link to one of the images :
https://drive.google.com/open?id=0B2T3Cvxr7LlOa19vemtuQmh4ZFE

It’s a `.bz2` file, but the uncompress `.img` didn’t work either (also it worked using `dd`).
Thanks for your work.

jviotti, 2016-11-30:
Hi @AlainGourves ,

I was able to reproduce your issue with a 15 GB USB drive, and found something very interesting.

This is the size, in bytes, of the image you shared with us (uncompressed):

```
$ stat -f '%z' pi-23-11-2016.img
15931539456
```

This is the size of my USB drive:

```
$ diskutil info /dev/disk2
...
   Disk Size:                15.5 GB (15522070528 Bytes) (exactly 30316544 512-Byte-Units)
...
```

As you can see, the image is larger than my drive (15931539456 - 15522070528 = 409468928, which equals 390.5 MB), so the "Not enough space on the drive. Please insert larger one and try again" is legit in my case.

Lets try using `dd` now:

```
$ sudo dd if=pi-23-11-2016.img of=/dev/rdisk2 bs=4m
Password:
3701+0 records in
3700+1 records out
15522070528 bytes transferred in 4638.619943 secs (3346269 bytes/sec)
```

Surprisingly, `dd` will not complain but only copy the first 15522070528 bytes of the image, **and completely ignore the rest!**.

In summary, I think Etcher is actually protecting you from yourself, while `dd` is not flashing the whole thing, but please let me know if this is not the case!

closed 2016-11-30 by jviotti

jviotti, 2016-11-30:
Oh, wait. You're re-flashing the images you've previously backed up, which should definitely fit your drive (assuming you're flashing to the same drive). There was a very similar issue we've fixed in v1.0.0-beta.17, which might be the reason I was unable to reproduce.

Can you try v1.0.0-beta.17 and let us know how that goes?

lurch, 2016-12-01:
Slightly off-topic:

> Surprisingly, dd will not complain but only copy the first 15522070528 bytes of the image, and completely ignore the rest!

There _is_ actually a use-case for that - when you're doing something like `dd if=/dev/zero of=/dev/sdc` to completely wipe out the contents of a drive. `dd` is a very low-level tool, which means it's easy to abuse or get things wrong if you're not *very* careful ;-)

Back on-topic:
Specifically, the bug that was fixed in v1.0.0-beta.17 that @jviotti 's referring to is that previously Etcher would only write images with a size equal to an integer number of megabytes. 15931539456 equates to 15193.5 MB, and Etcher v1.0.0-beta.17 now copes with the extra-half-megabyte on the end https://github.com/resin-io-modules/etcher-image-write/pull/58 .

AlainGourves, 2016-12-02:
My shitty USB/SD card adaptor died in the meantime, I will try the last beta as soon as I get the one currently lost between China and France.

Still, I have an idea (although I’m not an expert of the command line) :
If `dd` is a low-level tool easy to abuse, could the culprit be the `bs=1m`option used to create the backup image ? I assume that `dd`create blocks of the same size, even if `total size of the SD card modulo 1024 != 0`, thus creating an image file greater than the original. 
That’s what I understand from `dd`man page :
>Normally, data resulting from input or conversion or both are aggregated into output blocks of the
>specified size.  After the end of input is reached, any remaining output is written as a block.  This
>means that the final output block may be shorter than the output block size.

lurch, 2016-12-02:
Nah, I think the snippet you've quoted basically says that the output file may not be an exact multiple of the block-size. So the output file will always be the exact size of the input file (the raw SD card in your case), regardless of the block size.

I think `My shitty USB/SD card adaptor died` _may_ be the cause of the errors in the log you posted ;)
So you were actually seeing two problems at once - errors in your logs because of a shitty SD adaptor, and the "Not enough space on the drive" error caused by Etcher beta16 not coping with half-megabyte sized disk images.

Hopefully both problems will be fixed by Etcher beta17 and a new SD adaptor :-)