balena-io/etcher #898: Invalid message error with Etcher

Agmka, 2016-11-25:
- **Etcher version:** version 1.0.0-Beta.16
- **Operating system and architecture:** Windows 7 Professional (32 Bit)
- **Do you see any meaningful error information on DevTools?**
Nothing opens when I press  `Ctrl+Alt+I`,

I am using Etcher to Flash an ISO file of Zorin OS and  errors occur. See the attached screen dump

[image]

jviotti, 2016-11-25:
Hm, that's really strange. I can't spot any issue on the JSON object being presented. I'll install Windows 7 and give it a go and see if I can reproduce over here.

WasabiFan, 2016-11-25:
@jviotti there are two JSON objects there. Is it possible that two messages are being combined in the data stream?

Agmka, 2016-11-25:
I don't know, (don't think your question was at me). I need more RAM, the
reason I am trying to install Zorin OS. I will reinstall Etcher and try
again.

Agmka, 2016-11-27:
The Etcher program resolved. I installed more RAM and Etcher Flashed the ISO file with no problem, indicating it was a RAM problem. However I can't install Zorin OS as there is no folder on my desktop when I reboot with this Flash Drive flashed by Etcher. I know it is a Zorin issue, but can anybody advise me what to do?

Agmka, 2016-11-27:
I see now there is a way I can ask for help on the Zorin site. Going to do this now.

jviotti, 2016-11-28:
> I installed more RAM and Etcher Flashed the ISO file with no problem, indicating it was a RAM problem.

I doubt its a RAM problem. Its probably an sporadic throttling issue of the IPC communication between the CLI and the GUI as @WasabiFan pointed out. I'll try to trigger this issue myself here.

lurch, 2016-11-28:
If @Agmka is indeed running low on RAM, then maybe his system was swapping heavily (i.e. really high IO load), and that's why the GUI missed one of the IPC messages and tried consuming two at once?

Given that it looks like the JSON objects are printed one-per-line, maybe this issue would be resolved by splitting the IPC message on newlines, and only trying to parse the last line as a JSON object?

jviotti, 2017-01-04:
I sent a fix here: https://github.com/resin-io/etcher/pull/997. See the commit body for an explanation of the underlying issue.

@lurch Do you think it makes any difference to emit different messages for each line encountered here instead of just emitting the last one?

If the machine is having issues processing IPC messages at a fast rate, emitting every message might cause the machine to keep missing more messages.

jviotti, 2017-01-05:
Okay, I ended up only emitting the last one, since missing progress state events doesn't have any negative effect, and there is no advantage on processing multiple progress state objects coming at almost the same time.

closed 2017-01-05 by jviotti

lurch, 2017-01-13:
Is it "guaranteed" that e.g. we won't get a progress message after an error message?

With
```
progress X
progress Y
```
then it's obviously fine to only send the last `progress Y` message, but with
```
error Z
progress Y
```
then we wouldn't want to miss the error message by only sending the progress message?

jviotti, 2017-01-13:
Great observation! I'll send a PR fixing this in a minute.