balena-io/etcher #900: Error while flashing Raspberry pi

santoo-v, 2016-11-26:
- v1.0.0-beta.16 - 2016-10-28
- Windows 7 64bit
- **Do you see any meaningful error information on DevTools?**

[image]

tracker.js:10 Error: operation timed out after 4040 ms, 5 tries with error: ENOENT: no such file or directory, open 'C:\Users\skva\AppData\Local\Temp \_diskpart-1480188143206'
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird-retry\lib\bluebird-retry.js:95:35
    at tryCatcher (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\util.js:16:23)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\catch_filter.js:24:41
    at tryCatcher (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\util.js:16:23)
    at Promise._settlePromiseFromHandler (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:504:31)
    at Promise._settlePromise (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:561:18)
    at Promise._settlePromise0 (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:681:18)
    at Async._drainQueue (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\async.js:138:16)
    at Async._drainQueues (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\async.js:148:10)
    at Immediate.Async.drainQueues (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\async.js:17:14)
    at runCallback (timers.js:574:20)
    at tryOnImmediate (timers.js:554:5)
    at processImmediate [as _immediateCallback] (timers.js:533:5)

jviotti, 2016-11-28:
This is coming from https://github.com/resin-io-modules/etcher-image-write/blob/master/lib/win32.js#L67, which means Etcher was not able to clean the drive after 5 tries for some reason. The final attempt failed with ENOENT, which is weird, and might indicate a subtle bug in the way we handle retrying, but we should investigate further why cleaning the drive has failed so many times in the first place.

santoo-v, 2016-11-28:
Thanks for response. I was however was manually be  able to the disk part operations.

So, I think as this is very specific to my machine or   memory card, you may close the bug report. 

Thanks again.

jviotti, 2016-11-28:
Hi @santoo-v,

Thanks for the response! While this might not be a common issue, I'd like to keep it open for a while, to see if I can somehow reproduce.

If it happened to you, then its likely that it will affect at least a minority of our users, and therefore deserves to be properly investigated :)

LittleDan9, 2017-01-21:
Experiencing similar issue also trying to write a Rasperian Image. I tried on three different cards. On two different readers. During the process on one of the much slower readers, Windows attempted to mount the flash disk 4 times, I assume this was after each attempt to erase the disk. SanDisk Ultra Plus 32GB HCI Cards. Version 1 Beta 18

[image]

jviotti, 2017-01-23:
Hi @LittleDan9 ,

Thanks a lot for the info. What SDCard brand and models were you using? We are having a really hard time reproducing this internally to be able to fix, so we think it might be related to certain SD Cards.

jviotti, 2017-01-23:
BTW, I just noticed this is a duplicate of https://github.com/resin-io/etcher/issues/989. Let's continue discussion there!

closed 2017-01-23 by jviotti

shimano1974, 2017-12-26:
Hello, i have used etcher many times and all the sudden today when i launched etcher before trying to mount an image it gives msg No such file or directory: undefined the file you are trying to access doesnt exist. Nothing loaded yet. So i tried old versions new versions. Deleting files temp files from app data. Uninstalling and reinstalling so i change machines and go to my desktop and launch a version 1.1.2 X64 i have used 100 times and it did same thing. Unknown?

jhermsmeier, 2017-12-26:
Hey @shimano1974 please open a new issue for that; this is an unrelated issue.