balena-io/etcher #912: can re-use my drive!

zactrixo, 2016-11-30:
- **Etcher version:**: 1.0.0-beta.17
- **Operating system and architecture:**: OS X El Capitan version 10.11.6
- **Do you see any meaningful error information on DevTools?**

Hi so i recently created an instalation drive using etcher and an ISO image for windows 10 but for some reason it doesnt work so i wanted to clear the drive to try another way but everytime i try trough disk utility it only says that the name isnt valide and that the operation failed (i am using the default name)

lurch, 2016-11-30:
> an ISO image for windows 10 but for some reason it doesnt work

https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

> i wanted to clear the drive to try another way but everytime i try trough disk utility it only says that the name isnt valide and that the operation failed (i am using the default name)

@jviotti knows more about OS X than I do, so maybe he'll be able to help out here.

zactrixo, 2016-11-30:
managed to solve it by completly reseting the drive on another computer and found out that OS X can alter NTFS file wich is why i couldnt reset it on my personnal computer but thank you for trying to help anyway!

closed 2016-11-30 by zactrixo

lurch, 2016-11-30:
Glad to hear you got it sorted :-)

jviotti, 2016-11-30:
Just for reference, we have guides to re-format drives in various operating systems here: https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives