balena-io/etcher #919: Settings Ui update

konmouz, 2016-11-30:
A few updates on how settings are displayed:
- The settings page is now replaced by a modal window (3 options are not enough to justify a whole page and '<back' button navigation was a bit tricky)
- Advanced settings have been moved to the [drive selection](https://github.com/resin-io/etcher/issues/888#issuecomment-263923906)

[image]

jviotti, 2016-11-30:
Looks good! (no pun intended :P)

lurch, 2016-12-01:
LOL at the button name :-)

konmouz, 2016-12-21:
Yesterday, I was light testing my flows with Gergely, and he suggested that 'auto-unmount on success' shouldn't be an option and should be enabled by default. Any ideas or objections on this?

lurch, 2016-12-21:
I guess a reason to keep it (i.e. to turn off auto-unmount) might be if somebody wants to e.g. edit `config.txt` on the boot partition, before removing the SD card?

Also, looking again at the image in the first post, I see there's no 'X' in the top-right corner to allow you to dismiss the settings modal without changing the existing settings? IMHO there should be.

konmouz, 2016-12-21:
but it will do exactly the same thing with the 'looks good' button if you don't change anything. And if you do change, it is not clear if changes will be saved or not.

jviotti, 2016-12-21:
Auto unmount on success is the default action for the huge majority of users, including myself. What is the use case to keep it mounted?

The only ones I can think of:

- Mounting the drive and editing files on it, which will be handled by our configuration feature
- Quickly checking that the image was flashed correctly (if it mounts, its *probably* fine), but we take care of this during our validation phase 

I'd love to hear about other use cases I'm not considering and that are not currently addressed.

lurch, 2016-12-21:
> And if you do change, it is not clear if changes will be saved or not.

IMHO the traditional "OK, Cancel" UX for a traditional preferences dialog is fairly clear-cut isn't it?

1. App starts up with previously-saved settings
1. User selects preferences dialog
      * Application displays preferences dialog initialised to previously-saved settings
1. User makes changes to preferences
    * Application only updates preferences dialog, doesn't touch saved settings
1. User clicks `OK`
    * Application saves (and uses) new settings
1. User selects preferences dialog
    * Application displays preferences dialog initialised to saved settings
1. User makes changes to preferences
    * Application only updates preferences dialog, doesn't touch saved settings
1. User clicks `Cancel`
    * Application ignores any changes made in the preferences dialog

Why shouldn't the settings modal in Etcher operate the same way? :confused:

konmouz, 2016-12-21:
>IMHO the traditional "OK, Cancel" UX for a traditional preferences dialog is fairly clear-cut isn't it?

I think this applies to something more complicated like the 'configure image'. I don't see why a user would like to cancel selections on a three tick options modal window. Most settings in modern UIs don't even have a confirmation CTA, like burger menus or 'x' on modals (changes will be saved anyway).

lurch, 2016-12-21:
> I think this applies to something more complicated...

I don't mean to sound argumentative, but is it possible that the Etcher settings window will get more complicated in future? (by which point changing the UX would be even more jarring?)

But then I'm a bit old-fashioned, so wouldn't really know what "modern UIs" do ;)

jviotti, 2016-12-21:
I think the difference is that on the settings modal actions take effect instantly after you tick/untick on them, instead of being saved into a temporary stage that is later applied or cancelled with a submission button.

konmouz, 2016-12-21:
> I don't mean to sound argumentative, but is it possible that the Etcher settings window will get more complicated in future? (by which point changing the UX would be even more jarring?)

I get the point, but it is the same flow with the current one. Instead of clicking '<back' you click 'looks good'.

lurch, 2016-12-22:
To me, the fact that the current settings page is on the same "level" as the main UI, reinforces the fact that the settings are applied instantly. But with the settings being in a modal window, they're conceptually at a different "level" to the main UI, and so I wouldn't expect changes made in the settings modal to get "applied" to the main UI until the settings modal has had its `OK` button clicked.

Hmmm, although I've just noticed that Gedit's preferences dialog window only has a `Close` button, rather than `OK` & `Cancel`, and does in fact do 'live changes' in the main window when settings in the prefences window are changed.
[image]

alexandrosm, 2016-12-22:
that fits my "modals define transactions" intuition. When you have a modal,
you're essentially "freezing" the product and stepping out to specify a
complex requirement, which you then go and apply atomically.

konmouz, 2016-12-22:
I completely agree for the 'ok' button, and I think it more clear for the user compared to the 'close' one.