balena-io/etcher #929: Disk list opening "inside" aplication

eSkiSo, 2016-12-02:
- **Etcher version:** Etcher-1.0.0-beta.17
- **Operating system and architecture:** Windows 10 64bits
- **Do you see any meaningful error information on DevTools?** Nop

When you have more then 4 drives the list grows inside the application and you can only select another disk by tabbing
 
[image]

After tab (to select disk under those)
[image]

*Usb pen not present at the time*

eSkiSo, 2016-12-02:
Downloaded portable version and it looks different, so it may just be that the normal installer has packed with an older version?

eSkiSo, 2016-12-02:
First download also downloaded that weird win32 version (that i installed and was causing this issue), new version downloaded.
[image]

lurch, 2016-12-03:
On the gitter channel @eSkiSo also identified that there's a link to `https://resin-production-downloads.s3.amazonaws.com/etcher/1.0.0-beta.0/Etcher-win32-x86.exe` which is presumably wrong... [screenshot](https://files.gitter.im/resin-io/etcher/q7LS/blob)
ping @craig-mulligan
(and this is presumably also the cause of #789 )

WasabiFan, 2016-12-03:
Ah, interesting; so presumably, the Etcher site is using that version as a fallback if it can't contact the server to get an updated list. Can that be changed so that it defaults to the GitHub releases page or some other complete list (so the user chooses manually)? That's what I have done with JS-based release pages like this in the past.

alexandrosm, 2016-12-03:
That does not look like beta17. It seems exceedingly weird @jviotti, if
that's what we're shipping, we have an emergency!

alexandrosm, 2016-12-03:
We don't want to make gh releases as those are very hard / impossible to
get analytics out of

WasabiFan, 2016-12-03:
> We don't want to make gh releases as those are very hard / impossible to
get analytics out of

I'm not sure what you mean. GitHub provides download counts for individual files in every release.

Either way, it doesn't really matter what page you link to; what I'm thinking is that you should set the fallback URL to be a list of releases instead of that old one directly.

alexandrosm, 2016-12-03:
agreed, don't want to fall back to the oldest one. in fact, @jviotti what's
stopping us from simply re-publishing the website upon a new release? that
would save us this weird indirection/fallback situation

lurch, 2016-12-03:
> what's stopping us from simply re-publishing the website upon a new release?

I guess that'd simply be another step to add to https://github.com/resin-io/etcher/blob/master/docs/PUBLISHING.md

jviotti, 2016-12-04:
> That does not look like beta17. It seems exceedingly weird @jviotti, if
that's what we're shipping, we have an emergency!

It seems to be an issue on the website, which is pointing to a very old release instead of to the latest one for certain user agents.

> in fact, @jviotti what's
stopping us from simply re-publishing the website upon a new release? that
would save us this weird indirection/fallback situation

We've discussed this with @craig-mulligan on the summit (having a static set of links updated on each release rather than "clever" logic to find the latest one automatically), and I think this is how the re-designed site works already, but we're not deploying it anywhere yet, though I'm not sure why. @craig-mulligan?

As a quick solution, lets solve this problem on the user-agent/latest version detection logic and lets focus on getting the new site up and running.

craig-mulligan, 2016-12-04:
> but we're not deploying it anywhere yet, though I'm not sure why. 

I haven't had any time since the summit, and we have the new lander ideas to include, hopefully decemeber will be quiet and we can upgrade everything. 

> As a quick solution, lets solve this problem on the user-agent/latest version detection logic and lets focus on getting the new site up and running.

Sure I'll knock up a PR quickly @jviotti

craig-mulligan, 2016-12-04:
@jviotti I added a downloads section: https://etcher.io/#downloads and updated the fallback.

lurch, 2016-12-04:
Nice. I guess that has the unfortunate side-effect of highlighting just how "big" the Etcher downloads are though ;-)   (maybe it's worth adding a small note that this is because Etcher is built on top of Electron, and not because Etcher is "bloatware"?)

@jviotti & @dlech Should the `.deb` file be included in that list too? Or maybe the https://github.com/resin-io/etcher/blob/master/README.md#installers section of the GH Readme should also be included on the etcher.io website somehow?

lurch, 2016-12-04:
Has the "slogan" on the website now 'reverted' back to `Burn images to SD cards & USB drives, safe & easy.`? I'm fairly sure that @jviotti updated it to `Flash OS images to SD cards & USB drives, safely and easily.` as part of https://github.com/resin-io/etcher/issues/797#issuecomment-257968786
(Or maybe I'm just mis-remembering?)

dlech, 2016-12-04:
> Should the .deb file be included in that list too? Or maybe the https://github.com/resin-io/etcher/blob/master/README.md#installers section of the GH Readme should also be included on the etcher.io website somehow?

I don't think the debian file needs to be in the Downloads section. But I do think that there should be an "Other installation methods" section that lists supported package managers. I guess for now, this is just the debian package, but it could be expanded to include a Chocolaty package for Windows, brew on Mac, rpm on other Linux distros, etc.

jviotti, 2016-12-05:
@craig-mulligan 

> @jviotti I added a downloads section: https://etcher.io/#downloads and updated the fallback.

Can you point us to the commit/PR? What is the fallback now? Can we close this issue?

> I don't think the debian file needs to be in the Downloads section. But I do think that there should be an "Other installation methods" section that lists supported package managers. I guess for [...]

Lets include this in the re-designed website, in order to avoid work in a landing page that will be replaced in the near future.

jviotti, 2016-12-05:
@craig-mulligan re: https://github.com/resin-io/etcher-homepage/commit/43034e794be88adfbdeafbaf117ba1e4b423ba47

Unless we get the re-designed website up and running very soon, we'll have to manually update the fallback the next time we release. What about simply making the button scroll the website to the "downloads" section, and completely get rid of the concept of a "fallback"?

alexandrosm, 2016-12-05:
> resin-io/etcher-homepage@43034e7

@jviotti -- is it that bad to update the landing page on every release?

lurch, 2016-12-05:
OTOH if we *do* end up updating the landing page on every release (presumably via some semi-automated step of the publishing process), then I guess we could just 'hardcode' all the relevant URLs in the page anyway, and then there wouldn't be any need for a "fallback URL" at all.

jviotti, 2016-12-05:
The thing is that as it is now, the latest version is dynamically calculated, but we still have to remember to update the fallback. When we have the new landing page, it will be integrated into the main Etcher repository, which means it will be able to get the latest version from `package.json` automatically, so we won't have to manually update every time. 

Until then, it sounds simple to point to the downloads table and avoid having to maintain even the fallback. I agree its not a big deal, but sounds like an easy win to me.

jviotti, 2017-01-25:
Closing since this issue was already resolved.

closed 2017-01-25 by jviotti