balena-io/etcher #932: Etcher should bring itself to the foreground when an image is selected

WasabiFan, 2016-12-04:
- **Etcher version:** Beta 17
- **Operating system and architecture:** Windows 10 x64

When I click the button to open the file selection window, the file picker opens on top of Etcher. I can then switch windows, minimize the main Etcher window, etc -- in my case, I was opening a browser to download the image I wanted after I realized that I didn't have it locally. Once I select a file, the picker goes away but Etcher stays minimized/in the background. I think that Etcher should bring itself to the foreground when this happens so that I don't have to go searching for the main Etcher window.

jviotti, 2016-12-04:
Sounds like a good addition at first sight, however is this the normal behaviour in other applications in the operating systems we're targeting? Maybe @konmouz can provide some input here.

WasabiFan, 2016-12-04:
I can't speak to other operating systems, but I definitely expected Etcher to pop back up after selecting the image; I can't say whether that's a universal action or not.

In general for most apps, the file picker behaves as a modal dialog; I can't, for example, put something between the file picker and the Word window that launched it. In that case, there is still only one "window" entry in the taskbar and the dialog is rendered at the same level as its parent window. I also can't manipulate the parent window's position or buttons while the picker is open. This is the same for Skype, Edge and a few other apps I tested, including a test app I just wrote. My guess is that this divergent behavior stems from the way Electron handles dialogs.

I can't imagine a scenario where you would click the "Open" button and _not_ want to proceed with Etcher's process immediately, so it seems like a safe bet.

lurch, 2016-12-04:
Yeah, I'd classify this as a "modal dialog" bug, not as a "Etcher doesn't pop back to the foreground" bug ;-)

I just checked with `atom`, and when the file-dialog is open in that, and you minimise the parent application, the file-dialog gets hidden too, and only gets redisplayed when the parent app is restored (which is the behaviour I'd expect). This shows that this is (possibly) a bug just with `Etcher`, and not with all `electron`-based apps.

jviotti, 2016-12-05:
> I just checked with atom, and when the file-dialog is open in that, and you minimise the parent application, the file-dialog gets hidden too, and only gets redisplayed when the parent app is restored 

I tested this in OS X, and I'm unable to minimize/close the parent window until I dismiss the modal, which is the intended behaviour. I'll check in other OSes soon, but I infer from your comments that this is not the case there.

If so, we can always workaround ourselves, but I think it'd be good checking that this is not something that has been fixed in later Electron versions (which can be the reason @lurch can't reproduce on Atom), or if we should raise in Electron directly instead.

lurch, 2016-12-05:
I was testing on Ubuntu 14.04 x64 - hadn't checked any other OSes yet.

konmouz, 2016-12-05:
Indeed the file picker behaves pretty much as a modal for Etcher. Indeed on OSX you can't mesh with the parent window while having the file browser open, which is a clear flow. I guess we can deactivate the minimizing of the main window while the picker is open, or indeed bring etcher to the foreground when an image is confirmed.

jviotti, 2017-01-24:
This has been fixed for OS X (at least) by https://github.com/resin-io/etcher/pull/1009. I'm testing on the remaining operating systems.

jviotti, 2017-01-24:
The issue is fixed for GNU/Linux as well.

jviotti, 2017-01-25:
I'm closing this issue. Its being fixed in all OSes.

closed 2017-01-25 by jviotti

zwackelmanderl, 2022-05-08:
Not solved have the current version - etcher always goes to the background
I'm uninstalliung this s.....oftware