balena-io/etcher #950: Select drive as source

alexandrosm, 2016-12-06:
This is a three-purpose feature request.

Overtly, it's about having the ability to select a drive as a source. So, imagine having two sd drives connected, and selecting the one as a source, and the other as a destination. I think this is a perfectly legitimate Etcher feature. Interestingly, a system drive would be a legitimate thing to select as a source in this case.

Covertly, this is about allowing the selection of things "other than files from the filesystem" as sources. In the future we could add urls, torrents, and other things as sources. If we work out the interaction for having files and drives, we can extend it to other things as they become available.

Even more covertly, this is one of the features we'd need for a hardware version of Etcher. ;)

jviotti, 2016-12-06:
I like this idea. In order to achieve it, we should first sort out how to determine how much of the drive to copy, which is exactly what we need for the backup feature as well (https://github.com/resin-io/etcher/issues/266).

I'll mark this issue as "pending design", so we can start figuring out how to properly present this paradigm shift to the user while we sort out the technical side in the backup ticket,

jviotti, 2016-12-06:
/cc @konmouz

WasabiFan, 2016-12-06:
Theoretically, if you always represent these data sources/destinations as abstract streams, would you just be able to drop-in a disk stream in place of a file one (or a Node.js buffer or whatever internal structure you use)? My thinking is essentially that you could have both the source and destination be completely generic internally and then just specialize the UI to facilitate selection. With something like this, it would presumably be easy for someone to implement a new source or destination, because you wouldn't need to adapt internal switching logic.

lurch, 2016-12-07:
@WasabiFan Yeah, and that's kinda covered by the `etcher-image-write` and `etcher-image-stream` modules as mentioned in https://github.com/resin-io/etcher/blob/master/docs/ARCHITECTURE.md

I agree with @jviotti that this PR sounds very much like a logical follow-on from #266

alexandrosm, 2016-12-07:
To me it sounds like a precursor. If you had a drive as a source and a file as a destination, voilà, backup

konmouz, 2016-12-07:
We can change the UI from 'Image' and 'Drives' to 'Source' and 'Destination'. Probably when you select a drive as source, it should look unavailable on the destination side. Regarding backup, it is true that it will be covered via this feature, however, it might worth having a 'backup' CTA as a more direct approach (mainly for new users).

alexandrosm, 2016-12-07:
agreed that there may be a need for a backup side-flow even if one could
put it together with these primitives.

having abstract source and destination/target selectors was one of the
earliest things we were talking about with @jviotti and I kinda let it drop
so we could focus on the core, interesting to see it coming back, means it
wasn't a terrible idea, probably.

lurch, 2016-12-07:
I guess when selecting the 'Destination' as a file though, you'd only want to be able to select a single file, rather than selecting multiple files as in the multi-write mode. And it probably doesn't make much sense to read from a file and write to another file (unless you wanted to specifically enable that as a verification / testing mode).
So I guess the allowed combinations would look something like:
 * read-from-file -> write to one or more drives (typical disk-flashing mode)
 * read-from-drive -> write to one or more drives (disk-cloning mode)
 * read-from-drive -> write to a single file (backup mode)

> Probably when you select a drive as source, it should look unavailable on the destination side.

Agreed. And when selecting a file as destination, you'd need to make sure it's not on the drive you're going to be reading from (which we can do, now that drivelist reports which mount-points each drive uses).

When reading-from-drive we'll need to unmount the drive first (so that we can access it in raw mode), so it might be a nice feature to re-mount the drive again at the same mount-point(s) after we've finished reading from it? (if it's possible to do that reliably)
Hmm, and perhaps that means we need another category of drives for drivelist to detect - drives that we "know" we can't unmount (which will include at a minimum the "boot drives"), and therefore will never be able to read from?

jviotti, 2016-12-07:
I think we should really consider this paradigm change now, since it means the whole Etcher workflow would change, and better to at least consider how this would look like before adding more features, otherwise it will be a pain to introduce later on.

konmouz, 2016-12-07:
>- read-from-file -> write to one or more drives (typical disk-flashing mode)
- read-from-drive -> write to one or more drives (disk-cloning mode)
- read-from-drive -> write to a single file (backup mode)

Just to add 'read-from-online source -> write to one or more drives' which I guess UI-wise would be slightly different from 'read-from-file -> write to one or more drives'.
Also, if it makes sense 'read-from-online source -> write to a single file'. Well, this sounds like download, so let me know if there is such a use case.

lurch, 2016-12-07:
Not sure if 'read-from-online source -> write to a single file' "works" as a concept, but when doing 'read-from-online source -> write to one or more drives' it might be nice to have an option to cache (or save) the image file (and its metadata?) locally, so that if you want to flash an online-image multiple times, you can just use the saved-version instead of streaming it again.

jviotti, 2016-12-07:
> Also, if it makes sense 'read-from-online source -> write to a single file'. Well, this sounds like download, so let me know if there is such a use case.

I think we can leave that out for now.

alexandrosm, 2016-12-08:
I am not sure why we are enumerating combinations if the two sides will be
independent?

konmouz, 2016-12-08:
I guess for purposes like if you select an A drive as source, probably A will be unavailable on the destination side. Am I wrong?

alexandrosm, 2016-12-08:
sure, just want to make sure we are thinking about the two things as
independent and not tying ourselves to specific patterns.

maybe we should list the special cases instead (like the one you mentioned
above)?

konmouz, 2016-12-08:
For v.01, just to double check:
- Can a destination be online?
- Can you edit the files of a connected device or drive? Which means that source and destination is the same. Probably this is configuration.

lurch, 2016-12-08:
Sorry if I was causing confusion. Yeah the two sides are "independent", but there are some combinations that don't make sense, so I was just trying to whitelist the valid combinations, rather than blacklist the invalid combinations.

In the current version of Etcher, the only "source" is "local file" and the only "destination" is "local drive".
I guess our currently-planned "sources" are: "local file", "online stream", "local drive"; and our "destinations" are: "local file", "local drive(s)". (But obviously there's nothing precluding other options being added to either of these lists).

alexandrosm, 2016-12-08:
I would replace "online stream" with simply "url" but otherwise yeah, it
all checks out. In cases like this I generally prefer blacklists than the
other way around, as they are true-er to the generative nature of
independent primitives.

konmouz, 2016-12-08:
regrading "online stream", except of a url, would it be possible to have a dropdown/ filter search UI element to select an image from an online source? At least for popular images would be great. (e.g. type 'Ras' and get a list of different Raspbian Image Versions)

lurch, 2016-12-08:
Fair enough. In that case the blacklist is:
 * local file -> local file
 * url -> local file

(and with the additional filtering that @konmouz mentioned of not being able to select the same local-drive as both source and destination; and if doing local-file -> local-drive or local-drive -> local-file we should ensure that the local-file isn't on a filesystem that's on the drive being written to or read from)

lurch, 2016-12-08:
> would it be possible to have a dropdown/ filter search UI element to select an image from an online source?

Yeah, something like that _is_ planned in the Etcher roadmap :)  @alexandrosm might be able to find up the doc and send it to you on flowdock.

jviotti, 2018-10-15:
Done in the latest SDK

closed 2018-10-15 by jviotti