balena-io/etcher #952: New drives are always removable

alexandrosm, 2016-12-06:
Not sure if this belongs here or in drivelist, but it just dawned on me that Etcher should consider drives that appeared after it was started as removable, even if detected as "system" for whatever reason. Is it a correct assumption to say that a system drive does not appear/disappear at runtime?

jviotti, 2016-12-06:
> Is it a correct assumption to say that a system drive does not appear/disappear at runtime?

Maybe. This means that removable USB hard drives, for instance, will be considered safe to write to, but the UX pattern definitely makes sense to me.

lurch, 2016-12-07:
I guess it also depends what we decide "system drive" to mean? AFAIK there's not really yet any good definition of what that means, and I wonder if there should be? (I think at the moment it's any drive which drivelist identifies as non-removable?)

jviotti, 2016-12-07:
Yeah, as mentioned in https://github.com/resin-io-modules/drivelist/issues/126, the proposal is to rename "system" to mean "non-removable" and add a new "system" property that is only true when that drive contains an operating system installed (e.g: the hard drive containing `/` in Linux, or usually `C:` in Windows).

After making such changes, the feature discussed here starts making more sense, since we'll whitelist non-removable drives that are non-system drives if they are inserted after Etcher is already running.

konmouz, 2016-12-07:
Will Etcher still auto-select a drive with the multi-write functionality? Does it make sense to auto-select a drive that appeared after Etcher started? Currently, the auto-select feature is great but wouldn't be that straightforward in the next version.

lurch, 2016-12-07:
I think the auto-selection does still make sense (even with multi-write). It would only auto-select if there was a single removable drive detected (even if that removable drive is inserted _after_ Etcher has started). Anything else will require manual user interaction.

alexandrosm, 2016-12-08:
However, this does imply that even after auto-selection, we would have a CTA button to open the drive menu, right?

konmouz, 2016-12-08:
Currently we use 'change' for that. There will be a similar interaction.

estevesd, 2017-08-15:
I'd love to see this fixed. I'd like to copy the image directly to a mSATA drive. For that I'm using a mSATA to USB adapter which is not detected as a removable drive. So back to Terminal.

lurch, 2017-08-15:
@estevesd Or you could simply enable the "unsafe mode" in Etcher's settings, which will then allow you to write to "system" drives :grinning:

konmouz, 2017-08-15:
At some point, it would be good to remove this option from settings and move it directly under the drives list as proposed.

lurch, 2017-08-15:
@konmouz Yeah, I still think #888 is worth doing, it just isn't top of our priority lists at the moment.

konmouz, 2017-08-15:
@lurch, makes sense.