balena-io/etcher #953: An easy way to test the update dialog / notifications

lurch, 2016-12-07:
**Development / testing only issue, not intended for use by the general public!**

At the moment, the only way to see the update dialog is either to run an old version of Etcher (which obviously only shows you the old update dialog), wait for a new version of Etcher to be released, or 'hack' the Etcher source code to fake an update notification.

As discussed in https://github.com/resin-io/etcher/pull/928#issuecomment-265185635 it would be nice if there was an easy way (for testing purposes only) to force the update dialog to display. Which would hopefully also make issues like [this](https://github.com/resin-io/etcher/pull/928#issuecomment-265216059) less likely to slip through the net.

konmouz, 2016-12-07:
On mac there is the following functionality which is pretty handy:

[image]

jviotti, 2016-12-07:
I like @konmouz suggestion, however this is something we should try to "hide" from the user, since its a developer testing feature, and an environment variable fits that definition very well.

In a side note, I'll change the "type:electron" label, since we usually reserve that one for issues that come from Electron (e.g: electron bugs that affects us).

lurch, 2016-12-07:
Yeah, (AFAIK) Etcher automatically checks for updates on every startup, so there doesn't really need to be an explicit "Check for Updates..." option. I guess that _would_ make sense if a program was likely to be left open for days at a time (and you wanted to check for updates without restarting the program), but for Etcher I can't see that being the case ;-)

Although I wonder @jviotti if there should also be a similar developer-testing environment variable to override the "don't remind me for 7 days" setting?

jviotti, 2016-12-07:
> Although I wonder @jviotti if there should also be a similar developer-testing environment variable to override the "don't remind me for 7 days" setting?

I don't want to bake in may developer testing features in Etcher. You can very easily unset that setting from DevTools (see the LocalStorage section in the Resources panel) (https://developers.google.com/web/tools/chrome-devtools/manage-data/local-storage?utm_source=dcc&utm_medium=redirect&utm_campaign=2016q3).

BTW, after writing the above paragraph, I think I have a new idea to introduce this change and other developer testing utilities:

What about exposing a collection of functions to the global scope that would do many useful stuff (like forcing the dialog to appear, or clear the "don't remind me for 7 days" setting, etc) so the developers can call them from DevTools? Since DevTools provides very nice autocompletion, it would be really easy to discover what's there and its still very developer-centric.

For example:

- Open DevTools
- Go to the Console pane
- Type:

```js
EtcherDevelopmentUtils.showUpdateNotifierDialog();
```

If this is indeed useful for out workflow (lets try brainstorming more things that we could add here), we could even create a DevTools extension that we load by default, which would provide a nice interface as an extra DevTools pane to call the development utilities (see http://electron.atom.io/docs/tutorial/devtools-extension/).

WasabiFan, 2016-12-07:
Just throwing this out there: [https://en.wikipedia.org/wiki/Konami_Code](Konami code) :laughing:

lurch, 2016-12-07:
Sounds good. And being able to (re)show the update dialog with `EtcherDevelopmentUtils.showUpdateNotifierDialog();` is even handier than setting the env-var because you wouldn't have to restart Etcher to get the dialog to display again :-)
(although it might be useful for this specific example to actually keep the env-var too, so that we can test the dialog-gets-displayed-at-startup feature)

Not sure that it warrants a whole DevTools Extension though ;-)

Might also be a useful place to prototype #857 before worrying about UX concerns!

jviotti, 2016-12-07:
> Just throwing this out there: Konami code 😆

lol, or we can add certain keyboard arrow combinations to trigger stuff :P 

> (although it might be useful for this specific example to actually keep the env-var too, so that we can test the dialog-gets-displayed-at-startup feature)

I propose to just start with the DevTools function and re-consider that if there is really a use case for that, since running the developer command when the application starts has the exact same effect really.

> Not sure that it warrants a whole DevTools Extension though ;-)

Not now, definitely, but if we keep adding lots of developer extra stuff, it might be worth considering at some point.

> Might also be a useful place to prototype #857 before worrying about UX concerns!

Oh yeah, that's an excellent idea!

lurch, 2016-12-07:
> > (although it might be useful for this specific example to actually keep the env-var too, so that we can test the dialog-gets-displayed-at-startup feature)

> I propose to just start with the DevTools function and re-consider that if there is really a use case for that, since running the developer command when the application starts has the exact same effect really.

...assuming that some botched PR-merge doesn't mean that the update-check stops getting run at startup. That's why I'm suggesting the env-var might still be useful. So yeah, in hindsight I'm changing the original focus of this PR away from "a way to force the update dialog to be displayed at startup" to instead "a way to force the notification system to think that there's actually an update available" ;-)  Perhaps we could even have the env-var specify the new "fake version" that the notification system reports?

jviotti, 2016-12-07:
Makes sense, lets go for the environment variable as well.

closed 2017-04-27 by jviotti