balena-io/etcher #975: [feature] etcher download widget

alexandrosm, 2016-12-14:
Etcher is commonly used in instructions pages. 

What those pages usually do is ask their users to go to the Etcher page and download Etcher. 

That requires the users to: 
- click to visit the Etcher website
- wait for it to load
- understand the page
- download and install Etcher
- close the Etcher website and switch back to the instructions page

Some few websites, wanting to avoid that roundtrip, hardlink to the installers, which is a terrible idea due to our constant updates being released.

I wonder whether we could create some kind of embeddable widget that acts as a tiny form of the Etcher website. It detects the users' OS, offers them the most recent version for it, and gets out of the way, unless the users want to click through and see more details or view more download types, etc. It would remove quite a few steps from the process, making Etcher even more instruction-page friendly.

Thoughts, especially from the people with instruction pages?

WasabiFan, 2016-12-14:
As someone who maintains such a page, I think this would be helpful; we currently link to etcher.io and then expect users to download the app manually, as you mentioned. It isn't a major issue as we have it right now, but the link is a bit hidden because it's directly above a bunch of Etcher screenshots. A badge-like image or other control would be nice for that.

jviotti, 2016-12-14:
I love this idea, and looks like a perfect task for @Shou. Maybe @konmouz can provide some design input here?

lurch, 2016-12-14:
I guess it'd need to be some kind of mini-popup-window though, because as discussed in another issue, browser OS auto-detection isn't that great.

konmouz, 2016-12-14:
Agree as well, sounds a great idea and can have a serious impact on Etcher downloads as well. Design-wise we need to be careful and make this pop-up clear (make sure it doesn't look like a common ad pop-up that users will dismiss immediately). Also, the fact that it is free and open-source should be highlighted as well.
Do you think it possible to control the initial CTA? That would be even better. But I guess in some instructions the link will be already embedded in the text, which probably means that we can control only the pop-up...

jviotti, 2016-12-14:
What does CTA stand for?

lurch, 2016-12-14:
I guess the ideal scenario would just be to have the user insert a `<script src="https://etcher.io/etcher_install_button.js"></script>` tag where they want the CTA ( = Call To Action, e.g. a button) to appear, and then we could just write HTML into the tag, so we get to control both the CTA, and the action of clicking the CTA.

Can't remember where off the top of my head, but I'm sure I've seen that technique used elsewhere before.

alexandrosm, 2016-12-14:
I don't think we should go for pop-ups, they've completely gone out of
favour these days. I also don't think it should take energy away from v1,
just kinda saying it here for storage. Unless there's enough excitement of
course.

@lurch, browser detection isn't *perfect*, but if there's a "see more
downloads" link that takes you to the regular flow, we should be ok in 99%
of cases.

WasabiFan, 2016-12-14:
Another option is a plain page that could be included in an `iframe` (which would really just be some buttons). Or a badge image and straight link.

konmouz, 2016-12-14:
I thought that we need a pop-up anyway for technical reasons. Having something direct would be ideal, especially since we have some sort of control on the download CTA.

lurch, 2016-12-14:
What about a pop-over instead of a pop-up? :-) ( [this sort of thing](http://www.w3schools.com/bootstrap/bootstrap_popover.asp) )

If it was just a button / image, I don't think there'd be room for the "see more
downloads" text, without making the buttons / image to big, which would probably be a turn-off for the instructions-authors we're targetting this at? (all IMHO of course)

konmouz, 2016-12-14:
That could work. I guess our options are: "Download Latest Etcher", "Check Etcher Versions", "Read more about Etcher". Last two options can be merged in one.

lurch, 2016-12-14:
IMHO it'd be nice if it said e.g. "Download Etcher for Windows x64" (with the OS being auto-detected) with maybe small "Other versions..." and "More info..." links underneath?
I definitely don't think we need the word "Latest" in there, because isn't that kinda implicit?

alexandrosm, 2016-12-14:
I *really* don't want to be too intrusive to the host page, so unless an
iframe doesn't work for some reason, let's stick with that instead of
pop-anythings, unless there's some specific consideration that hasn't come
up yet?

Shou, 2016-12-15:
I believe an iframe would be ideal as it transparently does not muck with the embedding webpage (same-origin policy). A direct script would work too, but it's less clear what exactly it will do (now and in the future) and may make people reluctant to use it. I don't think we need to do anything directly to the embedding website anyway so there shouldn't be any downsides to an iframe.

lurch, 2016-12-15:
I guess the only disadvantage with iframes is that it's the host-page which sets the size of the iframe, so we'd probably want to give a recommended size for the iframe, which then gives us less flexibility to change that size in the future without looking 'odd' on already-written host pages? \*shrug\*

alexandrosm, 2016-12-15:
We can always be reasonably responsive :)

konmouz, 2016-12-19:
Some exploration around the widget idea. @taahirisaacs any ideas? also can you help with the visual polishing?

[image]
[image]