balena-io/etcher #983: Do not allow automatic target selection for a retry operation

saylinglow, 2016-12-20:
- **Etcher version:** 1.0.0-beta.17
- **Operating system and architecture:**macOS Sierra v10.12.2 2.5GHz Intel Core i5 Mac Mini
My Mac Mini (Mid 2011) has a SD card reader, a SATA SSD as the Mac operating system disk and also a SATA drive as a data disk
- **Do you see any meaningful error information on DevTools?**

I was creating a Raspberry Pi boot image on a SD card plugged into the built-in SD card reader of my Mac Mini. The process completed with an error (I think it was something about not being able to unmount the SD card disk) and the 'Retry' button was in red. I clicked 'Retry' but unfortunately, the SD card had been unmounted and I didn't notice a different disk (my SATA data disk) had been selected as the target drive for the image copy. So I've spent the afternoon recovering the data for my SATA disk as Etcher had put the Raspberry Pi image on to my SATA drive.

I would suggest more checks are made in Etcher before allowing a retry operation; do not allow automatic target selection for retry; use the same unix device mount point for a retry, or get the user to actively choose the target disk if the previous target has been unmounted.

jviotti, 2016-12-20:
Ugh, that sucks! Intercepting the auto-selection logic if a previous drive was selected but don't currently exists sounds tricky and would result in unwanted state to be preserved in the application.

I still don't understand why Etcher auto-selected your SATA disk.

- Did you manually pull the SDCard out? An unmount operation shouldn't eject the drive, and it should still remain accessible
- Is your SATA drive external? Can you show us the output of `diskutil info /dev/diskN` where `N` corresponds to your SATA drive? Ideally, Etcher shouldn't be allowing you to write to it in any case, but the OS sometimes detects these external HD as exactly the same thing as an USB drive, and therefore we can't tell them apart without some weak heuristics

jviotti, 2016-12-20:
We should make sure "Retry" is not available if the same drive is no longer accessible though, or can you see a better UX solution @konmouz ?

saylinglow, 2016-12-20:
I didn’t pull out the SDcard.

Fearns% diskutil info /dev/disk1
   Device Identifier:        disk1
   Device Node:              /dev/disk1
   Whole:                    Yes
   Part of Whole:            disk1
   Device / Media Name:      ST2000LM003 HN-M201RAD

   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

   Content (IOContent):      GUID_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 SATA
   SMART Status:             Verified

   Disk Size:                2.0 TB (2000398934016 Bytes) (exactly 3907029168 512-Byte-Units)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          Internal
   Removable Media:          Fixed

   Solid State:              No
   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported
   Device Location:          "Lower"
Fearns% diskutil info /dev/disk1s1
   Device Identifier:        disk1s1
   Device Node:              /dev/disk1s1
   Whole:                    No
   Part of Whole:            disk1

   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

   Partition Type:           EFI
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 SATA
   SMART Status:             Verified
   Volume UUID:              0E239BC6-F960-3107-89CF-1C97F78BB46B
   Disk / Partition UUID:    457CCA5E-39CB-4EFE-BF6D-3E794199D3F7

   Disk Size:                209.7 MB (209715200 Bytes) (exactly 409600 512-Byte-Units)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          Internal
   Removable Media:          Fixed

   Solid State:              No
   Device Location:          "Lower"

Fearns% diskutil info /dev/disk1s2
   Device Identifier:        disk1s2
   Device Node:              /dev/disk1s2
   Whole:                    No
   Part of Whole:            disk1

   Volume Name:              SootyTern
[... 294 lines cut ...]
    1.73 GB   	Cached files
    562 MB    	Swap Used

Diagnostics Information: <etrecheck://help/diagnostics> ⓘ <etrecheck://help/diagnostics>
    Dec 20, 2016, 08:29:13 PM    ~/Library/Logs/DiagnosticReports/Exif Everywhere Plugin Agent_2016-12-20-202913_[redacted].crash
        /Library/Internet Plug-Ins/Exif Everywhere.webplugin/Contents/Exif Everywhere Plugin Agent.app/Contents/MacOS/Exif Everywhere Plugin Agent
    Dec 20, 2016, 01:35:43 AM    /Library/Logs/DiagnosticReports/installd_2016-12-20-013543_[redacted].cpu_resource.diag [Details] <etrecheck://detail//Library/Logs/DiagnosticReports/installd_2016-12-20-013543_%5Bredacted%5D.cpu_resource.diag>
        /System/Library/PrivateFrameworks/PackageKit.framework/Resources/installd
    Dec 20, 2016, 01:24:52 AM    /Library/Logs/DiagnosticReports/storedownloadd_2016-12-20-012452_[redacted].cpu_resource.diag [Details] <etrecheck://detail//Library/Logs/DiagnosticReports/storedownloadd_2016-12-20-012452_%5Bredacted%5D.cpu_resource.diag>
        /System/Library/PrivateFrameworks/CommerceKit.framework/Versions/A/Resources/storedownloadd
    Dec 18, 2016, 11:19:38 PM    /Library/Logs/DiagnosticReports/PenTabletDriver_2016-12-18-231938_[redacted].crash
        com.wacom.PenTabletDriver - /Library/Application Support/Tablet/PenTabletDriver.app/Contents/MacOS/PenTabletDriver
    Dec 18, 2016, 11:18:41 PM    Self test - passed
    Dec 18, 2016, 11:14:53 PM    /Library/Logs/DiagnosticReports/quicklookd_2016-12-18-231453_[redacted].crash
        /System/Library/Frameworks/QuickLook.framework/Versions/A/Resources/quicklookd.app/Contents/MacOS/quicklookd
    Dec 18, 2016, 11:14:53 PM    /Library/Logs/DiagnosticReports/ReportCrash_2016-12-18-231453_[redacted].crash
        /System/Library/CoreServices/ReportCrash
    Dec 18, 2016, 03:35:13 PM    /Library/Logs/DiagnosticReports/mds_2016-12-18-153513_[redacted].crash
        /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/Metadata.framework/Versions/A/Support/mds

best wishes
Sayling

Sayling Low
sayling.low@btinternet.com

saylinglow, 2016-12-20:
I think that that would be good enough. If you make the user go through choosing the target again it’s no longer obviously a retry.

Thanks for your quick response.

Sayling

Sayling Low
sayling.low@btinternet.com

lurch, 2016-12-21:
Sorry to hear about what happened @saylinglow , I hope your data-restoration is going okay.
Can you confirm that you definitely didn't have "Unsafe mode" enabled in the Etcher preferences?
Just for the sake of completion, could you also post the `diskutil` output for your SD card reader?

@jviotti Do you think it'd be worth automatically making drives marked with `Protocol: SATA` into "System drives" ? (if they're not already detected as system drives)

konmouz, 2016-12-21:
@jviotti I agree with you. Maybe also worth displaying the last settings (image + drive) on this screen. But indeed it is a bit weird that retry flashed another disk.

saylinglow, 2016-12-21:
This is with a different SD card inserted. (a Raspberry Pi image that has got corrupted).

Fearns% diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *256.1 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:                  Apple_HFS TawnyOwl                255.2 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3

/dev/disk1 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *2.0 TB     disk1
   1:                        EFI EFI                     209.7 MB   disk1s1
   2:                  Apple_HFS SootyTern               2.0 TB     disk1s2

/dev/disk2 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *7.8 GB     disk2
   1:             Windows_FAT_32 boot                    58.7 MB    disk2s1
   2:                      Linux                         3.9 GB     disk2s2

/dev/disk3 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *2.0 TB     disk3
   1:                        EFI EFI                     209.7 MB   disk3s1
   2:                  Apple_HFS Fearns_tm2              2.0 TB     disk3s2

Fearns% diskutil info /dev/disk2
   Device Identifier:        disk2
   Device Node:              /dev/disk2
   Whole:                    Yes
   Part of Whole:            disk2
   Device / Media Name:      Built In SDXC Reader

   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 Secure Digital
   SMART Status:             Not Supported

   Disk Size:                7.8 GB (7822376960 Bytes) (exactly 15278080 512-Byte-Units)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          Internal
   Removable Media:          Removable
   Media Removal:            Software-Activated

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

Fearns% diskutil info /dev/disk2s1
   Device Identifier:        disk2s1
   Device Node:              /dev/disk2s1
   Whole:                    No
   Part of Whole:            disk2

   Volume Name:              boot
   Mounted:                  Yes
   Mount Point:              /Volumes/boot

   Partition Type:           Windows_FAT_32
   File System Personality:  MS-DOS FAT16
   Type (Bundle):            msdos
   Name (User Visible):      MS-DOS (FAT16)

   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 Secure Digital
   SMART Status:             Not Supported
   Volume UUID:              F1B22FD9-94DC-3F90-B1F2-0F6DBBB5D885

   Disk Size:                58.7 MB (58720256 Bytes) (exactly 114688 512-Byte-Units)
   Device Block Size:        512 Bytes

   Volume Total Space:       58.7 MB (58662912 Bytes) (exactly 114576 512-Byte-Units)
   Volume Used Space:        15.1 MB (15065088 Bytes) (exactly 29424 512-Byte-Units) (25.7%)
   Volume Available Space:   43.6 MB (43597824 Bytes) (exactly 85152 512-Byte-Units) (74.3%)
   Allocation Block Size:    8192 Bytes

   Read-Only Media:          No
   Read-Only Volume:         No

   Device Location:          Internal
   Removable Media:          Removable
   Media Removal:            Software-Activated

Fearns% diskutil info /dev/disk2s2
   Device Identifier:        disk2s2
   Device Node:              /dev/disk2s2
   Whole:                    No
   Part of Whole:            disk2

   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

   Partition Type:           Linux
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 Secure Digital
   SMART Status:             Not Supported

   Disk Size:                3.9 GB (3894411264 Bytes) (exactly 7606272 512-Byte-Units)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          Internal
   Removable Media:          Removable
   Media Removal:            Software-Activated
+++

I've successfully recovered all the data (about 1GB worth, partly from a physical backup from 30nov and also by re-associating the recovered disk with the Time Machine backup)

jviotti, 2016-12-21:
> @jviotti Do you think it'd be worth automatically making drives marked with Protocol: SATA into "System drives" ? (if they're not already detected as system drives)

I think its a good assumption.

In any case, this is getting really weird, since based on the `diskutil info` output, that drive should have been detected as a system drive based on the following attributes:

```
Removable Media:          Fixed
Device Location:          Internal
```

Which directly matches the following condition from `darwin.sh`:

```sh
removable=`echo "$diskinfo" | get_key "Removable Media"`
location=`echo "$diskinfo" | get_key "Device Location"`

[[ ( "$location" == "Internal" ) && ( "$removable" != "Yes" ) && ( "$removable" != "Removable" ) ]]
```

@saylinglow Can you try running the following Bash script on your Mac and report back what you get? https://github.com/resin-io-modules/drivelist/blob/master/scripts/darwin.sh

jviotti, 2016-12-21:
The fact that the removable drive was not available anymore after an unmount is also very weird. @lurch , thoughts?

WasabiFan, 2016-12-21:
@jviotti where is the logic that protects against selecting a system drive? Is it in the selection modal, or immediately before the write procedure (or both)? If it's the former, is it possible that the protection logic wasn't run at all?

jviotti, 2016-12-21:
@WasabiFan It comes from https://github.com/resin-io-modules/drivelist, which returns objects containing the boolean property. That is saved to a Redux store, and everything else reads from there, so this logic not actually running shouldn't be a possibility.

saylinglow, 2016-12-21:
@jviotti 
output from drivelist.sh :--

Fearns% ~/drivelist.sh
device: /dev/disk0
description: "M4-CT256M4SSD2"
size: 256060514304
mountpoints:
  - path: /
raw: /dev/rdisk0
protected: False
system: True

device: /dev/disk1
description: "ST2000LM003 HN-M201RAD"
size: 2000398934016
mountpoints:
  - path: /Volumes/SootyTern
raw: /dev/rdisk1
protected: False
system: False

device: /dev/disk2
description: "Built In SDXC Reader"
size: 7822376960
mountpoints: []
raw: /dev/rdisk2
protected: False
system: False

device: /dev/disk3
description: "JMicron"
size: 2000398934016
mountpoints:
  - path: /Volumes/Fearns_tm2
raw: /dev/rdisk3
protected: False
system: False

device: /dev/disk4
description: "5ASG"
size: 250059350016
mountpoints:
  - path: /Volumes/TawnyOwl_bk
raw: /dev/rdisk4
protected: False
system: False

WasabiFan, 2016-12-21:
@jviotti what I'm trying to figure out is what part of Etcher checks that property. Is there something triggered when you hit the "flash" button that confirms that the selected drive is valid? Or if I managed to select a system drive (possibly through faulty retry logic 😉 ) would Etcher still be willing to write to it?

jviotti, 2016-12-21:
@WasabiFan Oh, I see what you mean. The check is being done here: https://github.com/resin-io/etcher/blob/master/lib/gui/modules/drive-scanner.js#L61. We then attach a listener to the DriveScanner in order to populate the Redux store, from where other components read from, so if it got to the store, then the check was indeed made.

In any case, by looking at @saylinglow output, Drivelist is indeed detecting this as `system: false` for some very strange reason... I'll keep investigating...

jviotti, 2016-12-21:
@saylinglow Can you give this new Bash script a go? I modified `darwin.sh` to include some extra debugging information:

```bash
#!/bin/sh

function get_key {
  grep "$1" | awk -F "  +" '{ print $3 }'
}

function get_until_paren {
  awk 'match($0, "\\(|$"){ print substr($0, 0, RSTART - 1) }'
}

DISKS="`diskutil list | grep '^\/' | get_until_paren`"
mount_output="`mount`"

for disk in $DISKS; do
  diskinfo="`diskutil info $disk`"

  device=`echo "$diskinfo" | get_key "Device Node"`
  echo "# Device Node: $device"

  # See http://superuser.com/q/631592
  raw_device=`echo "$device" | sed "s/disk/rdisk/g"`

  description=`echo "$diskinfo" | get_key "Device / Media Name"`
  echo "# Device / Media Name: $description"

  volume_name=`echo "$diskinfo" | get_key "Volume Name"`
  echo "# Volume Name: $volume_name"

  removable=`echo "$diskinfo" | get_key "Removable Media"`
  echo "# Removable Media: $removable"

  protected=`echo "$diskinfo" | get_key "Read-Only Media"`
  echo "# Read-Only Media: $protected"

  location=`echo "$diskinfo" | get_key "Device Location"`
  echo "# Device Location: $location"

  size=`echo "$diskinfo" | sed 's/Disk Size/Total Size/g' | get_key "Total Size" | perl -n -e'/\((\d+)\sBytes\)/ && print $1'`

  mountpoints=`echo "$mount_output" | perl -n -e'm{^'"${disk}"'(s[0-9]+)? on (.*) \(.*\)$} && print "$2\n"'`

  # Omit mounted DMG images
  if [ "$description" == "Disk Image" ]; then
    continue
  fi

  echo "device: $device"

  # Attempt to use the volume name if applicable,
  # since it provides a much more readable name.
  if [[ $volume_name =~ .*Not\ applicable.* ]]; then
    echo "description: \"$description\""
  else
    echo "description: \"$volume_name\""
  fi

  echo "size: $size"

  if [ -z "$mountpoints" ]; then
    echo "mountpoints: []"
  else
    echo "mountpoints:"
    echo "$mountpoints" | while read mountpoint ; do
      echo "  - path: $mountpoint"
    done
  fi

  echo "raw: $raw_device"

  if [[ "$protected" == "Yes" ]]; then
    echo "protected: True"
  else
    echo "protected: False"
  fi

  if [[ "$device" == "/dev/disk0" ]] || \
     [[ "$removable" == "No" ]] || \
     [[ ( "$location" == "Internal" ) && ( "$removable" != "Yes" ) && ( "$removable" != "Removable" ) ]] || \
     echo "$mountpoints" | grep "^/$"
  then
    echo "system: True"
  else
    echo "system: False"
  fi

  echo ""
done
```

saylinglow, 2016-12-21:
@jviotti 

Here's the output of the modified drivelist script:

Fearns% chmod 755 drivelist2.sh
Fearns% ~/drivelist2.sh
# Device Node: /dev/disk0
# Device / Media Name: M4-CT256M4SSD2
# Volume Name: Not applicable (no file system)
# Removable Media: Fixed
# Read-Only Media: No
# Device Location: Internal
"Upper"
device: /dev/disk0
description: "M4-CT256M4SSD2"
size: 256060514304
mountpoints:
  - path: /
raw: /dev/rdisk0
protected: False
system: True

# Device Node: /dev/disk1
# Device / Media Name: ST2000LM003 HN-M201RAD
# Volume Name: Not applicable (no file system)
# Removable Media: Fixed
# Read-Only Media: No
# Device Location: Internal
"Lower"
device: /dev/disk1
description: "ST2000LM003 HN-M201RAD"
size: 2000398934016
mountpoints:
  - path: /Volumes/SootyTern
raw: /dev/rdisk1
protected: False
system: False

# Device Node: /dev/disk2
# Device / Media Name: Built In SDXC Reader
# Volume Name: Not applicable (no file system)
# Removable Media: Removable
# Read-Only Media: No
# Device Location: Internal
device: /dev/disk2
description: "Built In SDXC Reader"
size: 7822376960
mountpoints: []
raw: /dev/rdisk2
protected: False
system: False

# Device Node: /dev/disk3
# Device / Media Name: JMicron
# Volume Name: Not applicable (no file system)
# Removable Media: Fixed
# Read-Only Media: No
# Device Location: External
device: /dev/disk3
description: "JMicron"
size: 2000398934016
mountpoints:
  - path: /Volumes/Fearns_tm2
raw: /dev/rdisk3
protected: False
system: False

# Device Node: /dev/disk4
# Device / Media Name: 5ASG
# Volume Name: Not applicable (no file system)
# Removable Media: Fixed
# Read-Only Media: No
# Device Location: External
device: /dev/disk4
description: "5ASG"
size: 250059350016
mountpoints:
  - path: /Volumes/TawnyOwl_bk
raw: /dev/rdisk4
protected: False
system: False

Fearns%

jviotti, 2016-12-21:
Awesome, thanks a lot @saylinglow !

I see some strange strings being printed in the "Device Location" key for the first couple of drives:

```
# Device Location: Internal
"Upper"

...

# Device Location: Internal
"Lower"
```

I have no clue where that comes from, but its clearly causing the equality assertions to fail :/

jviotti, 2016-12-21:
@saylinglow What happens if you run the following command?

```sh
diskutil info /dev/disk1 | grep "Device Location" | awk -F "  +" '{ print $3 }'
```

jviotti, 2016-12-21:
Oh, damn, so there are two "Device Location" keys in your output. WTF?

```
   Device Location:          Internal
   ...
   Device Location:          "Lower"
```

jviotti, 2016-12-21:
I guess the best think we can do is join every value with some delimiter, and check that the string we are interested in (e.g: "Internal") is part of that list.

saylinglow, 2016-12-21:
@jviotti 

```
Fearns% diskutil info /dev/disk1 | grep "Device Location" | awk -F "  +" '{ print $3 }'
Internal
"Lower"
Fearns% 

```
I think the upper and lower refer to the two positions within the Mac Mini for an internal SATA hard disk.

Sayling

jviotti, 2016-12-21:
I see. Thanks a lot for your help @saylinglow , at least we got to the bottom of it :D I'll send a fix to Drivelist soon, and we'll keep this issue opened to handle the "retry" UX.

jviotti, 2016-12-21:
Done: https://github.com/resin-io-modules/drivelist/pull/129

@saylinglow Can you confirm the following script works for you now (e.g: reports `system: True` for `/dev/disk1`)?

```bash
#!/bin/sh

function get_key {
  awk -F "  +" "
    /${1/\//\/}/ {
      if (values) {
        values = values \" \" \$3
      } else {
        values = \$3
      }
    }

    END { print values }
  "
}

function get_until_paren {
  awk 'match($0, "\\(|$"){ print substr($0, 0, RSTART - 1) }'
}

DISKS="`diskutil list | grep '^\/' | get_until_paren`"
mount_output="`mount`"

for disk in $DISKS; do
  diskinfo="`diskutil info $disk`"

  device=`echo "$diskinfo" | get_key "Device Node"`

  # See http://superuser.com/q/631592
  raw_device=`echo "$device" | sed "s/disk/rdisk/g"`

  description=`echo "$diskinfo" | get_key "Device / Media Name"`
  volume_name=`echo "$diskinfo" | get_key "Volume Name"`
  removable=`echo "$diskinfo" | get_key "Removable Media"`
  protected=`echo "$diskinfo" | get_key "Read-Only Media"`
  location=`echo "$diskinfo" | get_key "Device Location"`
  size=`echo "$diskinfo" | sed 's/Disk Size/Total Size/g' | get_key "Total Size" | perl -n -e'/\((\d+)\sBytes\)/ && print $1'`

  mountpoints=`echo "$mount_output" | perl -n -e'm{^'"${disk}"'(s[0-9]+)? on (.*) \(.*\)$} && print "$2\n"'`

  # Omit mounted DMG images
  if [ "$description" == "Disk Image" ]; then
    continue
  fi

  echo "device: $device"

  # Attempt to use the volume name if applicable,
  # since it provides a much more readable name.
  if [[ $volume_name =~ .*Not\ applicable.* ]]; then
    echo "description: \"$description\""
  else
    echo "description: \"$volume_name\""
  fi

  echo "size: $size"

  if [ -z "$mountpoints" ]; then
    echo "mountpoints: []"
  else
    echo "mountpoints:"
    echo "$mountpoints" | while read mountpoint ; do
      echo "  - path: $mountpoint"
    done
  fi

  echo "raw: $raw_device"

  if [[ "$protected" == "Yes" ]]; then
    echo "protected: True"
  else
    echo "protected: False"
  fi

  if [[ "$device" == "/dev/disk0" ]] || \
     [[ "$removable" == "No" ]] || \
     [[ ( "$location" =~ "Internal" ) && ( "$removable" != "Yes" ) && ( "$removable" != "Removable" ) ]] || \
     echo "$mountpoints" | grep "^/$"
  then
    echo "system: True"
  else
    echo "system: False"
  fi

  echo ""
done
```

Happy to take comments on better approaches!

saylinglow, 2016-12-21:
@jviotti 

Here's the result from your latest script:

```
Fearns% chmod 755 drivelist3.sh
Fearns% ~/drivelist3.sh
device: /dev/disk0
description: "M4-CT256M4SSD2"
size: 256060514304
mountpoints:
  - path: /
raw: /dev/rdisk0
protected: False
system: True

device: /dev/disk1
description: "ST2000LM003 HN-M201RAD"
size: 2000398934016
mountpoints:
  - path: /Volumes/SootyTern
raw: /dev/rdisk1
protected: False
system: True

device: /dev/disk2
description: "Built In SDXC Reader"
size: 7822376960
mountpoints: []
raw: /dev/rdisk2
protected: False
system: False

device: /dev/disk3
description: "JMicron"
size: 2000398934016
mountpoints:
  - path: /Volumes/Fearns_tm2
raw: /dev/rdisk3
protected: False
system: False

device: /dev/disk4
description: "5ASG"
size: 250059350016
mountpoints:
  - path: /Volumes/TawnyOwl_bk
raw: /dev/rdisk4
protected: False
system: False

Fearns% 

```

Looks like it's recognised the second internal disk as system = True

Sayling

jviotti, 2016-12-21:
Perfect! Hopefully this fix will be included in the next Etcher version, and your SATA wipe shouldn't have happened at all. Sorry again for the inconveniences and thank you very much for your help in solving this issue!

jviotti, 2018-10-15:
Closing as the real fix is to solve the way to detect drives

closed 2018-10-15 by jviotti