balena-io/etcher #984: Show list of supported file extensions in alphabetical ordering

tooomm, 2016-12-22:
- **Etcher version:** Etcher-1.0.0-beta.17-win32-x64

`Open Image` window should have a ordered list of supported formats, preferably in alphabetical order. It's a bit of a mess right now: :)

[image]

lurch, 2016-12-22:
Haha, reminds me of #787 ;-)

For this particular fix though, we'd either need to modify https://github.com/resin-io/etcher/blob/master/lib/gui/models/supported-formats.js#L101 to always sort the extensions alphabetically (or use some other well-defined ordering method), or modify the order the items are stored in https://github.com/resin-io-modules/etcher-image-stream/blob/master/lib/supported.js

jviotti, 2016-12-23:
Let's sort in alphabetical order here and keep our "relevance" based order everywhere else.

tooomm, 2017-01-05:
This can be closed I guess...

closed 2017-01-05 by tooomm

jviotti, 2017-01-05:
Ah, good catch. Turns out the commit referenced this issue with "See" instead of "Closes"/"Fixes".