balena-io/etcher #985: Etcher freezing on "Starting..."

Facel3ss1, 2016-12-23:
- **Etcher version: 1.0.0-beta.17**
- **Operating system and architecture: macOS 10.12.1 64-bit**
- **Do you see any meaningful error information on DevTools?** No

When trying to flash an image to an SD Card, Etcher freezes on the "Starting..." dialog on the flash button: 
[image]
No errors come up in DevTools or the GUI...
[image]
...but if I try running the command in the DevTools in Terminal.app it comes up with this error message:
[image]
Any help would be appreciated. Thanks!

jviotti, 2016-12-24:
Hm, very weird. So the command exits with an error code other than zero, but we don't know anything else about it.

BTW, did you run the command after closing Etcher? If so, the error might be due to the IPC server ID not longer being valid, and that might be obscuring the original issue.

If you can reliably reproduce (let me know if not), you should be able to execute the command while Etcher is frozen, but without closing it, and see if that reveals anything else.

Facel3ss1, 2016-12-24:
Okay, I have a lot to say, so bear with me:

Thank you for your reply (especially this close to Christmas! 🎄)! I did run the command in Etcher while the Terminal was closed. So, I ran it while Etcher was open as you said and that just freezes too (it freezes on 'diskutil' at the top'). No errors:
[image]
As for reproducibility, I know I used Etcher to flash the iso of the [Debian+PIXEL image](https://www.raspberrypi.org/blog/pixel-pc-mac/) to my USB stick and this worked successfully. In this case (where Etcher is freezing), I am trying to flash an img ([LibreELEC for RPi Zero](https://libreelec.tv/downloads/)) to my SD card. The download from LibreELEC came as a img.gz. I have tried flashing the img.gz as it is and unzipped as a img. In both cases Etcher freezes. I have also tried flashing the Debian+PIXEL image to the SD card. It Freezes 😢 ! And if I try to plug in my USB stick from earlier (to reproduce the scenario where it worked before) , macOS doesn't recognise it (probably because it has partitions etc after flashing Debian+PIXEL onto it). This suggests to me there's a problem with flashing to SD cards. My SD card in particular is a micro SD to SD card adapter. The micro SD card inside is a micro SDHC I 8GB class 6 (It doesn't say the brand). Let me know if you need any more info.

alexandrosm, 2016-12-24:
Have you tried, or can you try, to flash a different SD card? Perhaps the
issue is with the card itself? Etcher shouldn't behave this way for sure,
but i wonder if it's to do with all SD cards on your machine or just this
specific one has some special factor that's causing the behavour.

jviotti, 2016-12-24:
Can you show us the exact diskutil error you see?

I guess this could indicate some issues when unmounting the drive. What happens if you manually unmount using `diskutil unmountDisk /dev/diskN`? Does it hang as well?

Regarding recovering your drive, you should be able to reformat using the Disk Utility.app.

Let us know how that goes! Merry Christmas!

jviotti, 2016-12-24:
If manually unmounting works fine, I'll create a custom build with extra debugging information.

Facel3ss1, 2017-01-02:
Hello again, sorry for the long reply (with Christmas & New Year etc). @alexandrosm I have tried using a different SD card and it flashed successfully 😄 !! The SD card in question was a Maplin 8GB micro SD card (I am using the same adapter as before)

Facel3ss1, 2017-01-02:
ygu

Facel3ss1, 2017-01-02:
Sorry about the close and reopen, my computer spazzed :/

Facel3ss1, 2017-01-02:
@jviotti I have tried with the original SD card and it's working again as before 😕 . I know this doesn't help in terms of reproducability. One difference I noted was that the disk it was using was different. Before when it wasn't working is ws showing /dev/disk4 (You can see this in the screenshot above). Now, it is showing my SD card as /dev/disk1. And the diskutil 'error from before wasn't an error. Terminal.app shows the current running process in the title bar, and it had froze on 'diskutil' when it wasn't working before. I thank you for your help anyway! Keep up the good work!

jviotti, 2017-01-03:
Ugh, it sucks to not be able to reliably reproduce. I guess this could have happen if there was a file descriptor opened to the drive which diskutil was unable to close even with the force option we pass.

Do you remember if any application was using the drive in question while you were trying to flash to it?

Facel3ss1, 2017-01-03:
@jviotti I don't think so. I cannot remember using another program apart from Etcher... If I was using another program (which is unlikely), it would be Finder. If you have any other queries, I would be happy to respond. Also, if this behaviour occurs again, then I can record that here if you want me to.

jviotti, 2017-01-04:
The `diskutil` command we run is:

```sh
/usr/sbin/diskutil unmountDisk force /dev/diskN
```

where `N` corresponds to the disk number.

How long did you wait for `diskutil` before killing the command? It might be worth to try manually running that command and see if you can manage to reproduce that way.

closed 2017-03-24 by jviotti