balena-io/etcher #986: " , Unexpected token U in JSON at position 0" - I need help with this error

paulallen8691, 2016-12-23:
- **Etcher version:**   1.0.0-beta.17
- **Operating system and architecture:** Win10 Home
Version 1607
OS build 14393.576
RAM 12G
64 bit
- **Do you see any meaningful error information on DevTools?**
Not Installed

Can anyone shed some light on the cause of the following error while trying to write to an SD card? ...
I've uninstalled and reinstalled the card reader.
I've tried a different card with the same response.
I don't see anything within this group that talks about this error.
A google search seems to indicate that this error is related to Javascript.
Any help is appreciated..

" , Unexpected token U in JSON at position 0"
[image]

WasabiFan, 2016-12-23:
> Do you see any meaningful error information on DevTools? Not Installed

@jviotti and company are probably going to need the output from DevTools to figure this out. There shouldn't be a reason that it is "not installed" -- In Etcher after this error occurs, just press <kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>I</kbd> and the pane should pop up, then click on "console" if it isn't already selected and post the contents from there.

This should be a pretty easy bug to fix as long as the maintainers know the log output (assuming the required information is in there).

jviotti, 2016-12-24:
This is somehow similar to https://github.com/resin-io/etcher/issues/981, where we're receiving standard output from the script as IPC messages. 

Given that the help output from the CLI was shown, it seems that we're having some issues passing the arguments to the child script when spawning it, but I can't see anything obvious at first glance.

I'll investigate further and report back, probably after New Year's.

jviotti, 2016-12-24:
What is the full path of the image you're trying to flash BTW?

paulallen8691, 2016-12-25:
G:\Engineering Reference\Raspberry pi\Raspian images\2016-11-25-raspbian-jessie.img

lurch, 2016-12-26:
Hmmm, I wonder if this might be another problem with filenames being passed to commands unquoted, which might lead to errors when filenames / paths contain spaces.

@paulallen8691 As a test, could you try copying the image you're trying to flash to just `G:\` and see if you get the same error?

jviotti, 2016-12-26:
@lurch 

> Hmmm, I wonder if this might be another problem with filenames being passed to commands unquoted, which might lead to errors when filenames / paths contain spaces.

Yeah, I was thinking about just that. 

@paulallen8691 

You can also open DevTools by pressing `Ctrl+Alt+I`. If you do so and click the "Console" tab, Etcher will print the command it will execute as a child process right after you click flash. Do you mind copying that here?

paulallen8691, 2016-12-29:
Sorry for not getting back sooner - family commitments.
> @paulallen8691 As a test, could you try copying the image you're trying to flash to just G:\ and see if you get the same error?

I tried this with the same results.

DevTools is still eluding me - I tried Ctrl+Alt+I to no avail.
Is_ there a package/app that should/could be installed?

I'm trying all this on an ASUS X550E  laptop.

Thanks for your help to date.

jviotti, 2016-12-30:
> DevTools is still eluding me - I tried Ctrl+Alt+I to no avail.
Is_ there a package/app that should/could be installed?

Hm, that's weird. Just in case, notice its "I" from "India." Are you trying to open DevTools after the error? It might be that Etcher gets so terribly broken that the bindings from the main process don't work anymore (although that shouldn't happen), so you could try opening it as soon as the application starts.

[This is how the binding is being declared](https://github.com/resin-io/etcher/blob/master/lib/gui/etcher.js#L66):

```js
electron.globalShortcut.register('CmdOrCtrl+Alt+I', () => {
  mainWindow.webContents.openDevTools({
    mode: 'undocked'
  });
});
```

paulallen8691, 2016-12-30:
A couple of comments:

> It might be that Etcher gets so terribly broken that the bindings from the main process don't work anymore (although that shouldn't happen), so you could try opening it as soon as the application starts.

Yep that looks like the case... If I let the error pop up before starting DevTools it (DevTools)will not open.. A reboot is required before it will open.
Opening Dev tools prior to the error allows it to run.
The console needs to be open and active prior to the error otherwise DevTools shows just a white window.

Now I have some valid debug info...
I initially noticed a lot of ERR_CONNECTION_REFUSED to api.mixpanel.com
I have an aggressive hosts file which specifically filters
mixpanel.com
api.mixpanel.com
www.mixpanel.com
The error still occurs after editing my hosts file to allow this connection. (As a side note I'm not too enthralled that my use is being tracked).

The debug info:

tracker.js:10 Fri Dec 30 2016 09:36:54 GMT-0500 (Eastern Standard Time) Application start
tracker.js:10 Fri Dec 30 2016 09:36:54 GMT-0500 (Eastern Standard Time) Checking for updates
https://usage.trackjs.com/usage.gif?token=032448bc3d9e4cffb1e9b43d29d6c142&…8b-3f67fc3442c3&application=etcher&x=dc992f1b-2cc1-4555-891b-3b5cd9dc226b& Failed to load resource: net::ERR_CONNECTION_REFUSED
tracker.js:10 Fri Dec 30 2016 09:37:15 GMT-0500 (Eastern Standard Time) Select image ({"size":4371513344,"path":"G:\\2016-11-25-raspbian-jessie.img","logo":false,"bmap":false})
tracker.js:10 Fri Dec 30 2016 09:37:46 GMT-0500 (Eastern Standard Time) Flash ({"image":"G:\\2016-11-25-raspbian-jessie.img","device":"\\\\.\\PHYSICALDRIVE1"})
tracker.js:10 WRITER: Attempting to elevate

tracker.js:10 WRITER: Running: set ELECTRON_RUN_AS_NODE=1 && set IPC_SERVER_ID=etcher-server-9732 && set IPC_CLIENT_ID=etcher-client-9732 && call C:\Program Files (x86)\Etcher\Etcher.exe C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\writer-proxy.js C:\Program Files (x86)\Etcher\resources\app.asar G:\2016-11-25-raspbian-jessie.img --robot --drive \\.\PHYSICALDRIVE1 --no-unmount --check

tracker.js:10 SyntaxError: Invalid message from the writer process
    at Server.ipc.server.on.error (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:101:22)
    at Server.emit (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\event-pubsub\es6.js:64:21)
    at Server.gotData (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\node-ipc\dao\socketServer.js:185:14)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)
    at Pipe.onread (net.js:543:20)a.(anonymous function) @ tracker.js:10(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:13708logException @ C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\modules\analytics.js:157reportException @ C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\modules\error.js:45ImageWriterService.flash.then.catch @ C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\pages\main\controllers\flash.js:82processQueue @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16170(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16186$eval @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17444$digest @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17257(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17483completeOutstandingRequest @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:5955(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:6234d.util.hasFunction.f.(anonymous function) @ tracker.js:3
tracker.js:10 Fri Dec 30 2016 09:38:07 GMT-0500 (Eastern Standard Time) Flash error
tracker.js:10 Fri Dec 30 2016 09:38:11 GMT-0500 (Eastern Standard Time) Restart after failure

jviotti, 2016-12-30:
Hi @paulallen8691 ,

> The error still occurs after editing my hosts file to allow this connection. (As a side note I'm not too enthralled that my use is being tracked).

That should be fine. Notice that you can disable error reporting and analytics from the settings page to prevent any communication to the hosts you're blocking anyway.

So the error seems to be caused by an unhandled error on the IPC routines. I'll be focusing on this very soon, and will probably ping you to give my fix a go with a custom build :) Thanks a lot for the help so far!

paulallen8691, 2016-12-30:
> ...Notice that you can disable error reporting and analytics from the settings page ...

Found it

> ...and will probably ping you to give my fix a go with a custom build :) Thanks a lot for the help so far!

It's my pleasure to help debug this issue.

As another test I purchased an external card reader (rather than relying on the internal reader): Insignia NS-CR25A2 with the same error occuring.

jviotti, 2016-12-31:
@paulallen8691 Can you check if you can reproduce the issue by using the standalone Windows version (that doesn't require installation)? Here's the 64-bit one: https://resin-production-downloads.s3.amazonaws.com/etcher/1.0.0-beta.17/Etcher-1.0.0-beta.17-win32-x64.zip

paulallen8691, 2016-12-31:
Same:

Sat Dec 31 2016 18:36:50 GMT-0500 (Eastern Standard Time) Flash ({"image":"G:\\2016-11-25-raspbian-jessie.img","device":"\\\\.\\PHYSICALDRIVE1"})
tracker.js:10 WRITER: Attempting to elevate

tracker.js:10 WRITER: Running: set ELECTRON_RUN_AS_NODE=1 && set IPC_SERVER_ID=etcher-server-12692 && set IPC_CLIENT_ID=etcher-client-12692 && call D:\Etcher\Etcher.exe D:\Etcher\resources\app.asar\lib\src\child-writer\writer-proxy.js D:\Etcher\resources\app.asar G:\2016-11-25-raspbian-jessie.img --robot --drive \\.\PHYSICALDRIVE1 --no-unmount --check

tracker.js:10 SyntaxError: Invalid message from the writer process
    at Server.ipc.server.on.error (D:\Etcher\resources\app.asar\lib\src\child-writer\index.js:101)
    at Server.emit (D:\Etcher\resources\app.asar\node_modules\event-pubsub\es6.js:64)
    at Server.gotData (D:\Etcher\resources\app.asar\node_modules\node-ipc\dao\socketServer.js:185)
    at emitOne (events.js:96)
    at Socket.emit (events.js:188)
    at readableAddChunk (_stream_readable.js:176)
    at Socket.Readable.push (_stream_readable.js:134)
    at Pipe.onread (net.js:543)a.(anonymous function) @ tracker.js:10(anonymous function) @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:13708logException @ D:\Etcher\resources\app.asar\lib\gui\modules\analytics.js:157reportException @ D:\Etcher\resources\app.asar\lib\gui\modules\error.js:45ImageWriterService.flash.then.catch @ D:\Etcher\resources\app.asar\lib\gui\pages\main\controllers\flash.js:82processQueue @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:16170(anonymous function) @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:16186$eval @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:17444$digest @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:17257(anonymous function) @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:17483completeOutstandingRequest @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:5955(anonymous function) @ D:\Etcher\resources\app.asar\node_modules\angular\angular.js:6234d.util.hasFunction.f.(anonymous function) @ tracker.js:3
tracker.js:10 Sat Dec 31 2016 18:37:06 GMT-0500 (Eastern Standard Time) Flash error
tracker.js:10 Sat Dec 31 2016 18:37:09 GMT-0500 (Eastern Standard Time) Restart after failure

jviotti, 2017-01-09:
I've been investigating this issue without much success. The help menu from the CLI will be shown in the following scenarios:

- No image has been passed (in this case a "Missing image" error gets printed at the bottom, which is not the case here)
- The image doesn't exist (prints a "No such file or directory at the bottom")
- The drive is missing but robot is set (in this case it prints a "Missing drive" message)
- The `-h`/`--help` options are passed
- An exception is thrown from the CLI (in which case a message would be printed at the bottom indicating the message)

Everything except from `-h` goes through the same error handling function, so I'll increase the logging there when the CLI is ran from the GUI to see if we can find anything else about it.

jviotti, 2017-01-09:
Okay, I found the reason of the help message: if something fails when parsing the arguments, we print help, *and then* we print the actual message that described the error, however since they are printed in two separate `console.error` calls, they arrive to the GUI via IPC in order, so we first try to parse the first print (which is only the help), which fails, so we abort everything and never consider the next line, which contains an explanation of the error.

I'm solving this whole mess now: the CLI will emit JSON "error" objects that the GUI can understand even on argument parsing errors, which will reveal the actual cause of the error with a nice error message and a stack trace.

jviotti, 2017-01-09:
I'm going to do some cross-platform testing to ensure everything works property (since I'm messing with platform dependent new line separators in weird ways), and I'll publish a custom build so we can finally see the root cause of this issue.

jviotti, 2017-01-11:
Here's the custom build: https://drive.google.com/open?id=0B7tkbonGU-RyZ3h5UjB2NTc1dlE

Please run it and let us know the error message you get! Luckily this should provide all the information we need to actually fix the bug!

jviotti, 2017-01-26:
I'm closing this since I believe its fixed in v1.0.0-beta.18, but please re-open if not!

closed 2017-01-26 by jviotti