balena-io/etcher #987: Make website visits coming from the update prompt on etcher identifiable

alexandrosm, 2016-12-25:
This should be simple to do but v. valuable. We should make the link to the etcher.io website coming from the "update" link be identifiable, perhaps by having a ?ref=etcher_update querystring. That will allow us to separate that traffic in the analytics, and perhaps even offer a slightly different landing page.

lurch, 2016-12-26:
I guess we could also do something similar for the Etcher logo displayed at the bottom of the main program window, e.g. have it link to `https://etcher.io/?ref=etcher`.
And would we then want to use a link like `https://etcher.io/?ref=github` for links in the README and documentation?

jviotti, 2016-12-26:
Sounds good, and its a really easy improvement. I assigned @Shou to it.

alexandrosm, 2016-12-26:
Check the source code :) but the github one, yes!

jviotti, 2016-12-26:
What do you mean? Editing https://github.com/resin-io/etcher/blob/master/lib/gui/components/update-notifier/templates/update-notifier-modal.tpl.html#L15 should do it.

closed 2016-12-30 by jviotti

lurch, 2016-12-30:
@alexandrosm 

> but the github one, yes!

Does that mean that you **do** still want the links from github.com to etcher.io to have referrer query-URLs? And if so, should this issue be re-opened?

alexandrosm, 2017-01-16:
actually, I think github has proper referrer, because we can see it in our
analytics. so no real problem there!