balena-io/etcher #989: Corrupted Flash Drive

Lefel95, 2016-12-28:
- **Etcher version: 1.0.0-beta.17**
- **Operating system and architecture: Windows 10 64bits**
- **Do you see any meaningful error information on DevTools? yes**

I used Etcher to burn a Linux OS in a flash drive and it worked, but when I tried to burn a Linux Mint in the flash Drive it says its corrupted.

Here are the Images that shows the errors:
[image]
[image]
[image]

jviotti, 2016-12-29:
Hi @Lefel95 ,

Thanks for reporting! Can you show us where Etcher says your drive is corrupted? Regarding the error that you show there, it seems that for some reason `diskpart` was unable to clean your drive.

Is this something you can actively reproduce only with a certain image? If so, can you link us directly to the image so we can try it ourselves?

If you can actively reproduce, can you try the following right after you get the error shown above?

- Open `cmd.exe`
- Execute `diskpart.exe`
- Execute `list disk` to show the available drives. Take note of the id of the removable drive
- Execute `select disk N`, where `N` is the id of the removable drive obtained above (make sure you select the right one, otherwise you might wipe out another drive unintentionally)
- Execute `clean`

The last command hopefully reveals more information about the issue.

MetaFight, 2017-01-21:
I get the same error.  In my case, when I try to use `diskpart` to `clean` or `format` my disk I get an error telling me to check the Event Log.

The event log shows an error for source `VDS Basic Provider` with the message:

> Cannot zero sectors on disk \\?\PhysicalDrive1. Error code: 5AA@0101000F

I'm fairly certain my SD card is damaged and this has nothing to do with Etcher.

jviotti, 2017-01-23:
Hi @MetaFight ,

Thanks a lot for the report. It might indeed be something related to faulty SDCards, since no-one from the Etcher can reproduce yet. I see you're in Oxford, and since we have an office in London, would you be willing to "donate" your faulty SD Card to the Etcher team for further investigation? We can have a Weengs guy pick it up from any address you can provide to us! Please e-mail me at juan@resin.io so!

Ping @apostolism ^

LittleDan9, 2017-01-23:
I am experiencing the same issue but only in Windows. I am using the SanDisk microSDHC UHS-I 32GB SDSQUSC-032G-ANCIA. What I found interesting it that the same card in the same card reader was able to be successfully written and verified in Ubuntu through a VM in that same Windows machine that was producing the issue. A major concern is that after receiving this error in Windows etcher did brick two cards. I seriously think something with Windows attempting to mount the disk after the initial format of the card may be part of the problem.

jviotti, 2017-01-23:
Hm, so it might really be Windows and not the SDCard. Are you using an internal SDCard reader, or an external USB one? Do you see any different if you try the internal vs external?

> A major concern is that after receiving this error in Windows etcher did brick two cards.

I believe you should be able to recover these by using the `diskutil` instructions I sent above ^

MetaFight, 2017-01-23:
I'm using an internal reader. And, in my case, windows was happy to read the SD card, but not my raspberry pi.

Also, in windows, some partitioning tools refused to work with the card while others didn't complain. Also, occasionally the card seemed bricked after a failed format, but using the official SD formatting tool always revived it.

Needless to say, I won't be relying on this SD card for critical work.

MetaFight, 2017-01-23:
Also, I'm happy to donate my card. I've already replied by email.

jviotti, 2017-01-23:
Hi @MetaFight ,

We got your e-mail. Thanks a lot for the good willing. We'll hopefully arrange something very soon :) If we can reproduce the issue with that SD Card, that would be a huge step forward with this issue.

LittleDan9, 2017-01-23:
Both disks were new and they were able to read and write a test file after using the SD Formatter. Once using Etcher in Windows 10 and receiving the error the disks became non-functioning. I had attempted to use diskpart to repair, but the event log shows the following: `The IO operation at logical block address 0x0 for Disk 4 (PDO name: \Device\000001be) failed due to a hardware error.` Each card that failed provided a different address and or device location. I also attempted to recover these disks using Linux also without success. As well as using the SD Formatter Tool. When attempting to write to them I would be presented with a write protected error message, which they were not.

> Are you using an internal SDCard reader, or an external USB one? Do you see any different if you try the internal vs external?

I have tried an external (Insignia NS-CR2021 USB 2.0 SD/MMC Memory Card Reader). Additionally, I have tried an internal reader, but I only attempted to use that one on the bricked cards in an attempt to determine if it was my USB bus or something else. That being said, as I mentioned above the success that did have with Etcher was using VM Workstation running Ubuntu whose Host was the Windows 10 machine that is producing the errors. The SD Reader used on success was the external Insignia.

> using the official SD formatting tool always revived it.

I wish my result was the same. The SD formatting tool attempts to work, but without any success and producing the write protected message or simply becoming non-responsive and crashing. As I stated previously, the cards were operational and new prior to the attempt to "etch" them, so I definitely think the bricking is somehow related. I also think they are still within the return / exchange period from the store I bought them from, so I'll be exchanging them.

lurch, 2017-01-23:
I once had a cheap SD card-reader that would always report cards as read-only, when they weren't. I carefully took it apart and found that the write-protect detection switch (a little metal lever) had got bent into the 'always locked' position. Once I carefully bent it back to where it _should_ be, it worked fine afterwards.

*Really* interesting that the hardware that doesn't work natively in Windows, **does** work when used from within a Linux VM on the same computer. Makes me wonder if perhaps the driver Windows is using doesn't support the low-level functions that Etcher is using (but it works within the VM, because the VM is probably talking to it at the USB-level rather than the Windows-driver level?).
Could you find up the manufacturer website for your SD card reader and maybe see if there's an updated driver available?

LittleDan9, 2017-01-23:
> I once had a cheap SD card-reader that would always report cards as read-only, when they weren't. I carefully took it apart and found that the write-protect detection switch (a little metal lever) [...]

It doesn't always report this. It only reported this after the Etcher write failed. It is not as if every card I put in it is write protected. Additionally, the same card worked on that reader previous to the Etcher fail.

> Could you find up the manufacturer website for your SD card reader and maybe see if there's an updated driver available?

For this device in particular, Windows is simply using the Generic STORAGE DEVICE USB Device drivers that are baked into Windows. Insignia does not offer any special drivers for this device. Beyond that, I don't know that I will be using Etcher in Windows for the time being as I don't want any more bricked SD cards.

Let me know if there is anything additional I can provide to aid in diagnosis of the problem.

lurch, 2017-01-23:
On the cards that are "bricked", are you able to run the commands from [here](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#gnulinux) from within the Linux VM to 'clean' them?
Alternatively, do you have e.g. a digital camera that you could use to try formatting the cards? (I've seen that mentioned elsewhere as a "last ditch" attempt which sometimes works when nothing else does)

LittleDan9, 2017-01-23:
> On the cards that are "bricked", are you able to run the commands from here from within the Linux VM to 'clean' them?

`Disk 1:
1+0 records in
1+0 records out
512 bytes copied, 0.0100766 s, 50.8 kB/s
`
`Disk 2:
1+0 records in
1+0 records out
512 bytes copied, 4.44272 s, 0.1 kB/s
`
dd isn't reporting any errors, but it as you can see it is not zeroing out the disk.

Disk 1 appears to have a partition table becuase in GParted it is listing a fat32 file system with 16Mib Used and 28.77 Gib Unused. This may because I had already attempted to recover this disk after the Etch initially failed. Additionally GParted is reporting the following warning: 
`plain floppy: device "/dev/sdb" busy (Resource temporarily unavailable):
Cannot initialize '::'
mlabel: Cannot initialize drive`

Disk 2 does not have a partition table or perhaps a damaged partition table. Additionally, I am unable to write any partition table to it. The warning message in GParted for this disk is:
`Unable to detect file system! Possible reasons are:
~ The file system is damaged
~ The file system is unknown to GParted
~ There is no file system available (unformatted)
~ The device entry /dev/sdb1 is missing
`

lurch, 2017-01-24:
> but it as you can see it is not zeroing out the disk.

What makes you think that? All that that `dd` command does is zero out the first 512 bytes of the disk, which is where the MBR is stored, and after that everything else should see the disk as "unformatted".
But it obviously shouldn't take 4 seconds to write only 512 bytes, so something weird is definitely going on :-/

AFAIK there's nothing that can be done to an SD card - from the software side - that can 'brick' it, so I've no idea how your SD cards have got into this state. Were they purchased from a reputable store, or is it possible that they're fakes?

Something else that it might be worth trying, is flashing an Ubuntu ISO onto a USB drive, getting your computer to live-boot from that, and then try running GParted again, in case running Linux on the hardware directly gives you different results to running Linux inside a VM?

LittleDan9, 2017-01-24:
My bad, didn't look to hard at the dd command, just copied and pasted it this morning half asleep. For some reason I thought it was zeroing the whole 32GB. 
>Were they purchased from a reputable store, or is it possible that they're fakes?
They were purchased form a reputable source, not to worried about, still have 10 days left to exchange/return them. I bought more than two, and those two happen to be the only two to display symptoms and where the only ones I chose to "burn" using Etcher in Windows. All the other disks worked perfect in the Ubuntu VM with Etcher.

I have a laptop with Debian on it, I'll see if there is GParted provide better result indirectly with Hardware. I'll need to charge it and blow the dust off it, I'll report back in tomorrow.

LittleDan9, 2017-01-24:
Unfortunately, after posting the above, I double checked my receipts. I needed to do the exchange last night, so I was unable to test it at the hardware level. I think the evidence is inconclusive that Etcher was the cause for the bricking of the SD Cards. It may have just been two bad cards that Etcher in Windows happen to be the first application to test their limitations. I'll be imaging the replacement cards in the coming week, however, don't plan to use Windows to perform the operations.

lurch, 2017-01-24:
You could always try using h2testw or F3 on your new cards to verify that they're also not bad / dodgy / fakes.

LittleDan9, 2017-01-24:
> You could always try using h2testw or F3 on your new cards to verify that they're also not bad / dodgy / fakes.

@lurch Thanks for the advice, I'll definitely do it that. 

I did have a card that has been used, same model 32GB SanDisk, that needed wiped and reimaged. I attempted this with the Windows Version and continued to receive the timeout error. This card however was not damaged as a result of the timeout. I was able to reformat using the SD Formatter and image it using Etcher in the VM on Ubuntu. So I think there is definitely a bug in Etcher for Windows. I still do not see enough evidence to say that Etcher was or was not the cause for dead cards but I think I'll stick to using Linux to use etcher. Plus, I have yet to have a successful write from Etcher in Windows.

DavidPetit, 2017-02-28:
I am using windows 10 64bits and I also get the same errors systematically.
1) I format the card
2) I try to write an image using etcher
3) I get those errors. And then it corrupts the card.

After this, I format the card again and it can be used again. It happenrd to be with 2 new cards. 

I finally used win32 disk imager and it worked great on both cards. I don't know what cause this bug but it is not because of the card because, I can always format again to get back the card working and then write the image with another software with success. There has to be a bug in Etcher IMO.

lurch, 2017-02-28:
@DavidPetit It's great that you can reproduce this - would you be willing to assist us with further debugging please? :-)  AFAIK no-one on the Etcher team has been able to reproduce this yet.

> It happened to be with 2 new cards.

Are these cards the same manufacturer / brand / capacity? Can you provide more details please? Do you have any other SD cards from a different manufacturer? (so we can see if this is specific to some SD cards)

Is your SD card reader internal or external? (again, the more details you can provide about brand or model, the better!) Do you have a different SD card reader that you can use / borrow? (so we can see if this is specific to some SD readers)

Out of curiosity, which image is it that you're writing?
You say that when you format the card it can be used again - is that with the official SD Formatter tool, or the built-in Windows disk formatter? Do they both work equally well?

If you format the card to "get it working" and then - without running Etcher or Win32DiskImager - you run the commands from [here](https://github.com/resin-io/etcher/issues/989#issuecomment-269580783) and let us know the results, that'd be really useful thank you!

DavidPetit, 2017-02-28:
@lurch 
I am glad to help. :)

I  reproduced the bug and took a screenshot with the feedback error popup after Etcher failed:
[image]. 

After this, if you click on the disk in the explorer it will say "Please insert a disk into xxx (E:)". And if I look at the properties from Windows it reports 0 bytes for all the values.

To get back the SD card working, I need to use the official SD formatter tool (I also tried to format through my Android phone it worked too). However, using windows formatter does not work, it will say "Windows was unable to complete the format" (The windows formatter cannot read the capacity and is stating "Unknown").

The 2 cards are micro SD cards, 16G Toshiba Exceria (class 1) and 8G PNY (class 3). I am using a Micro SD to SD adaptor to plug them into the reader in my laptop. My laptop is Asus Zenbook Pro UX501VW: https://www.asus.com/Notebooks/ASUS-ZenBook-Pro-UX501VW/

The 2 images I tried are Raspbian and Picroft.

That's all. I hope yoy guys will be able to solve this.

Cheers.

DavidPetit, 2017-02-28:
As for the last question with diskpart.exe commands, I just tried and it says:
Diskpart has encoutered an error: Insufficient system ressources exist to complete the requested service. See the system event log for more information.

So I checked the event logs this is the corresponding error: 
"Cannot zero sectors on disk \\?\PhysicalDrive2. Error code: 5AA@0101000F"

Also this same error happened for each time Etcher tried to write. The time hour matches when I launched the app and also you can see the 5 attempts corresponding to 5 times this error for a particular time when it was running.

jviotti, 2017-02-28:
Thanks a lot for the help @DavidPetit .

> Diskpart has encoutered an error: Insufficient system resources exist to complete the requested service. See the system event log for more information.

> "Cannot zero sectors on disk \?\PhysicalDrive2. Error code: 5AA@0101000F"'

This is very valuable information. Do you mind triggering the issue again, while keeping an eye on the task manager, to see how much remaining RAM you have when the problem occurs?

What are the specs of your computer BTW? (architecture, RAM, etc)

jviotti, 2017-02-28:
Also, @DavidPetit does this happen as soon as you click flash, or once the flash completed?

lurch, 2017-02-28:
Ditto, thanks for the info @DavidPetit :+1: 
With regards to why Win32DiskImager works and Etcher doesn't - I believe Win32Diskmager does a lower-level drive-locking than Etcher does, and hence it doesn't need to use the `diskpart` command (and as you've seen, Etcher actually tries to run diskpart 5 times before giving up and displaying an error message to the user).

@jviotti https://msdn.microsoft.com/en-us/library/windows/desktop/ms681385(v=vs.85).aspx also confirms error code 0x5AA (which is the same error code that @MetaFight had) as:
**ERROR_NO_SYSTEM_RESOURCES** 'Insufficient system resources exist to complete the requested service.'
But it's hard to imagine that a simple `diskpart` command really needs that many resources?!? :-/

https://duckduckgo.com/?q="Cannot+zero+sectors+on+disk" isn't very helpful either - seems to mainly be people with genuinely broken USB drives (which obviously isn't the case here, as SD Formatter is able to fix things).

@DavidPetit You mentioned that on a corrupted drive, the Windows disk-formatting doesn't work, but SDFormatter does. Just out of curiosity, are you able to run the Windows disk-formatting successfully _after_ formatting the card with SDFormatter?

lurch, 2017-02-28:
@DavidPetit A couple more ideas I've just thought of (if you're still willing to provide more help)... Do you have a spare USB drive (rather than SD card) that you're able to test Etcher with? Are you able to reboot your laptop into safe-mode, to see if Etcher / diskpart behaves any more reliably on your SD cards then?

DavidPetit, 2017-03-01:
> What are the specs of your computer BTW? (architecture, RAM, etc)

CPU: Intel Core i7 6700 HQ @ 2.60GHZ
RAM: 16GB of RAM DDR4 2133 MHz
GPU: GeForce GTX 960M 
SSD: 128GB PCIe x4
HDD: 1TB SATA3
Windows 10 64 bits

> Do you mind triggering the issue again, while keeping an eye on the task manager, to see how much remaining RAM you have when the problem occurs?

Just did and I can confirm the consumption of RAM is not increasing and remains at a stable 50% during Etcher attempts to flash the card. At the same time, the CPU load is very low too (not more than 12-13% during the process).
The error in the event log is still the same one: "Cannot zero sectors on disk \\?\PhysicalDrive2. Error code: 5AA@0101000F" and the popup the same. This bug is repeatable and the error is consistent and always the same.

> Just out of curiosity, are you able to run the Windows disk-formatting successfully after formatting the card with SDFormatter?

Yes. Just did it and it works.

> Do you have a spare USB drive (rather than SD card) that you're able to test Etcher with? 

Flashing a USB drive with Etcher works fine.

> Are you able to reboot your laptop into safe-mode, to see if Etcher / diskpart behaves any more reliably on your SD cards then?

I rebooted in safe mode and tried. But in this mode, the SD card does not even show up when I insert it in the reader (whereas USB drive shows up). So I could not try to do anything with a SD card in safe mode... And after rebooting in "normal"mode, the card appears when inserting in the reader as usual.

lurch, 2017-03-01:
Curioser and curioser... I can't think what else to suggest :-/ Any ideas @jviotti ?
Given that Etcher is able to flash a regular USB drive okay, I _suspect_ that a USB-connected SD reader might also work (i.e. perhaps this is only a bug that affects internal SD slots on Windows computers?). But obviously that doesn't get us any closer to 'properly' fixing the problem in Etcher :(

I don't suppose your internal SD slot has any kind of custom Windows driver? Could you try to see if maybe there's an updated driver available? (although with your laptop being so new, I guess that's unlikely)
Actually, getting the properties of your internal SD reader in Windows Device Manager might be interesting...

Thanks again for your help so far!

DavidPetit, 2017-03-01:
The card reader description: Alcor Micro PCIE Card Reader

I was able to update the driver but it does not help for Etcher. Still same bug.

lurch, 2017-03-01:
Hmmm, I wonder if this is some kind of inherent 'incompatibility' between PCI-connected card-readers on Windows, and the `diskpart` command? :-/  Perhaps this is just limited to Windows 10?
It would explain why this problem is 100% reproducible, but still rare, because I suspect most people are using USB-connected SD card-readers. Even on some laptops, the internal SD slot is still internally connected over USB.

@DavidPetit One more thing that you could try for me, if you have the technical know-how - could you flash an Ubuntu ISO onto your USB drive, temporarily boot your laptop from that (in LiveLinux mode), and then see if Etcher works correctly with your internal SD reader under Linux? (which would allow us to identify if this is an OS-problem or a hardware-problem)

lurch, 2017-03-01:
Update: I've now asked one of the other resin.io employees to see if they can reproduce this on their NUC (which has a PCI-connected SD card slot) running Windows 10, and he reports that Etcher and `diskpart` both work fine, so this clearly doesn't affect _all_ PCI-connected SD readers. Hmmm... :-S

DavidPetit, 2017-03-01:
> could you flash an Ubuntu ISO onto your USB drive, temporarily boot your laptop from that (in LiveLinux mode), and then see if Etcher works correctly with your internal SD reader under Linux?

I tried but unfortunately Ubuntu does not see the SD card... It can only see the USB devices and the SATA SSD and HDD. I probably need to install some driver but I already spent a lot of time just to make Ubuntu boot (it crashed multiple times before I could open a session) so I will just stop here.

Cheers for the rest!

lurch, 2017-03-01:
Hmm, so maybe this _is_ a problem just with particular non-standard PCI-connected internal SD readers? As you can tell I'm grasping at straws ;-)

Thanks again for all your help @DavidPetit it's been **incredibly** useful :smiley:

jviotti, 2017-03-01:
Amazing stuff going on here, but sadly I'm out of ideas as well. @lurch Do we know the laptop model? Maybe worth trying to get access a laptop with a Alcor Micro PCIE Card Reader?

lurch, 2017-03-01:
> Do we know the laptop model?

https://github.com/resin-io/etcher/issues/989#issuecomment-283041594 says `My laptop is Asus Zenbook Pro UX501VW` ;-)

jviotti, 2017-03-01:
Ah, great catch! The laptop is quite good looking btw :) 

We should try to get access to one of those somehow. We can ask the resin.io team, however I feel that we should have some kind of process for this. We've been encountering tons of issues that can only be reproduced in certain computers, so I wonder if there is a "service" we can use to rent or get access for some time to certain computer brands (startup idea? :P).

@lurch Any ideas?

lurch, 2017-03-01:
> so I wonder if there is a "service" we can use to rent or get access for some time to certain computer brands (startup idea? :P).
@lurch Any ideas?

Nope, no ideas. I _suspect_ that there's such a huge variety of different laptops, that any "rent this specific hardware for a short time-period" service would never recoup the hardware costs.

Ringworks, 2017-03-01:
Etcher has bricked my brand new 32gb SanDisk card.  After the image was verified and the drive was OK to remove I took at the card out and put it right back in to see how the image looked on the drive. My system no longer recognizes the SD Card.  The computer has a built in SD reader.  I tried the card on my laptop and in several lab systems with the same result.  

The system is running Windows 10 Pro.
Device Manager does not show any issues with the hardware.
I can use other SD cards without issue.

jviotti, 2017-03-01:
@Ringworks Most frequently the issue is that Windows can't read the partitions of the image you flashed to the drive (e.g: Linux partitions). Are you sure that's not the case?

Ringworks, 2017-03-01:
Update: I have been in support for 25 years, so I was able to do a little ground work and got this resolved.  Thanks for the comment jviotti, that was very good thought but in this case it was not the issue.

I have to take a step back, I was able to test my system USB ports with another USB stick and it worked properly, it was detected and I could access it.  This got me thinking that that maybe the USB and SD drivers are not totally shared.

On a hunch I uninstalled the USB drivers and rebooted the system letting them reinstall. One thing to note is that they where not marked as being 'bad' no yellow or red indicators in device manager.  After the reboot they re-detected and reinstalled I put in the SD card and it worked!

The timing of this is totally suspect. I have been using other SD card utilities (imagers and formaters) with out any issues all week I use Etcher and this problem occurs.  I have a feeling that the "auto eject" might not be working as intended because the formatting worked without a hitch

jviotti, 2017-03-02:
Hi @Ringworks ,

It looks like you're hitting this one: https://github.com/resin-io/etcher/issues/750. The eject Windows functionality seem to completely eject the internal reader, so you have to reboot to make Windows detect it. We hopefully have a way forward though :)

Ringworks, 2017-03-02:
The thing is I rebooted my box twice in the hopes that it was something along those lines.  It did not correct the issue, he SD card was not detected and at that point I even got another SD card and it was not recognized.  But we know what this issue is.  I would highly recommend disabling the Auto Eject in the software and just pop a message just a thought.

jviotti, 2017-03-03:
Hm, that sucks. I wonder what caused this. We're looking forward to re-implement the unmounting mechanisms ourselves in order to be able to debug these kinds of issues further (we're blocked by a black box at the moment).

@jhermsmeier Did you have a chance to look at the unmounting repo?

Ringworks, 2017-03-03:
I was able to reproduce the same behavior.  The SD card is in fact fine, it's the SD reader that gets broken after the 'auto eject' (Windows users don't normally think in terms for mounting an unmounting disks/SD/cards/volumes).   Running the "Troubleshooter" identified an issue with the USB interface and automatically fixed the problem, that is a an easier fix then the route I took the first time. 

To fix the broken SD card slot do the following:
1. Go to: Control Panel\All Control Panel Items\Troubleshooting\Hardware and Sound
2. Select Hardware and Devices
3. After the systems hardware is scanned it should detect the issue with the SD slot.   It will remove the drivers and re-add them.  
4. Reboot

The SD slot will now work again.

I will just uncheck the auto eject for now and look for a new version at some point in the future.

jviotti, 2017-03-03:
Great finds @Ringworks. We'll hopefully fix this soon, so stay tuned!

jhermsmeier, 2017-03-06:
> @jhermsmeier Did you have a chance to look at the unmounting repo?

@jviotti yeah, fiddled with it a bit, and started reading through the win32 API docs, but haven't gotten around to anything substantial yet.

> I tried the card on my laptop and in several lab systems with the same result.

@Ringworks where there any systems you tested where the card reader was external?

Ringworks, 2017-03-06:
I have not tested on an external card reader

Taidaan, 2017-05-04:
@jviotti - My apologies for resurrecting comments from 2 months ago, but I can confirm the issues @DavidPetit was having with his Asus ZenBook U501VW. I have the Asus ROG G501V, which is identical to his machine with respect to guts and specs. Specifically, my card reader shows as an "Alcor Micro PCIE Card Reader" as well.

While attempting to use Etcher on a Micro SD card, I kept getting errors when it tried to clean the disk. Event viewer shows: "Cannot zero sectors on disk \?\PhysicalDrive2. Error code: 5AA@0101000F".  However, Etcher works just fine when using the same micro SD on the same laptop through a near-ancient USB-attached IDE/SATA/CF/MS/XD/SD reader.

The evidence points to an issue with that specific Alcor card reader. If it's worth your time to track down this problem, let me know and I'll provide my time and laptop to assist however I can.

lurch, 2017-05-04:
@Taidaan thanks for your offer of help :-)

Can you confirm that Win32DiskImager does work with your internal card reader? Could you try the steps I mentioned in [this comment](https://github.com/resin-io/etcher/issues/989#issuecomment-283314724) ?

jhermsmeier, 2017-09-15:
Closing this, as drivelist utilizes the Win32 API for quite a while now.

closed 2017-09-15 by jhermsmeier

nilay994, 2017-11-30:
Is this issue open/active? Facing a problem currently:  <br/>
1) Cheap SD card reader <br/>
2) SD: SanDisk Ultra 32GB MicroSD HCI <br/>
3) Raspbian latest image <br/>
Using Etcher on Windows 7 to burn the image <br/>

lurch, 2017-12-01:
**This** issue is closed `jhermsmeier closed this on 15 Sep`, but if you're having problems please open a new issue, including as much detail as possible, and we'll do our best to help. Thanks.