balena-io/etcher #993: Add an information tooltip for the CRC32 checksum

jviotti, 2016-12-30:
From http://www.linuxlinks.com/article/20161223100458799/Etcher.html

> The user is presented with a CRC32 checksum for the image, but there is no explanation what this means or the purpose of the checksum. A little bit of research reveals the CRC is an acronym for Cyclic Redundancy Code, and the number 32 specifies the size of the hash value - 32 bits. The CRC is an error-detecting code.

Maybe we can add an "i" tooltip explaining how to manually calculate the CRC32 checksum of the image and validate it yourself.

jviotti, 2016-12-30:
Pinging @alexandrosm , @lurch , @Shou , and @konmouz for input on what/how to explain.

konmouz, 2016-12-30:
Or you could click the 'CRC32 CHECKSUM' itself to get more info (the way you click 'others' under 'select image'). @jviotti can you estimate the size of this explanation? Can we fit it in a tooltip or we need more space? Is it possible to have a tooltip for a basic description and a link to redirect to an online resource?

lurch, 2016-12-30:
This is obviously also related to #643 
Given that the usage of MD5 for checksums is _much_ more common than CRC32, maybe if we switched from CRC32 to MD5 (TBH I've got no idea why CRC32 was chosen originally?), then maybe it'd be more obvious and not actually need any explanation?

jviotti, 2016-12-30:
> @jviotti can you estimate the size of this explanation? Can we fit it in a tooltip or we need more space? Is it possible to have a tooltip for a basic description and a link to redirect to an online resource?

It depends on how we want to do it. We can either:

- Show a very small tooltip that describes what this is for in a few words (a short sentence)
- Show a modal with an extended explanation (I estimate about 2 or 3 sentences at most)
- Redirect to an article online (could be part of the user documentation)

I wonder if we should redirect somewhere else though... I have the feeling that any information should be shown in place unless there is a convincing reason to point the user somewhere else.

> Given that the usage of MD5 for checksums is much more common than CRC32, maybe if we switched from CRC32 to MD5 (TBH I've got no idea why CRC32 was chosen originally?), then maybe it'd be more obvious and not actually need any explanation?

I think we should still provide some explanation. Using MD5 might mean that we need to provide *less* explanation than with CRC32, but we should still provide some IMHO.

alexandrosm, 2016-12-30:
Or maybe this is another case where we need the pattern of "short
explanation + learn more link"

konmouz, 2016-12-30:
Yes, I think the best way is to 'show a very small tooltip that describes what this is for in a few words (a short sentence)' + 'learn more'

jviotti, 2017-01-24:
@konmouz The problem is that we can't put links in tooltips (since they will disappear when the user tries to click on them since the mouse is no longer on top of the element showing the tooltip).

I guess let's do a short modal with it then?

alexandrosm, 2017-01-25:
How about a tool tip that stays on for a second or two so the user can
click the link?

jviotti, 2017-01-26:
A popover seems to be what we want here: https://angular-ui.github.io/bootstrap/#/popover

konmouz, 2017-01-27:
@jviotti can we trigger this on hover?

jviotti, 2017-01-27:
Looks like we can do so by passing `popover-trigger="mouseenter"`.

jviotti, 2017-01-27:
I've been trying to write up something for this without luck. What should we explain? There is no need at all to show the final checksum (its already being used internally) so it yields little benefit to the user. Also, where do we link the user to? The wikipedia page of CRC32?

jviotti, 2017-01-27:
Maybe @lurch can help us with the contents?

lurch, 2017-01-27:
Yeah, I've actually always wondered why the "verification complete" screen shows the CRC32 checksum. Maybe we should only bother showing the checksum, if we're flashing an image where we also have the checksum too (in whatever format), and then display the checksum in that form (rather than always CRC32).
With text something like
"Image verification produced a matching _&lt;checksum-type&gt;_ [checksum](https://en.wikipedia.org/wiki/Checksum):
_&lt;checksum&gt;_"
?
And if no checksum was provided, we could just display something like
"Image verification successful"
?

It might over-complicate matters, but I guess one possible extension would be to have an "Always display checksum" option in the settings, for advanced users who want to double-check their checksums for images they flash where we don't have the checksum in advance? (and obviously these advanced users would already know what exactly a checksum is). And as mentioned in #643 this option (if enabled) should display an MD5 checksum, with text like
"Image verification produced a MD5 [checksum](https://en.wikipedia.org/wiki/Checksum):
_&lt;checksum&gt;_"

And then maybe mouse-hovering over the checksum link, a small tooltip could pop up saying something like "Checksums are used to verify data integrity and ensure that no errors were introduced" ?

jviotti, 2017-01-27:
Even if the image has a associated checksum, showing this to the user has no benefits at all, but actually confuses them. I remember reading an article about Etcher when the user was wondering what he should do with the checksum we displayed. I guess showing it to the user gives the impression there is an action to do?

Maybe we should just say "Etcher validated the image successfully" and treat the details as something that merely happens under the hood?

What do you think @konmouz ? I think showing the checksum without any benefit is poor UX.

lurch, 2017-01-27:
Fair enough. I still wonder if advanced users might optionally want the ability to display the checksum? (e.g. if they're streaming an online image in a future version of Etcher, they have no opportunity to checksum the local disk file).

So let me change my above suggestion to:
By default, we just display "Etcher validated the image successfully"

If the user enabled the "Display checksum after validation" option in settings, show:
"Etcher validated the image successfully.
_&lt;checksum-type&gt;_ [checksum](https://en.wikipedia.org/wiki/Checksum): _&lt;checksum-type&gt;_"
where _&lt;checksum-type&gt;_ is either MD5, or the type of the image-checksum if we have it in advance from some kind of meta-data.

lurch, 2017-01-27:
I'm still of the opinion that if a user is advanced enough to explicitly enable the "show checksum" option in the settings (I guess we could have a link to the Wikipedia checksum article from the settings page), then they're advanced enough to not need a tooltip on the "validation successful" page telling them what a checksum is.
But maybe that's just my opinion because I'm an advanced user myself? :-/

Many other websites (e.g. https://www.raspberrypi.org/downloads/raspbian/ and https://www.raspberrypi.org/downloads/noobs/ ) provide checksums with no explanation of what they are or what they're used for.

jviotti, 2017-01-27:
> Many other websites provide checksums with no explanation of what they are or what they're used for.

True fact. I think we have a sound proposal for @konmouz to review.

konmouz, 2017-01-31:
@jviotti based on the conversation above, I think we can hide the checksum within some sort of details window. Meaning that if there is a 'space' to list flash details then we can also display checksum. That helps us since it removes it from the main UI and therefore if users are not aware of it they will ignore it while advanced users will just use it. The option method that @lurch suggested complicates things and still advanced users have to search the options to figure out that about this feature. I am thinking after burning and while you can read instructions there might be some area for details or a small details button.

jviotti, 2017-01-31:
Cool. Looking forward to the details window wireframe!

lurch, 2017-01-31:
Suggestions for the kind of information that we could potentially display after flashing (with no opinions on which should be shown, and whether they should be shown on the main 'success' page or only in a 'details' view) :
 * whether the validation succeeded or not (pretty important! :wink: )
 * filename of the image we just wrote
 * size of the image
 * checksum type and checksum of the image - probably omit this if validation failed (and sorry to keep banging on about this, but I think the checksum should default to MD5, or be the checksum-type of the metadata-provided checksum if available)
 * total elapsed time (both flashing and validating)
 * flashing time
 * average (and possibly maximum and minimum) flashing speed (gives an indication of the card's write-speed)
 * validation time
 * average (and possibly maximum and minimum) validating speed (gives an indication of the card's read-speed)
 * size of the drive the image just got written to
 * name / drive-letter the image just got written to (e.g. if you have multiple USB sticks plugged in, you might want to know which one you just wrote to and can now remove?)
 * the image's logo (if we have it from the metadata)
 * link to the publisher's website (if we have it from the metadata)
 * etc.

jviotti, 2017-01-31:
> whether the validation succeeded or not (pretty important! 😉 )

This might change the flow a bit. Currently we display an error ribbon on top of the main page if the validation fails, while we move to a new screen if the validation succeeds.

I personally never really liked that flow. Maybe we should also move to a new screen if validation fails? @konmouz ?

konmouz, 2017-01-31:
Yes, agree. So we can also display error details as well. I have some high priority tasks to finish but I'll try to wireframe this process based on @lurch's list asap.

lurch, 2018-02-23:
@jhermsmeier Just like #643 this can probably be closed, given that the checksum (CRC32 or otherwise) is never actually shown to the user any more.

jhermsmeier, 2018-02-23:
Yeah, let's close this – we can always re-open it, should it come back up.

closed 2018-02-23 by jhermsmeier