balena-io/etcher #995: SD Card Capacity incorrectly Reported

TimHop, 2017-01-03:
- **Etcher version:**
- **Operating system and architecture:**
- **Do you see any meaningful error information on DevTools?**

TimHop, 2017-01-03:
Etcher-1.0.0-beta.17-win32-x64
Windows 10

A 16GB SD Card (14.9 formatted) reports as only 5.9GB
Therefore image can not be burned

jviotti, 2017-01-04:
Hi @TimHop,

How does Windows report it when you open My PC? Any reason why you decided to close this ticket?

TimHop, 2017-01-04:
Hi Juan,

I closed the call by mistake!

Windows reports SD card size correctly at Circa 14.9gb

 

On first use Etcher reported the correct size.  A Raspberry PI image was used to test.

Test was successful.

Card was re-formatted and the issue reported arose.

Formatting using SDFormatter Quick & Full and Windows also Quick and Full.

 

The SD card was imaged using another app with a 14gb .img file successfully.

A very odd issue?

 

 

Tim
----------------------------------------------------------------

---------------------------------------------------------------- 

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com/email-signature> 
Version: 2016.0.7996 / Virus Database: 4749/13706 - Release Date: 01/04/17

jviotti, 2017-01-05:
That is indeed very odd. Can you try running the following batch script (https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat) before and after re-formatting such drive, and report back the results?

I'll try to reproduce this from my side as well in the mid time. Did you re-format is using Windows tooling or did you use something else?

jviotti, 2017-01-05:
For reference, we get the drive size from the [`Win32_DiskDrive`](https://msdn.microsoft.com/en-us/library/aa394132(v=vs.85).aspx) class, which is supposed to be *the* source of truth.

TimHop, 2017-01-11:
> For reference, we get the drive size from the  <https://msdn.microsoft.com/en-us/library/aa394132(v=vs.85).aspx> Win32_DiskDrive class, which is supposed to be the source of truth.

Hi Juan,

Sorry for the delay, we are busy developing a new product for the Raspberry PI J

 

I have now seen this on more than one SD Card!

Cards have been formatted with both Windows generic tooling and SDFormatter both Quick and Full with the same results.

 

It appears that once a card has been imaged for the Raspberry PI then formatted to receive a new image this issue occurs?

 

SDFormatter after Full Format:

 

Windows Properties:

 

Etcher –Drive V:

 

 

 

 

Tim

Version: 2016.0.7996 / Virus Database: 4749/13712 - Release Date: 01/05/17

jviotti, 2017-01-11:
Hi @TimHop ,

It looks like GitHub didn't show any attachments you added on the email, so I can't see the Windows properties, not SDFormatter output :/

Can you try running the following batch script (https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat) on the computer while the drives are connected and report back what you get?

The above script is what we run to detect the drives. If the error is reproducible by running it manually (e.g: you can check the size being reported for each drive), I can start investigating if there a subtle bug in there.

lurch, 2017-01-13:
Also, is this an internal SD card reader (e.g. in a laptop) or a USB SD card reader?
It would be interesting to hear if you can try using a different SD card reader, and if that gives you the same weird results?

TimHop, 2017-01-13:
Hi Andrew,

On a PC not Laptop.

Tried both built in and external.

Same result L

 

 

Tim
---------------------------------------------------------------- 

 

Version: 2016.0.7996 / Virus Database: 4749/13761 - Release Date: 01/13/17

lurch, 2017-01-16:
@TimHop Thanks for the info. Are you able to download and run the `win32.bat` that @jviotti linked to in an earlier message? (you'll need to run it in a Command-Prompt window in order to see the output)

mwebber, 2017-01-25:
I am having a similar problem. This is Windows 7, on a laptop with an internal card reader.
Etcher 1.1.0-beta18, Windows 64-bit build.

Windows Explorer shows my "Secure Digital Storage Device (E:)" as 14.4GB free of 14.4GB. I had just formatted the card using standard Windows format (this makes it FAT32). The card I purchased was advertised as 16GB, so that makes sense.

at "Select Drive", Etcher reports "SD Storage Card - 0.0 GB E: TOO SMALL FOR IMAGE"

I ran the `win32.bat` script, and here's what it reports:
```
device: "\\\\.\\PHYSICALDRIVE0"
description: "ST980411ASG"
size: 80023749120
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "SD Storage Card"
size: 8225280
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints:
  - path: "E:"
```
Ejecting and reinserting the card, and restarting Etcher, makes no difference.

lurch, 2017-01-25:
How peculiar! Just for the sake of comparison, do you have any other SD cards and/or SD card readers you can test with, to see if they all give these strange results?

If you format the card with the official SDFormatter tool, does that `win32.bat` still produce the same output?

mwebber, 2017-01-26:
After formatting with the official SDFormatter tool, `win32.bat` reports:
```
device: "\\\\.\\PHYSICALDRIVE0"
description: "ST980411ASG"
size: 80023749120
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "SD Storage Card"
size: 8225280
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints:
  - path: "E:"
```
So, the same as before. And Etcher still thinks it has 0 space.

Note that if I use the Win32 Disk Imager application (https://sourceforge.net/projects/win32diskimager/files/latest/download), it _can_ successfully write to the same SDCard (note: needs to be run as Administrator to get access to physical drive).

jviotti, 2017-01-26:
The issue definitely comes from our VBScript program. 8225280 bytes equal 0.00822528, so it makes sense Etcher things it has 0 space.

We get the drive size from the size property of the [`Win32_DiskDrive class`](https://msdn.microsoft.com/en-us/library/aa394132(v=vs.85).aspx), which is described as:

> Size of the disk drive. It is calculated by multiplying the total number of cylinders, tracks in each cylinder, sectors in each track, and bytes in each sector.

Can you try running `wmic diskdrive list brief` in cmd.exe and report back the results?

Its weird that Windows explorer gets the right size, since I assume it must be using `Win32_DiskDrive` as well.

jviotti, 2017-01-26:
Another alternative would be to try getting the size of the logical disk instead. Let me try that here.

jviotti, 2017-01-26:
Can you try the following modification of the `win32.bat` script? I included logging to check what are the sizes of the logical disks:

```vbscript
<!-- : Begin batch script
@echo off

:: Ensure System32 is in the PATH, to avoid weird
:: 'cscript' is not recognized as an internal or external command"" errors.
set PATH=%PATH%;%SYSTEMROOT%\System32

cscript //nologo "%~f0?.wsf"
exit /b

----- Begin wsf script --->
<job><script language="VBScript">

Class List
	Private Dictionary

	Private Sub Class_Initialize()
		Set Dictionary = CreateObject("Scripting.Dictionary")
	End Sub

	Public Sub Add(element)
		Dictionary.Add element, ""
	End Sub

	Public Function GetArray()
		GetArray = Dictionary.Keys()
	End Function

	Public Function Count
		Count = UBound(Dictionary.Keys()) + 1
	End Function

	' Only works for simple types (e.g: not objects)
	Public Function Has(element)
		Result = False
		For Each Key In Dictionary.Keys()
			If Key = Element Then
				Result = True
			End If
		Next
		Has = Result
	End Function
End Class

Set WMIService = GetObject("winmgmts:\\.\root\cimv2")

Function BooleanToString(ByVal Value)
	If Value Then
		BooleanToString = "True"
	Else
		BooleanToString = "False"
	End If
End Function

Function GetOperatingSystemDevice()
	Set OperatingSystemsColumn = WMIService.ExecQuery("SELECT SystemDrive FROM Win32_OperatingSystem")
	Err.Clear
	For Each OperatingSystem in OperatingSystemsColumn
		On Error Resume Next
		GetOperatingSystemDevice = OperatingSystem.Properties_("SystemDrive")
		If Err.Number <> 0 Then
			GetOperatingSystemDevice = Nothing
			Err.Clear
		End If
	Next
End Function

Function GetLogicalDisks(ByVal DriveDevice)
	Set GetLogicalDisks = new List
	Set DrivePartitionsColumn = WMIService.ExecQuery _
		("ASSOCIATORS OF {Win32_DiskDrive.DeviceID=""" & _
			DriveDevice & """} WHERE AssocClass = " & _
				"Win32_DiskDriveToDiskPartition")

	For Each DrivePartition In DrivePartitionsColumn
		Set DriveLogicalDisksColumn = WMIService.ExecQuery _
			("ASSOCIATORS OF {Win32_DiskPartition.DeviceID=""" & _
				DrivePartition.DeviceID & """} WHERE AssocClass = " & _
					"Win32_LogicalDiskToPartition")

		For Each DriveLogicalDisk In DriveLogicalDisksColumn
			Set LogicalDisk = CreateObject("Scripting.Dictionary")
			LogicalDisk.Add "Device", DriveLogicalDisk.DeviceID
      WScript.Echo "Logical Size for " & DriveLogicalDisk.DeviceID & ": " & DriveLogicalDisk.Size
			LogicalDisk.Add "IsProtected", DriveLogicalDisk.Access = 1
			GetLogicalDisks.Add(LogicalDisk)
		Next
	Next
End Function

Function GetTopLevelDrives()
	OperatingSystemDevice = GetOperatingSystemDevice()
	Set GetTopLevelDrives = new List
	Set TopLevelDrivesColumn = WMIService.ExecQuery("SELECT * FROM Win32_DiskDrive")
	For Each TopLevelDrive In TopLevelDrivesColumn
		Set Summary = CreateObject("Scripting.Dictionary")

		DeviceID = Replace(TopLevelDrive.DeviceID, "\", "\\")
		Summary.Add "Device", DeviceID

		Summary.Add "Description", TopLevelDrive.Caption
		Summary.Add "Size", TopLevelDrive.Size

		Set Mountpoints = new List
		IsRemovable = InStr(TopLevelDrive.MediaType, "Removable") = 1
		IsProtected = False

		Set LogicalDisks = GetLogicalDisks(DeviceID)

		For Each LogicalDisk In LogicalDisks.GetArray()
			If Not Mountpoints.Has(LogicalDisk.Item("Device")) Then
				Mountpoints.Add(LogicalDisk.Item("Device"))
			End If

			If LogicalDisk.Item("IsProtected") Then
				IsProtected = True
			End If

			If LogicalDisk.Item("Device") = OperatingSystemDevice Then
				IsRemovable = False
			End If
		Next

		Summary.Add "Mountpoints", Mountpoints
		Summary.Add "IsRemovable", IsRemovable
		Summary.Add "IsProtected", IsProtected

		GetTopLevelDrives.Add(Summary)
	Next
End Function

For Each TopLevelDrive In GetTopLevelDrives().GetArray()
	Wscript.Echo "device: """ & TopLevelDrive.Item("Device") & """"
	Wscript.Echo "description: """ & TopLevelDrive.Item("Description") & """"
	Wscript.Echo "size: " & TopLevelDrive.Item("Size")
	Wscript.Echo "raw: """ & TopLevelDrive.Item("Device") & """"
	Wscript.Echo "system: " & BooleanToString(Not TopLevelDrive.Item("IsRemovable"))
	Wscript.Echo "protected: " & BooleanToString(TopLevelDrive.Item("IsProtected"))

	If TopLevelDrive.Item("Mountpoints").Count = 0 Then
		Wscript.Echo "mountpoints: []"
	Else
		Wscript.Echo "mountpoints:"
		For Each Mountpoint In TopLevelDrive.Item("Mountpoints").GetArray()
			Wscript.Echo "  - path: """ & Mountpoint & """"
		Next
	End If

	Wscript.Echo ""
Next
</script></job>
```

mwebber, 2017-01-26:
**Test 1**
Result after formatted using official SDFormatter. SDFormatter reports that it's 14GB (approx), and so does Windows Explorer.
You can see that the logical size for E is reported as 15.5GB, and size is 8225280

```
>wmic diskdrive list brief
Caption          DeviceID            Model        Partitions  Size
ST980411ASG      \\.\PHYSICALDRIVE0  ST980411ASG  2           80023749120
SD Storage Card  \\.\PHYSICALDRIVE1               1           8225280

>win32-new.bat
Logical Size for C:: 79866884096
Logical Size for E:: 15512698880
device: "\\\\.\\PHYSICALDRIVE0"
description: "ST980411ASG"
size: 80023749120
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "SD Storage Card"
size: 8225280
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints:
  - path: "E:"
```
At this point, Etcher thinks the card has zero space.

**Test 2**
I then wrote a 2.5GB .img file to the card using Win32DiskImager (running as an admin). That worked.

Windows explorer has a size for the card of 0 (gotten by right-click on the drive in Windows Explorer, and select Properties). That might be expected, I don't know.
```
>wmic diskdrive list brief
Caption          DeviceID            Model        Partitions  Size
ST980411ASG      \\.\PHYSICALDRIVE0  ST980411ASG  2           80023749120
SD Storage Card  \\.\PHYSICALDRIVE1               1           8225280

>win32-new.bat
Logical Size for C:: 79866884096
Logical Size for E::
device: "\\\\.\\PHYSICALDRIVE0"
description: "ST980411ASG"
size: 80023749120
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "SD Storage Card"
size: 8225280
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints:
  - path: "E:"
```
Hope that helps.

jviotti, 2017-01-26:
Hi @mwebber ,

Thanks a lot for the testing! In the first case, looks like we can indeed rely on the logical drives size. I'll do some modifications to the script and potentially ping you for one more try.

Regarding the second scenario, it makes no sense that after writing an image Windows Explorer sees the drive as size 0. I have no clue what could be causing that :S

jviotti, 2017-01-26:
Hi @mwebber ,

Can you try this new version of the script? What I'm doing here is get the size of every logical disk (e.g: each drive letter) and add them all together. If the sum of the logical size is greater than of the drive size, we display the logical size, otherwise we display the drive as we normally do.

This should at least mitigate your issue but its not way a definitive fix. Imagine the following scenario: you have an 8 GB drives with 4 partitions. One of them is FAT while the remaining are EXT4 partitions. Since Windows won't read EXT4 partitions, it won't assign a drive letter to them, and therefore their sizes won't be considered in the total logical size addition. This means that if the drive size is incorrectly reported as zero, after my fix, the drive size will only equal the size of the FAT partition, which might be 25% of the drive.

```vbscript
<!-- : Begin batch script
@echo off

:: Ensure System32 is in the PATH, to avoid weird
:: 'cscript' is not recognized as an internal or external command"" errors.
set PATH=%PATH%;%SYSTEMROOT%\System32

cscript //nologo "%~f0?.wsf"
exit /b

----- Begin wsf script --->
<job><script language="VBScript">

Class List
	Private Dictionary

	Private Sub Class_Initialize()
		Set Dictionary = CreateObject("Scripting.Dictionary")
	End Sub

	Public Sub Add(element)
		Dictionary.Add element, ""
	End Sub

	Public Function GetArray()
		GetArray = Dictionary.Keys()
	End Function

	Public Function Count
		Count = UBound(Dictionary.Keys()) + 1
	End Function

	' Only works for simple types (e.g: not objects)
	Public Function Has(element)
		Result = False
		For Each Key In Dictionary.Keys()
			If Key = Element Then
				Result = True
			End If
		Next
		Has = Result
	End Function
End Class

Set WMIService = GetObject("winmgmts:\\.\root\cimv2")

Function BooleanToString(ByVal Value)
	If Value Then
		BooleanToString = "True"
	Else
		BooleanToString = "False"
	End If
End Function

Function GetOperatingSystemDevice()
	Set OperatingSystemsColumn = WMIService.ExecQuery("SELECT SystemDrive FROM Win32_OperatingSystem")
	Err.Clear
	For Each OperatingSystem in OperatingSystemsColumn
		On Error Resume Next
		GetOperatingSystemDevice = OperatingSystem.Properties_("SystemDrive")
		If Err.Number <> 0 Then
			GetOperatingSystemDevice = Nothing
			Err.Clear
		End If
	Next
End Function

Function GetLogicalDisks(ByVal DriveDevice)
	Set GetLogicalDisks = new List
	Set DrivePartitionsColumn = WMIService.ExecQuery _
		("ASSOCIATORS OF {Win32_DiskDrive.DeviceID=""" & _
			DriveDevice & """} WHERE AssocClass = " & _
				"Win32_DiskDriveToDiskPartition")

	For Each DrivePartition In DrivePartitionsColumn
		Set DriveLogicalDisksColumn = WMIService.ExecQuery _
			("ASSOCIATORS OF {Win32_DiskPartition.DeviceID=""" & _
				DrivePartition.DeviceID & """} WHERE AssocClass = " & _
					"Win32_LogicalDiskToPartition")

		For Each DriveLogicalDisk In DriveLogicalDisksColumn
			Set LogicalDisk = CreateObject("Scripting.Dictionary")
			LogicalDisk.Add "Device", DriveLogicalDisk.DeviceID
			LogicalDisk.Add "Size", DriveLogicalDisk.Size
			LogicalDisk.Add "IsProtected", DriveLogicalDisk.Access = 1
			GetLogicalDisks.Add(LogicalDisk)
		Next
	Next
End Function

Function GetTopLevelDrives()
	OperatingSystemDevice = GetOperatingSystemDevice()
	Set GetTopLevelDrives = new List
	Set TopLevelDrivesColumn = WMIService.ExecQuery("SELECT * FROM Win32_DiskDrive")
	For Each TopLevelDrive In TopLevelDrivesColumn
		Set Summary = CreateObject("Scripting.Dictionary")

		DeviceID = Replace(TopLevelDrive.DeviceID, "\", "\\")
		Summary.Add "Device", DeviceID

		Summary.Add "Description", TopLevelDrive.Caption

		Set Mountpoints = new List
		IsRemovable = InStr(TopLevelDrive.MediaType, "Removable") = 1
		IsProtected = False

		Set LogicalDisks = GetLogicalDisks(DeviceID)
		LogicalSize = 0

		For Each LogicalDisk In LogicalDisks.GetArray()
			If Not Mountpoints.Has(LogicalDisk.Item("Device")) Then
				Mountpoints.Add(LogicalDisk.Item("Device"))
			End If

			LogicalSize = LogicalSize + LogicalDisk.Item("Size")

			If LogicalDisk.Item("IsProtected") Then
				IsProtected = True
			End If

			If LogicalDisk.Item("Device") = OperatingSystemDevice Then
				IsRemovable = False
			End If
		Next

		If LogicalSize > TopLevelDrive.Size Then
			Summary.Add "Size", LogicalSize
		Else
			Summary.Add "Size", TopLevelDrive.Size
		End If

		Summary.Add "Mountpoints", Mountpoints
		Summary.Add "IsRemovable", IsRemovable
		Summary.Add "IsProtected", IsProtected

		GetTopLevelDrives.Add(Summary)
	Next
End Function

For Each TopLevelDrive In GetTopLevelDrives().GetArray()
	Wscript.Echo "device: """ & TopLevelDrive.Item("Device") & """"
	Wscript.Echo "description: """ & TopLevelDrive.Item("Description") & """"
	Wscript.Echo "size: " & TopLevelDrive.Item("Size")
	Wscript.Echo "raw: """ & TopLevelDrive.Item("Device") & """"
	Wscript.Echo "system: " & BooleanToString(Not TopLevelDrive.Item("IsRemovable"))
	Wscript.Echo "protected: " & BooleanToString(TopLevelDrive.Item("IsProtected"))

	If TopLevelDrive.Item("Mountpoints").Count = 0 Then
		Wscript.Echo "mountpoints: []"
	Else
		Wscript.Echo "mountpoints:"
		For Each Mountpoint In TopLevelDrive.Item("Mountpoints").GetArray()
			Wscript.Echo "  - path: """ & Mountpoint & """"
		Next
	End If

	Wscript.Echo ""
Next
</script></job>
```

As far as I can see if Windows incorrectly detects the drive size there is not much we can do. Maybe @jhermsmeier has some ideas on how to get the drive size in a bullet proof way?

lurch, 2017-01-26:
@jviotti I realise it's probably not something you can do much about, but would it be worth trying to dig into why `wmic diskdrive list brief` is reporting the wrong size?

@mwebber Probably a silly question, but do you have any ideas about how we might be able to reproduce your problem, for debugging purposes? (and as I asked before, are you able to check with different SD cards to see if it's a card-specific problem, or something else?) Would you mind running [h2testw](https://sosfakeflash.wordpress.com/h2testw/) over your SD card as well, please?

jviotti, 2017-01-27:
> @jviotti I realise it's probably not something you can do much about, but would it be worth trying to dig into why wmic diskdrive list brief is reporting the wrong size?

Yeah, seems to be an OS/driver issue. The MBR/GPT should contain information about the size of whole drive which should be bullet proof, but accessing the partition table of each drive on each call to drivelist sounds like a bad idea. We probably can't do that with VBScript anyway.

lurch, 2017-01-27:
> The MBR/GPT should contain information about the size of whole drive which should be bullet proof

Nope, the MBR/GPT is _far_ from bulletproof - it's just data written to the drive (in a particular location), and so can be set to anything. E.g. if I do `dd if=/dev/zero of=/dev/sdX` then the whole drive will be written with zeros, and so the MBR and GPT won't even exist. But there's still OS-level info (the cylinders, heads, sectors, etc. that you referred to earlier) which can be retrieved and used to calculate the size of the drive (even if it's totally zeroed, or filled with random bytes).
That's why I've asked @mwebber to run the `h2testw` program, because I'm wondering if he's using a fake SD card with hacked firmware, which is reporting incorrect CHS values to the OS, which is why the `wmic diskdrive list brief` calculation is coming out wrong... But that's just a theory so far, until we get the h2testw results ;-)

jviotti, 2017-01-27:
True. Let's see the `h2testw` results :)

mwebber, 2017-01-27:
Ok, here's the latest results:

# Test 1
1) After successfully writing disk image (containing Armbian Linux) using Win32 Disk Imager application, I put it in my OrangePi. That booted ok, and it does something to the card to recover the extra space beyond the image. Now Linux as expected reports it as 1 14GB drive.

2) Put SDCard in Windows 7. Windows 7 tells me that it needs formatting. I presume that it's ext4, hence not recognised by Windows. Test results:
```
>wmic diskdrive list brief
Caption          DeviceID            Model        Partitions  Size
ST980411ASG      \\.\PHYSICALDRIVE0  ST980411ASG  2           80023749120
SD Storage Card  \\.\PHYSICALDRIVE1               1           8225280

>win32-new2.bat
device: "\\\\.\\PHYSICALDRIVE0"
description: "ST980411ASG"
size: 80023749120
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "SD Storage Card"
size: 8225280
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints:
  - path: "E:"
```
**h2testw** won't test it, since it doesn't contain a recognised filesystem.

# Test 2
Format card using the official SDFormatter tool.
Windows now thinks the drive has 14.4 GB, as expected.
```
>wmic diskdrive list brief
Caption          DeviceID            Model        Partitions  Size
ST980411ASG      \\.\PHYSICALDRIVE0  ST980411ASG  2           80023749120
SD Storage Card  \\.\PHYSICALDRIVE1               1           8225280

>win32-new2.bat
device: "\\\\.\\PHYSICALDRIVE0"
description: "ST980411ASG"
size: 80023749120
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "SD Storage Card"
size: 8225280
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints:
  - path: "E:"
```
**h2testw** unfortunately I ran out of time, but it had written 10GB without errors when I cancelled it.

Sorry I don't have a spare card I can test this with. I will see if I can try on another machine.

jviotti, 2017-01-27:
> because I'm wondering if he's using a fake SD card with hacked firmware, which is reporting incorrect CHS values to the OS,

Checking if you can reproduce this on other operating systems would help validate that theory.

lurch, 2017-01-28:
I found http://serverfault.com/questions/760253/how-to-list-basic-information-about-disks-cylinders-heads-sectors-sector-siz which provides info on how to get the CHS values from within Windows.
So @mwebber if you can run `wmic diskdrive > diskdrive.txt` and then paste the contents of diskdrive.txt here, that'd be useful. And if you can boot up the card in your OrangePi, the output of `sudo fdisk -l` will be really useful too. And allowing `h2testw` to run to completion will help too.
Sorry to ask so many questions, but the more info we have, the better we'll be able to diagnose this weird problem.

lurch, 2017-01-28:
(oops, mouse-slip)

lurch, 2017-02-15:
Pinging @mwebber again - any chance you could run the tests I mentioned in my last message?
We've just had a user in our gitter channel also report this problem, and he also gets `size: 8225280` from the drivelist `win32.bat`

mwebber, 2017-02-17:
My apologies for the lack of an earlier response; I've been on holidays. Latest results:
# Step 1
Format card using the official SDFormatter tool.
Windows now thinks the drive has 14.4 GB, as expected.
Ran `wmic diskdrive` and attached output as `step1.txt`
[step1.txt](https://github.com/resin-io/etcher/files/783347/step1.txt)
# Step 2
Run `h2testw` to completion , and it reported 17794 MByte
> Test finished without errors.
> You can now delete the test files *.h2w or verify them again.
> Writing speed: 9.97 MByte/s
> Reading speed: 14.7 MByte/s
> H2testw v1.4
# Step 3
Attempted to use Etcher to write an image, and as previously it reported no space.
Successfully wrote disk image (containing Armbian Linux) using Win32 Disk Imager application.
When I get home I'll boot up in the OrangePi and see what it says about the card.

lurch, 2017-02-17:
Thanks @mwebber !
So, looking at your output of `wmic diskdrive`, if I plug in the `TotalHeads`, `TotalCylinders`, `SectorsPerTrack` and `BytesPerSector` values for your `ST980411ASG` hard drive into the online calculator at http://www.csgnetwork.com/mediasizecalc.html then all the figures tally up, and I get the correct total size as 80023749120 bytes (i.e. 80GB).
But for some strange reason, `wmic diskdrive` reports your SD card as only having 1 `TotalCylinders` (!), and plugging the relevant values into the online calculator again I get a value of just 8225280 bytes (i.e. 8MB), which matches what we were seeing earlier.

However IIRC the Heads / Cylinders / Sectors values for an SD card are just "backwards derived" from the card's _actual_ capacity, which is calculated from values stored in an [internal register](http://electronics.stackexchange.com/questions/172797/is-it-possible-to-check-max-storage-address-of-a-sd-card-via-a-command) .
So it sounds to me (I'm kinda guessing here!) like your SD card-reader (or the OS-driver used by your SD card-reader) is reporting both the card's _actual_ size to Windows (and this is the "correct" size used by Windows Explorer and h2testw) and _also_ reporting a bogus `TotalCylinders` value to `wmic`, which is why `wmic` calculates the incorrect size, and unfortunately this is also the size reported to Etcher (via the VBScript in `win32.bat`) :-(

So my current theory is that if you tried another (different) SD card in your SD card-reader, Windows Explorer and h2testw would again get the size 'correct', but Etcher would get the size 'wrong' (due to the wrong number of cylinders being reported to wmic). And I suspect that if you used a different SD card-reader, wmic would then get the correct number of cylinders, and Etcher would be able to report the SD cards' size correctly.
Still, it'll be interesting to see what values your OrangePi reports.

If you're willing to do even more debugging, it might be interesting to try running a LiveUSB linux distro on your Windows PC, and seeing what values get reported then - which would allow us to track down whether it's a firmware-error built into the reader itself, or whether it's just an error in the Windows driver.

@jviotti Based on my assumptions above, it sounds like SDFormatter / Windows Explorer / h2testw are ignoring the wmic values, and getting the SD card's capacity by some other means? :-/
And there's a bit more info about the CHS values for an SD card (and links to further reading) at https://boundarydevices.com/how-big-is-your-sd-card/
For PCI-connected SD card readers (i.e. `/dev/mmcblk0` under Linux, which is the only time the CSD register is accessible) I believe the raw-capacity->CHS values calculation is done by the OS, but for USB-connected SD card readers (i.e. `/dev/sdc` under Linux) I wonder if it's the firmware in the reader that does the conversion?
Note also that for (mechanical) hard-drives the CHS values->raw capacity calculation always works exactly, but for SSDs, USB Flash drives, SD cards etc. the CHS values are only a "nearest approximation", i.e. (examples from my laptop) : 
```
Disk /dev/sda: 1024.2 GB, 1024209543168 bytes
255 heads, 63 sectors/track, 124519 cylinders, total 2000409264 sectors
Units = sectors of 1 * 512 = 512 bytes
```
255 * 63 * 124519 = 2000397735 which is slightly less than the _real_ total number of sectors.
2000409264 * 512 = 1024209543168

```
Disk /dev/sdb: 200.8 GB, 200788672512 bytes
255 heads, 63 sectors/track, 24411 cylinders, total 392165376 sectors
Units = sectors of 1 * 512 = 512 bytes
```
255 * 63 * 24411 = 392162715
392165376 * 512 = 200788672512
```
Disk /dev/mmcblk0: 62.7 GB, 62730010624 bytes
64 heads, 32 sectors/track, 59824 cylinders, total 122519552 sectors
Units = sectors of 1 * 512 = 512 bytes
```
64 * 32 * 59824 = 122519552 (ooh, this time it actually matches!)
122519552 * 512 = 62730010624

...and actually, based on the figures above, I guess if wmic always calculates the total size purely from the CHS values, then there'll be some SD cards / USB Flash drives where it actually reports the drive as being slightly smaller than it actually is? :-/

jviotti, 2017-02-17:
Interesting information. I'll try to find the source of SD Formatter and/or Win32 Disk Imager and see if I can find where they get the information from.

lurch, 2017-02-17:
I just tried google-searching, and couldn't find any recommendations _other_ than the `Win32_DiskDrive` that you're already using. But from a quick test on my Windows 7 laptop, `wmic` indeed reports the size of my 200GB microSD card as 200787310080 bytes ( = 255 * 63 * 24411 * 255) whereas my Linux laptop reports the same CHS values, but reports the raw size as 200788672512 bytes (an extra 1.3MB).
I'm now creating a raw image of the drive on my Windows7 laptop using Win32 DiskImager (which will take a while!) to see what size that comes out as. The "Disk Management" pane in the "Computer Management" control panel on Windows7 only reports the raw drive sizes in GB or MB, which isn't that helpful when you're trying to compare exact byte sizes ;)
I suspect the size reported by Windows Explorer is actually the (formatted) filesystem size, rather than the raw drive size.

AFAIK the SD Formatter tool is proprietary, so I doubt you'll be able to find its source code.

lurch, 2017-02-17:
...I just dived into the Win32DiskImager [code](https://sourceforge.net/p/win32diskimager/code/ci/master/tree/src/disk.cpp#l213), and it seems they're using [IOCTL_DISK_GET_DRIVE_GEOMETRY_EX ](https://msdn.microsoft.com/en-us/library/windows/desktop/aa365171(v=vs.85).aspx) and then dividing the returned `DiskSize` by `sectorsize`, so I wonder if that _might_ give the "true" (larger) sector count that `fdisk` in Linux does?

jhermsmeier, 2017-02-17:
Just a note on CHS values: CHS addressing has been deprecated, and shouldn't really be used anymore or relied upon, unless necessary (i.e. with older magnetic storage devices).

If I remember this correctly, these days devices often report CHS addresses & geometries as `[0xFE,0x3F,0xFF]` – which is `(1023, 254, 63)` – the maximum geometry / value and the equivalent of `-1` or `0xFFFFFFFF` in other places – usually meaning that the underlying device's size exceeds the 24bit CHS addressing spectrum, or that CHS addressing is not supported.
Using the maximum here (instead of zeroing it) enables old devices to still be able to address the device as far as they can (i.e. so that older BIOS can still boot from a modern drive).

lurch, 2017-02-17:
Aha, also found more useful info here https://bugs.launchpad.net/win32-image-writer/+bug/932321

> Just a note on CHS values: CHS addressing has been deprecated, and shouldn't really be used anymore

Yup, but from what I can tell, as noted above, it appears that [Win32_DiskDrive](https://msdn.microsoft.com/en-us/library/aa394132(v=vs.85).aspx) indeed still uses the CHS values (with a cylinder count greater than 1023) to calculate the raw size - gah! :-(

Hopefully `IOCTL_DISK_GET_DRIVE_GEOMETRY_EX` is the right path to head down, as backed up by the bug-report mentioned in this comment. Fingers crossed...  but I guess as this is Win32 API code, it might need a native `.exe` in drivelist's `win32.bat`, rather than being retrievable from pure VBScript?

jviotti, 2017-02-17:
We can apparently create a C# DLL and call it from VBScript: https://www.codeproject.com/articles/79314/how-to-call-a-net-dll-from-a-vbscript, though this starts to get ugly, I really want to avoid native dependencies on this module.

lurch, 2017-02-17:
> I really want to avoid native dependencies on this module.

Based on what I've discovered so far, I don't think that's possible, unfortunately :-(
At least with .Net we wouldn't need different versions for 32bit / 64bit. (Or would we, given that this is touching native platform APIs?)

lurch, 2017-02-18:
Okay... my 200GB microSD card _finally_ finished getting read by Win32DiskImager on my Windows7 laptop, so now it's conclusion time... ;-)

`Win32DiskImager` produced a 200788672512 byte file, which exactly matches the size reported by `fdisk` on Linux. But if I then try using Etcher v1.0.0-beta.18 on my Windows 7 laptop to write that same image file back to the same microSD card that I've just read it from, I'm unable to do so as it tells me the microSD card is "TOO SMALL FOR IMAGE" (i.e. I'm not able to select it as a destination drive).

So even with a "fully working" SD card-reader and a branded SD card (i.e. that doesn't report just 1 cylinder!) the issue described in this PR still needs fixing.
On my Windows7 laptop, this is what `wmic diskdrive list brief` reports:
```
Caption                                   DeviceID            Model                                     Partitions  Size          
HITACHI HTS543232A7A384 SATA Disk Device  \\.\PHYSICALDRIVE0  HITACHI HTS543232A7A384 SATA Disk Device  9           320070320640  
Lexar microSD RDR USB Device              \\.\PHYSICALDRIVE1  Lexar microSD RDR USB Device              2           200787310080  
SAMSUNG SSD PM851 M.2 22 USB Device       \\.\PHYSICALDRIVE2  SAMSUNG SSD PM851 M.2 22 USB Device       1           256052966400
```
and this is what drivelist's `win32.bat` reports:
```
device: "\\\\.\\PHYSICALDRIVE0"
description: "HITACHI HTS543232A7A384 SATA Disk Device"
size: 320070320640
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"
  - path: "D:"
  - path: "E:"
  - path: "Q:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "Lexar microSD RDR USB Device"
size: 200787310080
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints: []

device: "\\\\.\\PHYSICALDRIVE2"
description: "SAMSUNG SSD PM851 M.2 22 USB Device"
size: 256052966400
raw: "\\\\.\\PHYSICALDRIVE2"
system: True
protected: False
mountpoints:
  - path: "G:"
```
On my Linux laptop (with the same two external drives connected), drivelist's `linux.sh` reports:
```
device: /dev/sda
description: "SanDisk SD8SN8U1"
size: 1024209543168
mountpoints:
  - path: /
  - path: /boot/efi
  - path: /home
  - path: /var/lib/docker/aufs
raw: /dev/sda
protected: False
system: True

device: /dev/sdb
description: "SSD PM851 M.2 22"
size: 256060514304
mountpoints: []
raw: /dev/sdb
protected: False
system: False

device: /dev/sdc
description: "microSD RDR"
size: 200788672512
mountpoints: []
raw: /dev/sdc
protected: False
system: False
```

To try and recap: [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/) (since [this patch](https://bugs.launchpad.net/win32-image-writer/+bug/932321)) uses [IOCTL_DISK_GET_DRIVE_GEOMETRY_EX](https://msdn.microsoft.com/en-us/library/windows/desktop/aa365171(v=vs.85).aspx) to get the _actual_ size of the drive, and reads the full 200788672512 bytes. However the [Win32_DiskDrive](https://msdn.microsoft.com/en-us/library/aa394132(v=vs.85).aspx) that we're currently using in drivelist's `win32.bat` incorrectly (IMHO) still calculates the drive's size using the CHS values, and hence reports the drivesize to Etcher as only 200787310080 bytes.
So the 'fix' is to somehow modify drivelist's `win32.bat` to use `IOCTL_DISK_GET_DRIVE_GEOMETRY_EX` to get the size on Windows instead of using `Win32_DiskDrive`, and I _believe_ that will also fix the problem identified by @mwebber where Etcher only thinks his drive has a capacity of 8225280 bytes (and maybe fix @TimHop 's problem too). I'll create an appropriate issue in the drivelist repo....

P.S. As well as the size differences, another interesting point to note is that `win32.bat` reports that my "SAMSUNG SSD PM851 M.2 22 USB Device" (an M.2 SSD in an external USB enclosure) as `system: True`, but `linux.sh` reports "SSD PM851 M.2 22" as `system: False`.

jviotti, 2017-02-19:
Hi @lurch ,

Amazing findings! Maybe a small C# program is the way to go here. Alternatively, we could make `drivelist` a native C++ Nodejs add-on, and use the C Win32 API to do it. This means we avoid shipping a compiled binary with the code, and that we avoid an extra child process.

Regarding your system drive, can you open a new issue as well, including the output of `lsblk -b -d "$device" --output SIZE,RO,RM,MODEL` and `udevadm info --query=property --export --export --prefix=UDEV_ --name="$disk"`?

joeinaustin, 2017-04-18:
Just ran into the 0.0GB issue and see that this issue has been around for months.  I also have a win7 laptop with internal SD reader and a brand new 8GB sd card recognized by windows and sdformatter.

Any resolution in sight for Etcher?

jviotti, 2017-04-18:
Hi @joeinaustin ,

We're currently porting `drivelist` to C++, however the task is not trivial, and its likely to take a while, I'm afraid.

closed 2017-07-12 by jviotti

lurch, 2017-07-13:
Please see https://github.com/resin-io/etcher/pull/1574#issuecomment-315054168

PR #1574 (2017-07-12): upgrade: drivelist to v5.0.25
This version contains various improvements on the drive detection system
mainly on Windows. The improvements can be summarized as follows:

- Drivelist no longer spawns a .bat script to perform its job. This
  means that a lot of spawning related issues are now fixed

- Drivelist doesn't fetch drive sizes from WMI anymore, which was known
  to incorrectly report a very small size for certain drives, causing
  the famous "0.0 GB" Windows issues (see
  resin-io-modules/drivelist#142)

- Cleanup temporary scripts after execution

This PR also sets the DRIVELIST_DEBUG [... cut]