balena-io/etcher #996: No Partitions Found After Successful Flash

yldbouk, 2017-01-03:
- **Etcher version:** 1.0.0-beta.17
- **Operating system and architecture:** `MacOS Sierra 10.12.2 (16C67)` on a `MacBook Air (13-inch, Early 2015) | Processor: 1,6 GHz Intel Core i5 | Memory: 4 GB 1600 MHz DDR3`
- **Do you see any meaningful error information on DevTools?** Nope

So, when I try to flash an iso to an External Hard Drive, everything goes normal, it works like nothing bad happened. but when I look at the drive in Disk Utility, no partitions show up, and it says that it's not bootable. It gives the name:
`Generic USB Disk Media`
Here is a Screen shot of some info if you need it:
[image]
I don't know if i did something wrong or missed a step, but im pretty sure i did nothing wrong. DevTools didn't show anything. I don't like writing issues because it always gets awkward for me when im trying to end it, so...

Please help, Thank You.

jviotti, 2017-01-04:
What image did you flash? It might be that OS X can't detect the partitions of such image, which usually happens with Linux images (OS X can't read them).

yldbouk, 2017-01-04:
I flashed multiple images. I checked their health and it was fine. I could open them, and I had read/write perms

jviotti, 2017-01-04:
Hm, can you directly link me to an image you've flashes so I can try to reproduce here?

Horadrim, 2017-01-17:
I can confirm that the same thing happens, but under Linux (Ubuntu 14.04). No matter what image I try to write, ti just renders the usb stick unusable as no partitions are found afterwards. The last image I tried "burning" was lubuntu (lubuntu-15.10-desktop-i386.iso) downloaded directly from the Lubuntu website.

lurch, 2017-01-17:
This is weird. Does the image that you try flashing successfully validate okay after writing?
Can you give more details about the particular USB stick that you're using?

jviotti, 2017-01-17:
Hi @Horadrim ,

Can you compute a checksum out of your file and compare it to the one provided by Ubuntu, just to discard any incomplete image problems?

Can you show us what you see when inspecting the drive with say: `gparted`?

Horadrim, 2017-01-18:
@jviotti ,
I`ve downloaded a fresh copy of lubuntu 14.04 . These are the checksums:
https://i.gyazo.com/faf01a54c1035d441009b4dd5e8f06ff.png

This is what gparted shows after the "etching":
https://i.gyazo.com/a849784fd266b9fedec3bac4b2369b8d.jpg

jviotti, 2017-01-18:
Hi @Horadrim ,

Have you actually tried booting the images? I think we're seeing the effects of Ubuntu hybrid ISO's hacks: https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

lurch, 2017-01-18:
@Horadrim I believe this may be "expected behaviour"...
I just tried flashing the `lubuntu-15.10-desktop-i386.iso` onto a microSD card, and GParted looks like this:
[image]

Meanwhile, the "Disks" program shows slightly more information:
[image]

...and clicking on the 'play' icon, the ISO9660 filesystem does indeed get mounted, and I can see the contents just fine. I guess this is because this is a "standard" ISO, and not one of the "hybrid" ISOs used by Ubuntu https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

I don't want to reboot my laptop right now (I'm in the middle of writing this comment ;) ), but maybe this means Lubuntu ISOs need to be written to CD / DVD, and don't work when written to USB? (or maybe whether the USB boots in this case is one of those BIOS-dependent things?)

lurch, 2017-01-18:
Now downloading a 'regular' Ubuntu ISO to see if that behaves the same or differently to Lubuntu...

Horadrim, 2017-01-18:
Unfortunately after "etching" the ISO, the USB doesn`t boot. I tried on several computers (I`m working in an office with about 20 computers). If I use another software like LiLi for example, it does boot. So I exclude the USB stick as a factor.

jviotti, 2017-01-18:
LiLi will mount the image, extract the files, partition the drive, and install GRUB on it, while Etcher does a byte-by-byte flashing (which includes the partition table described by the image).

In any case, our flash strategy should work fine here. I'll give this a go myself.

lurch, 2017-01-18:
That's because LiLi does 'more' than Etcher. Etcher (currently) just blindly copies the image to disk byte-by-byte (just like `dd`), without doing anything special.
See also #759 (and the links contained therein) and #553 and #413

lurch, 2017-01-18:
So, here's the results of my tests...

After downloading `ubuntu-16.04.1-desktop-amd64.iso` and writing it to the same microSD as before, this is what GParted looks like (after ignoring the error referred to [here](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos) ) :
[image]

and this is what "Disks" looks like:
[image]

Just like last time, clicking on the 'play' icon in the Disks application mounts the ISO9660 filesystem and allows its contents to be displayed.

And when I tried booting the microSD cards each time, **both** of them worked! (on my 2015 Dell XPS 13)

So in conclusion: Lubuntu is using a "regular" ISO, and Ubuntu is using a "hybrid" ISO. It seems that @Horadrim 's computer is only able to boot USB flash drives created from "hybrid" ISOs, but my computer (which I guess must have a more-capable BIOS) is able to boot flash drives created from both "hybrid" _and_ "regular" ISOs.
And who knows, perhaps some (old?) BIOSes are only able to boot from MBR-partitioned USB Flash drives, and not even from USB Flash drives created from hybrid ISOs? \*shrug\*

@jviotti I think this PR should be closed, and linked to #413 ? And perhaps we should add some additional notes about ISOs to the docs?

jviotti, 2017-01-19:
> And perhaps we should add some additional notes about ISOs to the docs?

I think the "Why is my drive not bootable?" covers the gist of this issue, but do you have any ideas to expand it/make it clearer?

closed 2017-01-19 by jviotti

WasabiFan, 2017-01-19:
> ...do you have any ideas to expand it/make it clearer?

No one will ever find that documentation :wink: #1035 is a great step in the right direction, but I honestly think that more needs to be done in-app if you expect users to understand Etcher's limits. Essentially, if it's not in the app itself and actively presented to the user, assume they won't find it.

lurch, 2017-01-19:
@WasabiFan but as I pointed out earlier, sometimes regular ISOs written to USB Flash drives _are_ bootable, and sometimes only hybrid ISOs are bootable - it just depends on the BIOS.
Do you think we should modify Etcher to detect if the image being written is an ISO file, and if it is then further detect if it's a regular or hybrid ISO, and display corresponding warning messages as appropriate?

I guess those warning messages might get annoying, so for advanced users who know what they're doing should we add an option to the settings to hide image-format warning dialogs?

WasabiFan, 2017-01-19:
You could try to detect it; if it's reliable, that would be nice. It could potentially be something basic which relies on known file names.

Another thought: while the user is waiting for the flash to finish, they're sitting there with just a bar to stare at... what if you put "tips" somewhere in the UI that cycle through? That would give you an easy way of presenting small bits of information without cluttering the UI.