balena-io/etcher #1003: Trojan file: Etcher-1.0.0-beta.17-win32-x64.exe

TeacherMac, 2017-01-07:
Windows 7 64bit
360 Total Security

[image]

jviotti, 2017-01-09:
I thought these alerts were long gone... The issue is that malware detection systems rely on how trusty a code signing certificate is to prevent these warnings. When we first released Etcher, every single malware detection software detected Etcher as malware, which went down over time as more and more users installed and trusted our certificate.

In the case of 360 Total Security, we might need to contact them directly for them to whitelist our certificate. I'll do that myself.

jviotti, 2017-01-09:
Sent! I'll update/close this issue once we get a response.

TeacherMac, 2017-01-09:
Glad to help. Thanks for the update.

Greg Trobaugh
310.734.8336

jviotti, 2017-03-24:
This was fixed already!

jviotti, 2017-03-24:
Please re-open if the warning keeps appearing for you!

closed 2017-03-24 by jviotti

sumitxbee, 2019-07-16:
[image]
Hi, I've downloaded Etcher software from its official website https://www.balena.io/etcher/, thought i'd scan it with Virustotal online tool.. look what I've found **"Trojan/Win32.GenKryptik"** This is latest version and I'm on windows 10 pro machine, I'm going to switch to linux and used it to burn an iso file. please find attached picture file for details.