balena-io/etcher #1005: Dependencies (Fedora 25)

gimerstedt, 2017-01-08:
- **Etcher version:**
[ake1@localhost Downloads]$ ls -l | grep Etcher-
-rw-rw-r--. 1 ake1 ake1   71186664 Jan  8 13:25 Etcher-1.0.0-beta.17-linux-x64.zip
-rwxr-xr-x. 1 ake1 ake1   71565312 Nov 28 11:33 Etcher-linux-x64.AppImage

- **Operating system and architecture:**
[ake1@localhost Downloads]$ uname -a
Linux localhost.localdomain 4.8.15-300.fc25.x86_64 #1 SMP Thu Dec 15 23:10:23 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

As i'm writing this i'm realizing this might be an issue with Electron but i'll leave it here anyway.

Starting the application without libXScrnSaver installed fails (provides libXss.so.1) with the following error message:

[ake1@localhost Downloads]$ ./Etcher-linux-x64.AppImage 
/tmp/.mount_MKbrYU/usr/bin/etcher: error while loading shared libraries: libXss.so.1: cannot open shared object file: No such file or directory

Installing libXScrnSaver solves the issue but it seems like an unwanted error for a statically compiled binary.

jviotti, 2017-01-09:
Hi @gimerstedt,

Thanks for reaching out! Sadly, `xscrnsaver` is a dependency of Electron (see https://github.com/electron/brightray/blob/master/brightray.gyp#L4), so not much we can do from our side apart from listing it in our documentation (which we do, see: https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#runtime-gnulinux-dependencies).

Another solution would be to create an RPM package for distributions such as Fedora, however I don't have any experience with that myself, but I'd happy to merge a contribution for it :)

closed 2017-01-09 by jviotti

kshitij10496, 2017-02-21:
I too am facing this problem while trying to use Etcher on Fedora 25. I hope this gets fixed soon.
@jviotti  Is there any work-through that might work for the time being ?
Apart from this, kudos on the excellent product.
I have been using it for the past 6 months.

lurch, 2017-02-21:
As mentioned in the first post, the current workaround is to install the libXScrnSaver package.

jviotti, 2017-02-21:
Yep, `libxscrnsaver` is an Electron dependency that you have to install. I don't have Fedora at hand right now, but from the top of my head, `sudo dnf install libscrnsaver` should do it :)

jviotti, 2017-02-21:
See https://github.com/resin-io/etcher/issues/835 for reference, there is a way we can mock this on the AppImage, since its not used by Electron at all (apparently)

lurch, 2017-02-21:
Ahh, thanks for finding that other issue @jviotti . Marking this as a duplicate of #835