balena-io/etcher #1006: Multiple USB Keys generates an error on flash

flobbb, 2017-01-08:
- **Etcher version:**
1.0.0 - beta.17

- **Operating system and architecture:**
windows 7 64 bt

- **Do you see any meaningful error information on DevTools?**

Get Error:Command Failed
Only occurs if more than one usb key attached, 
usb keys are same size and have no label, 
drive letters assigned when error occured E: and F:

re-run witn only one key, no problems

jviotti, 2017-01-09:
This is a great find. I'll try to reproduce on Windows 10 (hopefully the error happens there as well), otherwise I'll try to get access to a Windows 7 PC.

jviotti, 2017-01-09:
Hm, I've plugged in two drives that are the same brand, and same size, but wasn't able to reproduce. I guess this might be a Windows 7 specific issue. I'll try to test in that system soon.

flobbb, 2017-01-09:
Don't forget there were no volume labels on either ky  as well. Not sure if that contributes to how they are identified.

jviotti, 2017-01-11:
I double-checked without the volume labels, but still couldn't reproduce. I'm still trying to get access to a Windows 7 box. I'll hopefully send an update about this soon.

jviotti, 2017-03-20:
Hi @flobbb Do you still encounter the same issue on v1.0.0-beta.19?

jviotti, 2017-05-09:
The "Command Failed" message is a hint that this happened while spawning the old `elevate.exe` utility. We replaced that with our own native elevation mechanism, so either this issue is fixed, or 1.0.0-rc.5 will show the real error.

Please re-open of you are still experiencing this issue on 1.0.0-rc.5.

closed 2017-05-09 by jviotti