balena-io/etcher #1007: Make my disk unreadable

tyx, 2017-01-09:
- **Etcher version:** beta17
- **Operating system and architecture:** OSX 10.10
- **Do you see any meaningful error information on DevTools?** Nop, etcher said "it's done !"

Hello guys,

Thanks for this nice tools ! I just got a strange error after etcher completed the flash:
[image]

Yes it's french but it is saying that mac os x cannot read my disk. (even with disk utility) and when I try to boot on this disk, it failed.

Any idea ?

Thanks

Page-, 2017-01-10:
Hey @tyx, what image were you flashing? My best guess is that it was a linux image of some sort which contains a filesystem OSX isn't able to read, so it treats it as an uninitialised/corrupt drive when in reality it's fine

tyx, 2017-01-10:
Hello @Page- 

Yep you're right, I try to flash `ubuntu-16.04.1-desktop-amd64`. I guess I should give another try to boot on it ?

tyx, 2017-01-10:
Ok, my apologies. I was thinking I had to press `options` key to boot on usb, it was `alt`.

Sorry, and thanks again for this app ;)

closed 2017-01-10 by tyx

Page-, 2017-01-11:
No problem :)

EvergreenTree, 2018-01-02:
I also found the problem, and I know it's not an error by etcher but MacOS instead. But I wonder if it's possible to have a bootable Ubuntu device which is also readable on the MacOS system? or if it's possible to install the image into one partition of the USB while using the others as usual? Thanks.

jviotti, 2018-01-02:
> But I wonder if it's possible to have a bootable Ubuntu device which is also readable on the MacOS system?

The problem is that macOS simply doesn't have support for Linux file-systems. There might be some commercial software to install support for them though.

> if it's possible to install the image into one partition of the USB while using the others as usual? Thanks.

An image contains its own partition table, and Ubuntu images use many partitions in order to work, so its not possible to flash the whole set of partitions from the image into a single partition on your external drive without heavily modifying the image.