balena-io/etcher #1019: Issue while Flashing Zorin OS on a 4gb USB Drive

UselessPoint, 2017-01-12:
- **Etcher version:1.0.0.beta17**
- **Operating system and architecture: Windows 7 64bit**
- **Do you see any meaningful error information on DevTools?**

tracker.js:10 Error: Command failed: C:\Users\HP2510P\AppData\Local\Temp\9435.tmp.exe E:

    at ChildProcess.exithandler (child_process.js:218)
    at emitTwo (events.js:106)
    at ChildProcess.emit (events.js:191)
    at maybeClose (internal/child_process.js:877)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226)

tracker.js:10 Error: Command failed: C:\Users\HP2510P\AppData\Local\Temp\5EF4.tmp.exe E:

    at ChildProcess.exithandler (child_process.js:218)
    at emitTwo (events.js:106)
    at ChildProcess.emit (events.js:191)
    at maybeClose (internal/child_process.js:877)
    at Socket.<anonymous> (internal/child_process.js:334)
    at emitOne (events.js:96)
    at Socket.emit (events.js:188)
    at Pipe._handle.close [as _onclose] (net.js:493)

These error happened while trying to flash Zorin OS Core 12 in a 4gb simple USB drive Formatted on FAT32.
I even tried after formatting the device in different methods, but the results were the same.

jviotti, 2017-01-12:
Hey there!

Thanks for reporting. The issue seems to happen when trying to eject the drive. I believe the drive should be flashed completely, and you can prevent the error from happening on the first place by unticking "Eject after success" on the settings page (the top right cog icon).

As a way to get to the bottom of it, you can download the `RemoveDrive` tool (which we use internally for ejecting the drives) here: http://www.uwe-sieber.de/drivetools_e.html.

Here's the direct download link for your convenience: http://www.uwe-sieber.de/files/removedrive.zip.

You can download and decompress such tool, open `cmd.exe`, `cd` to the `x64` directory of the uncompressed directory, and run the tool on the drive letter:

```
cd path\to\RemoveDrive\x64
RemoveDrive.exe E:
```

Assuming `E:` is the drive letter of the problematic drive.

That should hopefully provide us with more information about the error

WasabiFan, 2017-02-08:
I just experienced what I believe to be the same issue.

[image]

Running `RemoveDrive.exe` manually, I get no output for either of the two drive letters that the physical drive is mapped to.

jviotti, 2017-02-08:
@WasabiFan Can you confirm that the two drive letters that Etcher report (F: and H:) are indeed associated with the drive and are accessible from My PC? It seems that in some cases, like when having some multi SD Card adapters, Etcher erroneously thinks the drive is mounted in all of them, therefore causing the "Invalid drive" error.

WasabiFan, 2017-02-08:
Those two drive letters were reported by Windows as soon as I plugged in the drive (two notifications would simultaneously appear). Although I didn't confirm in Partition Manager, I am fairly sure that it's being detected properly. It's worth noting that I _am_ using a cheap multi-reader. There is a FAT partition as well as another partition in a format that Windows does not recognize (and rightfully so); it freely allows me to open the FAT partition, but it says something to the effect of "this drive must be formatted before it can be used" of the other. I didn't try re-flashing the card to see if it would repro (I was rushed to leave the house yesterday) but I haven't seen this before when flashing very similar images.

WasabiFan, 2017-02-09:
Interesting... it looks like you may be right. There _is no H: drive_. On any disk.

[image]

lurch, 2017-02-09:
Hmmm, so maybe the fix is for `drivelist` (or the scripts that it contains) to be modified so that they only report 'real' drives (volumes) in the `mountpoints` array? (and/or drive-letters which are only currently mounted?)  And then `RemoveDrive.exe` wouldn't be given invalid drive letters in the first place.
Looks like you might have to get your hands dirty with more VBScript @jviotti ? :-(
(And IIRC this also sounds like the issue that @halebr reported?)

jviotti, 2017-02-09:
At least we know what's going on. I have a USB hub in which I might be able to reproduce this, and if so, I can research how to modify the VBScript to catch this.

jviotti, 2017-02-09:
I tried to reproduce this with an [Anker USB 3.0 Hub](https://www.amazon.com/Upgraded-Anker-SuperSpeed-Including-Charging/dp/B005NGQWL2/ref=sr_1_5?ie=UTF8&qid=1486655288&sr=8-5&keywords=anker+usb+hub) but it seems to behave correctly for me (no more than one drive letter is assigned).

I'll try flashing an ev3dev image and see if it makes any difference.

jviotti, 2017-02-09:
Yeah, I'm able to reproduce with drives that have more than one partition!

jviotti, 2017-02-09:
OK, I think I found a good way to filter these out. Turns out that on these kinds of partitions, the partition size is set to `null` in the `Win32_DiskPartition` class. I edited the drive detection VBScript to filter out mount points of partitions whose size is `null`: https://github.com/resin-io-modules/drivelist/blob/972c3e52c16a9bb209727e2a6880fa748ad9c7da/scripts/win32.bat

@WasabiFan Please let me know if this fixes the issue for you!

lurch, 2017-02-09:
> I confirm it doesn't happen when not using the hub.

That's totally bizarre, what difference (and how?) should just using a regular USB hub make to drive-detection??

jviotti, 2017-02-09:
> That's totally bizarre, what difference (and how?) should just using a regular USB hub make to drive-detection??

I know. I've seen so many bizarre Windows things in Etcher that I'm not really surprised anymore :P

WasabiFan, 2017-02-09:
Cool, I'll test it out tomorrow!

> That's totally bizarre, what difference (and how?) should just using a regular USB hub make to drive-detection??
> I know. I've seen so many bizarre Windows things in Etcher that I'm not really surprised anymore :P

I guess this is one of the artifacts of maintaining complete API compatibility for over two decades :laughing:

jviotti, 2017-02-13:
@WasabiFan Did you get a change to test this out?

WasabiFan, 2017-02-13:
"tomorrow" for me means "next week" :wink: Let's see if I remember tonight... I'll try!

WasabiFan, 2017-02-14:
Looks fixed to me!

[image]

jviotti, 2017-02-14:
lol amazing! Thanks a lot for the testing @WasabiFan ! I'll get this into `master` soon.

closed 2017-02-16 by jviotti