balena-io/etcher #1024: Still presented with "Connect a drive" after recovering broken drive

codeinabox, 2017-01-13:
- **Etcher version:**
v1.0.0-beta.17
- **Operating system and architecture:**
OS X Safari 10.12.2 
- **Do you see any meaningful error information on DevTools?**
[image]

I followed all the instructions to recover my broken drive (see below) but still get presented with Connect a drive

```
diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *500.1 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage Macintosh HD            498.5 GB   disk0s2
   3:                 Apple_Boot Recovery HD             1.3 GB     disk0s3

/dev/disk1 (internal, virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                            ]]]]]]][=======-]lo... +498.1 GB   disk1
                                 Logical Volume on disk0s2
                                 CB1CF191-5B28-44C9-95EC-72B2EECF1D23
                                 Unlocked Encrypted

/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *31.0 GB    disk2
   1:                        EFI EFI                     209.7 MB   disk2s1

diskutil eraseDisk free UNTITLED /dev/disk2
Started erase on disk2
Unmounting disk
Creating the partition map
Waiting for partitions to activate
Finished erase on disk2
```

```

lurch, 2017-01-13:
Hopefully @jviotti will be able to help with this one (I don't have a Mac myself).

jviotti, 2017-01-13:
That is the strangest drive name I've ever seen :) No doubt `drivelist` parser is getting confused about it. It should be easy to fix though, let me try to reproduce here...

codeinabox, 2017-01-13:
Ha I don't know how my drive ended up with that name but renaming it to something alphanumeric fixed the issue. Thank you for the quick response.

jviotti, 2017-01-13:
Ah, turns out this was already fixed by https://github.com/resin-io-modules/drivelist/pull/127, which didn't make it to v1.0.0-beta.17. I plan to release a next version really soon (probably today, on Monday at worst) with the fix.

Renaming it to something else should get you up and running in the mid-time :D

closed 2017-01-13 by jviotti

lurch, 2017-01-13:
Hahaha, I didn't even notice the peculiar drive name, I just skimmed over that section. Doh! ;-)