balena-io/etcher #1025: Error: 'MySQL' is not recognised as an internal or external command

RobertLutken, 2017-01-13:
[image]
- 1.0.0-beta 17
- Windows 10 Pro 64-bit

lurch, 2017-01-13:
We've previously seen something very similar to this on Gitter https://gitter.im/resin-io/etcher?at=582a14e945c9e3eb4339d229
I didn't realise we didn't have a Github issue for it yet.

@jviotti Did we ever get any ideas what might be causing this?

@RobertLutken Would you be willing to help us debug this? Do you currently have MySQL installed on your PC, or have you had it installed previously?

RobertLutken, 2017-01-13:
@lurch I do indeed have MySQL installed, well MySQL workbench. 
A little bit of digging seems to refer to "A key is a string specified in the Add method that uniquely identifies a specific member of a collection." [MSDN](https://msdn.microsoft.com/en-us/library/aa232626(v=vs.60).aspx)
I'll have a dig around and try setup some different environment variables. 
Happy to help!

jviotti, 2017-01-13:
Hey there @RobertLutken !

Thanks a lot for reporting! We've very recently fixed the `Microsoft VBScript runtime error: This key is already associated with an
element of this collection` error (see https://github.com/resin-io-modules/drivelist/pull/130).

Regarding the MySQL error, I think its simply output coming from your environment every time you type a command, and should be safely ignored by our drive detection parser (see https://github.com/resin-io-modules/drivelist/pull/99).

So just to confirm the issue is gone, can you try the following `bat` scripts on your computer, and report back the results?

- After the fix: https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat
- Before the fix: https://github.com/resin-io-modules/drivelist/blob/cdfc551e5fbe874a09cc947589480548bd938712/scripts/win32.bat

Also, can you open `cmd.exe` and try typing some commands on it? I want to check if those MySQL warnings you get get printed every time you write a command there, independently on what they are. You can try `wmic diskdrive list brief` for instance.

daturach, 2017-01-26:
Hello

Same error message, in French this time  :-)

Etcher-1.0.0-beta.18-win32-x64
Windows 7, 64 bits Edition Familiale Premium

Anything I can do to help?

jviotti, 2017-01-26:
Hi @daturach ,

Can you try running the `.bat` scripts I linked to in my previous comment?

Also, what happens if you *any* command on `cmd.exe`? Do you get the MySQL warning as well?

jviotti, 2017-01-26:
For this user, it seems to happen everytime a `cmd.exe` window is opened. Maybe its the same thing?

https://i.stack.imgur.com/e8b8q.png

jviotti, 2017-01-26:
Hi @daturach ,

Can you also paste a screenshot of the error dialog?

daturach, 2017-01-27:
Hello @jviotti 

Thanks for your quick answer.
This is the error message I get when starting etcher:
[image]

As a side note, this dialog box is not even supporting accented chars...:-(

Running win32.bat before the fix:
[image]

This MySQL error message is displayed only once. Other commands run into the same cmd.exe does not show this message anymore.

Running win32.bat after the fix displays the list of devices as above but without the error message if run in the same session.

Command `wmic diskdrive list brief` shows list of devices but never displays the error message.

Hope that helps!
Thanks

jviotti, 2017-01-27:
I think I have a theory about this. While our YAML parser will ignore invalid keys (https://github.com/resin-io-modules/drivelist/pull/99), the JavaScript code that executes the `.bat` script as a child process will blow up with an error if the `stderr` from the child process is not empty:

```js
// See https://github.com/resin-io-modules/drivelist/blob/master/lib/scripts.js
if (stderr.trim().length) {
    return callback(new Error(stderr));
}
```

This seems to be what's happening given that the error contains *both* MySQL warning instead of just one (if the YAML parser would have been complaining about this, it would have complained about the first warning only, and we would have a YAML related stack trace).

I'm not very sure about what to do here. We can of course ignore `stderr` completely, but we might be hiding real errors accidentally, which is definitely not good. 

@lurch What do you think?

jviotti, 2017-01-27:
In the mid-time, in order to validate this theory, I'll produce a custom Windows build that logs a warning instead of throwing an error if the child process includes `stderr` output.

lurch, 2017-01-27:
Hmmm, given that we've _only_ ever seen this error for MySQL (for a still-unknown reason) but for multiple times, I'd be tempted to filter out only any stderr messages starting with `'MySQL' ` which matches both the English and French errors above.

Also, should there be a separate issue for the French error-dialog not correctly showing accented characters?

jviotti, 2017-01-27:
> Hmmm, given that we've only ever seen this error for MySQL (for a still-unknown reason)

It seems to be a MySQL `%PATH%` configuration issue. Try googling the warning, the error seems to be well known. The reason why we see it here is that when you execute `cmd.exe`, it tries to run some MySQL command by default, and thus we get it on every new `cmd.exe` instance, including the one that triggers the drive detection script.

> I'd be tempted to filter out only any stderr messages starting with 'MySQL' which matches both the English and French errors above.

Yeah, maybe, although the fact that a similar error might be reported in the future makes me feel a bit uncomfortable.

I have an alternative solution: What about allowing the drivelist client to pass a function to handler `stderr` from the scripts? We can throw by default, but Etcher could silently print them in DevTools (and we'll see it if a user has any problems and we ask him to send us the DevTools output), and send them to TrackJS.

A similar approach would be to allow some option in drivelist to print `stderr` output to `stderr` instead of throwing (basically proxying the output). Since in Etcher we send any `stderr` output to TrackJS, we accomplish the same thing automatically.

lurch, 2017-01-27:
> It seems to be a MySQL %PATH% configuration issue. Try googling the warning, the error seems to be well known.

I googled, and as expected the first few results are from people actually trying to run `mysql` (which we aren't). But then I found https://github.com/cmderdev/cmder/issues/443 which has lots of great info, especially https://github.com/cmderdev/cmder/issues/443#issuecomment-150202124 which suggests that using double-quotes is the way to fix this problem, i.e. maybe we simply need to change https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat#L6 to
```
set "PATH=%PATH%;%SYSTEMROOT%\System32"
```
:-D

daturach, 2017-01-27:
@lurch, jviotti

Running win32.bat with the PATH between "" does not produce any MySQL error.

jviotti, 2017-01-27:
Amazing discovery! This might indeed be the reason! I'm uploading a custom build with these changes and we'll see how it goes :)

jviotti, 2017-01-27:
@daturach Can you try the following custom build as well just to confirm? https://drive.google.com/open?id=0B7tkbonGU-RyNTRxdlBhRVgwQnc.

daturach, 2017-01-27:
Thanks
What's the process? unzip and run Etcher.exe?
I get the following:
[image]

jviotti, 2017-01-27:
Let me try here. I might be messed something up when editing the files straight in the asar archive.

jviotti, 2017-01-27:
Try this one: https://drive.google.com/open?id=0B7tkbonGU-RyUnJxbm5WeVpSM1U. Indeed messed up the previous one, sorry about that :)

daturach, 2017-01-27:
Ok thanks; it's looking fine. MySQL error message disappeared.
Next week, I will start flashing some images. Great work!

jviotti, 2017-01-27:
Woohoo! Thanks a lot for the help on this issue. I'll send a patch to apply this fix!

jviotti, 2017-01-27:
All the credit to @lurch and his amazing observations of course \o/

alexandrosm, 2017-01-28:
should we just be double-quoting everything-in-general and get it over
with? it seems so many issues end with "use double quotes"

closed 2017-01-30 by jviotti

jviotti, 2017-01-30:
I'll definitely check if we can use double quotes elsewhere. At least in the Windows script, everything else is quoted already.

jviotti, 2017-01-30:
I've sent two PRs addressing issues highlighted by `shellcheck`, a shell script linter, for both the GNU/Linux and OS X scripts:

- https://github.com/resin-io-modules/drivelist/pull/134
- https://github.com/resin-io-modules/drivelist/pull/135

This tool, among other improvements, highlighted all the places where we should quote for safety purposes.