balena-io/etcher #1031: Window fails to load and remains blank after upgrading Etcher

WasabiFan, 2017-01-17:
- **Etcher version:** Beta 18 (upgraded from 17)
- **Operating system and architecture:** Windows 10 x64
- **Do you see any meaningful error information on DevTools?** See below.

After clicking the final button in the Etcher installer with the "run Etcher" box checked, the application opened without any content other than some upper-right buttons and the bottom text.

[image]

I had closed the previous version of Etcher before starting the installer for the new one. The dev tools look like this:

[image]

The error text is:

```
Uncaught Error: Cannot find module 'etcher-image-stream'
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:4631 Uncaught Error: [$injector:modulerr] Failed to instantiate module Etcher due to:
Error: [$injector:nomod] Module 'Etcher' is not available! You either misspelled the module name or forgot to load it. If registering a module ensure that you specify the dependencies as the second argument.
http://errors.angularjs.org/1.5.7/$injector/nomod?p0=Etcher
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:68:12
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:2075:17
    at ensure (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:1999:38)
    at module (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:2073:14)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:4608:22
    at forEach (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:321:20)
    at loadModules (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:4592:5)
    at createInjector (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:4514:19)
    at doBootstrap (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:1751:20)
    at bootstrap (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:1772:12)
http://errors.angularjs.org/1.5.7/$injector/modulerr?p0=Etcher&p1=Error%3A%…r%5Cresources%5Capp.asar%5Cnode_modules%5Cangular%5Cangular.js%3A1772%3A12)
```

The window still fails to load if I close the app and open it again. It's worth noting that I had opened the Etcher installation directory in File Explorer (for #1030) during the installation, but did not open any files or otherwise prompt it to try loading them. There were no errors that the installer reported.

WasabiFan, 2017-01-17:
I can confirm that this isn't localized to one PC; upgrading on another one exhibits the same behavior. Although it looks like you've probably already figured it out.

jviotti, 2017-01-17:
This is really weird. I can reproduce the problem on Windows indeed, but not on OS X nor Windows. I took off the Windows builds from S3 for now. I'm investigating if why this dependency has not been included in the package.

WasabiFan, 2017-01-17:
Are the zip builds functional? (I can't test because they've been removed from S3 😆)

jviotti, 2017-01-17:
Hi @WasabiFan ,

Looks not. I took off the bad builds from S3 so users get pointed to v1.0.0-beta.17 while we fix the issue. 

I'm tracking it down... Seems like an npm issue on my Windows build machine. I'll keep you updated on this ticket!

jviotti, 2017-01-17:
So the issue seems to be that npm insists on installing a platform specific development dependency for some reason, causing a certain package (`etcher-image-stream`) to be removed.

I've downgraded npm without luck, however discovered that the installation works fine if I call npm from MinGW instead of from `cmd.exe`, which makes no sense at all :/

I'll build from MinGW and release so we have everything up and running and will continue to investigate the issue.

jviotti, 2017-01-17:
I'm closing this issue. Seems like an npm Windows glitch. I've released MinGW builds that work fine (at least here), and will be putting extra effort to finally move releases to the CI servers for the next one.

closed 2017-01-17 by jviotti