balena-io/etcher #1032: Image name extends beyond tooltip when hovering over text

WasabiFan, 2017-01-17:
- **Etcher version:** Beta 17
- **Operating system and architecture:** Windows 10 x64
- **Do you see any meaningful error information on DevTools?** N/A

When hovering over the name of an image that's long and does not have natural separators, the text extends beyond the tooltip box.

[image]

This can be fixed by adding `word-wrap: break-word` to `.tooltip-inner`.

I can't test this in the newest version of Etcher right now (see #1031) so this may have already been resolved... I recall some talk of replacing this with a modal.

jviotti, 2017-01-17:
Thanks a lot for reporting! The tooltip is still present on the latest version. We have some plans to provide a new UX for this, but we should definitely fix it in the mid time.

Shou, 2017-01-17:
This should do it.

closed 2017-01-17 by jviotti