balena-io/etcher #1035: Warn against flashing Windows images

jviotti, 2017-01-19:
A lot of users are constantly trying to flash Windows images, which will not work on Etcher given that Windows images require special treatment.

As a solution, let's develop a regex (or collection of regexes) based on all the Windows images file names [available from MSDN](https://msdn.microsoft.com/subscriptions/securedownloads/?#searchTerm=windows&ProductFamilyId=0&Languages=en&FileExtensions=.img,.iso&PageSize=10&PageIndex=0&FileId=0) and use that to show a warning.

See https://github.com/resin-io/etcher/issues/210 for the issue tracking Windows support.

lurch, 2017-01-19:
Is there a version of that page (or similar) that's viewable without needing a Microsoft login?

jviotti, 2017-01-20:
I don't think so :(

alexandrosm, 2017-01-21:
Can we just show a simple warning if the image has the "windows" word in
it? If some other equivalently simple pattern exists in the names, maybe
let's use that, but keep it overall very simple, rather than some
aggressive regex?

fhdk, 2017-01-23:
> Is there a version of that page (or similar) that's viewable without needing a Microsoft login?
@lurch I can generate a copy for you

closed 2017-02-24 by jviotti