balena-io/etcher #1042: Use /dev/rdiskN instead of /dev/diskN on macOS

Jamesits, 2017-01-24:
- **Etcher version:**
1.0.0-beta18
- **Operating system and architecture:**
macOS (general)

`/dev/diskN` is buffered while `/dev/rdiskN` is unbuffered. Accessing raw disk makes writing a lot faster in certain cases.

lurch, 2017-01-24:
I've seen this same suggestion being made on the RPI documentation repo quite a few times (see https://github.com/raspberrypi/documentation/pull/192 and the other issues it links too), but not being a Mac user I never quite understood the difference ;-)

lurch, 2017-01-24:
Is the `If this command fails, try using disk instead of rdisk` comment on https://www.raspberrypi.org/documentation/installation/installing-images/mac.md something we need to be careful of? Are there reasons why one method might fail when the other method might succeed?

Jamesits, 2017-01-24:
@lurch `/dev/diskN` is a 4K buffered device while `/dev/rdiskN` is a raw unbuffered one. When writing to disks not optimized for 4K r/w, there may be a significant performance drop when using buffered device. (Some user have noticed a 10x speed drop; my personal tests show it largely depends on the card.) Also in the use cases of Etcher we really don't need buffer. 

I've never encountered the situation when I'm unable to write to `/dev/rdiskN`, and Googling also give null result. So the fallback option on RasPi doc may be just for compatibility with the Linux world.

jviotti, 2017-01-24:
Hey there!

We present the `/dev/diskN` form for user friendliness purpose, but we use the raw device (`/dev/rdiskN`) internally (see https://github.com/resin-io/etcher/blob/master/lib/cli/writer.js#L67). The `raw` property comes from https://github.com/resin-io-modules/drivelist, which is set to `/dev/rdiskN` on OS X.

closed 2017-01-24 by jviotti