balena-io/etcher #1053: Not executing in manjaro-openrc

Mystified1234, 2017-01-25:
Find below details of the build, taken from Git from https://github.com/resin-io/etcher.git & then cloned into my etcher directory. From threre built with makepkg. 
 I'm unable to locate icons to execute or even run from terminal
makepkg
==> Making package: etcher 1.0.0_beta.18-1 (Thu Jan 26 05:14:25 AEDT 2017)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found etcher.in
  -> Found Etcher.desktop
  -> Found etcher.png
  -> Downloading Etcher-1.0.0-beta.18-linux-x64.zip...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 53.3M  100 53.3M    0     0   285k      0  0:03:11  0:03:11 --:--:--  314k
==> Validating source files with md5sums...
    etcher.in ... Passed
    Etcher.desktop ... Passed
    etcher.png ... Passed
==> Validating source_x86_64 files with md5sums...
    Etcher-1.0.0-beta.18-linux-x64.zip ... Passed
==> Extracting sources...
  -> Extracting Etcher-1.0.0-beta.18-linux-x64.zip with bsdtar
==> Starting build()...
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Compressing man and info pages...
==> Checking for packaging issue...
==> Creating package "etcher"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: etcher 1.0.0_beta.18-1 
mystified1234@gmail.com

jviotti, 2017-01-26:
Hi @Mystified1234 ,

I don't understand very well what the issue is. Did you get Etcher running locally first (using `make electron-develop`)?

What do you get when running the remaining package?

jviotti, 2017-03-24:
I'm closing this, since the issue is how to package an application correctly using `makepkg` on Manjaro rather an an Etcher issue. I'm afraid I can't help with the former.

closed 2017-03-24 by jviotti