balena-io/etcher #1054: error Starting Etcher on Windows (Beta18)

timjstone, 2017-01-26:
- **Etcher version: Beta18**
- **Operating system and architecture: Windows 10 **

when starting Etcher, i get the following, and Etcher doesn't start

 [Window Title] Error

[Main Instruction]
Command failed: C:\Users\timjs\AppData\Local\Temp\B39C.tmp.bat

[Content]
Error: Command failed: C:\Users\timjs\AppData\Local\Temp\B39C.tmp.bat

    at ChildProcess.exithandler (child_process.js:218:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:877:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)

[OK]

jviotti, 2017-01-26:
Looks like an issue with our drive scanner script in your PC. Can you try running the following `.bat` file and report back what you get? https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat

timjstone, 2017-01-26:
output as follows:

device: "\\\\.\\PHYSICALDRIVE1"
description: "SanDisk SSD U110 16GB"
size: 16006394880
raw: "\\\\.\\PHYSICALDRIVE1"
system: True
protected: False
mountpoints:
  - path: "D:"

device: "\\\\.\\PHYSICALDRIVE2"
description: "SanDisk Ultra Fit USB Device"
size: 124218178560
raw: "\\\\.\\PHYSICALDRIVE2"
system: False
protected: False
mountpoints:
  - path: "H:"

device: "\\\\.\\PHYSICALDRIVE0"
description: "Samsung SSD 850 PRO 512GB"
size: 512105932800
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

jviotti, 2017-01-26:
Can you try running `echo %ERRORLEVEL%` **right** after running the `.bat` command?

timjstone, 2017-01-26:
returns 0

I tested both by running it from the command line and including the the cmd file. both returned 0

Tim

jviotti, 2017-01-27:
Hi @timjstone ,

That's weird. Your error comes from a `.bat` file in a temporary location. That location is created by Electron itself when calling `child_process.execFile` in order to extract the executable out of the asar.

The only `.bat` script we have is the drive scanning script, so it must be related to it, however running it independently seems to work fine for you.

I have two theories:

- The `win32.bat` script fails depending on a certain drive that is plugged to your computer

To validate this theory, try running Etcher and the win32.bat with different combinations of drives plugged in, and see if you can reproduce the failure only when a certain drive is there.

- I pointed you to the `win32.bat` from the `master` branch, which has some fixes on top of what was released on v1.0.0-beta.18

Here is the `win32.bat` that was shipped with v1.0.0-beta.18: https://github.com/resin-io-modules/drivelist/blob/5.0.6/scripts/win32.bat. Can you try running that one instead?

Let me know how that goes!

timjstone, 2017-01-27:
i have checked the script and yes it is identical to the temporary bat files created and run. the file you pointed to me works the same (i.e. no errors when run manually)

i have disconnected all the drives i can (the laptop i am using has two built in drives both SSDs (one on a M2 interface)

running as an administrator still fails (so doesn't seem to be permissions related). the error seems to imply that it is having a problem closing the child process.

for reference i am using the newest Windows 10 preview build 15014

lurch, 2017-01-27:
@jviotti Would it be easy to add some custom logging here, to determine whether the .bat file can't be executed at all, or if t started executing and failed mid-way through? (and if so, at which point it failed?)

jviotti, 2017-01-27:
@lurch If the script didn't complete successfully we don't inspect its `stdout` at all, so maybe the extra logging won't do much difference? 

Most of these errors hide their root cause as subtleties in the error message. I wonder if by copy pasting the error, some of this information got lost.

@timjstone Do you mind posting a screenshot of the error window?

timjstone, 2017-01-27:
[image]

jviotti, 2017-01-27:
Weird. Maybe this is also failing because of stderr output, like https://github.com/resin-io/etcher/issues/1025? @timjstone Do you mind trying this custom build? https://drive.google.com/open?id=0B7tkbonGU-RyNTRxdlBhRVgwQnc

timjstone, 2017-01-28:
[image]

unfortuately, no change

Tim

jviotti, 2017-03-03:
I'm honestly clueless about this one. I sent a PR to `drivelist` to append the exit code to the error message: https://github.com/resin-io-modules/drivelist/pull/149. I'll try to increase logging in some other areas and then maybe we can try again with another custom build.

jviotti, 2017-03-06:
@timjstone Can you run the following custom build:  
https://drive.google.com/open?id=0B7tkbonGU-RyQ0tHYTVDbkRCMEU?

Once the error happens, click OK, and then once Etcher opens up, press Ctrl+Alt+I to open DevTools, go to the Console pane, and copy paste what you see there here.

timjstone, 2017-03-06:
Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
tracker.js:10 Mon Mar 06 2017 19:21:45 GMT+0000 (GMT Standard Time) Application start
tracker.js:10 Mon Mar 06 2017 19:21:45 GMT+0000 (GMT Standard Time) Checking for updates
tracker.js:10 drivelist stderr:  +0ms
tracker.js:10 drivelist stdout: device: "\\\\.\\PHYSICALDRIVE1"
description: "SanDisk SSD U110 16GB"
size: 16006394880
raw: "\\\\.\\PHYSICALDRIVE1"
system: True
protected: False
mountpoints:
  - path: "D:"

device: "\\\\.\\PHYSICALDRIVE2"
description: "SanDisk Ultra Fit USB Device"
size: 124218178560
raw: "\\\\.\\PHYSICALDRIVE2"
system: False
protected: False
mountpoints:
  - path: "H:"

device: "\\\\.\\PHYSICALDRIVE0"
description: "Samsung SSD 850 PRO 512GB"
size: 512105932800
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE3"
description: "SDHC Card"
size: 15924142080
raw: "\\\\.\\PHYSICALDRIVE3"
system: False
protected: False
mountpoints:
  - path: "E:"

 +0ms
tracker.js:10 Error: Command failed: C:\Users\timjs\AppData\Local\Temp\6926.tmp.bat
 (code 1)
    at ChildProcess.exithandler (child_process.js:218)
    at emitTwo (events.js:106)
    at ChildProcess.emit (events.js:191)
    at maybeClose (internal/child_process.js:877)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226)a.(anonymous function) @ tracker.js:10

timjstone, 2017-03-06:
for info, when i try to select a drive in step 2, it says no removable drives.

Tim

jviotti, 2017-03-06:
Hi Tim,

Thank you very much for the fast response, that's really appreciated!

So it looks like the drive detection shell script exits with code 1, even though it outputs valid stuff to `stdout`. The question is what triggers the exit code 1.

I'm pretty sure that if it was a VBScript error, stderr would include at least a useful error message pointing the VBScript problematic line, which makes me think this comes from the batch wrapper:

```sh
set "PATH=%PATH%;%SYSTEMROOT%\System32"
cscript //nologo "%~f0?.wsf"
exit /b %ERRORLEVEL%
```

Here comes the nice part. If a command fails, it sets `%ERRORLEVEL%` to some number greater than 0, **however**, if after an error, a command succeeds, then `%ERRORLEVEL%` is NOT updated back to 0, which means that the exit code 1 could be coming from something else happening before this.

@timjstone Can you confirm that the output you pasted above includes all your connected drives?

> for info, when i try to select a drive in step 2, it says no removable drives.

Yeah, that's expected. The drive scanner is failing, so you'll not be able to select any drives.

jviotti, 2017-03-06:
In order to confirm this theory, I'll create a new custom build that resets the error level at the beginning of the batch script.

jviotti, 2017-03-06:
@timjstone Here's a new custom build containing the patch described above: https://drive.google.com/open?id=0B7tkbonGU-Ryb1hybnVvYlYtQ1k

Let us know how it goes, and thank you very much for you help debugging this issue!

timjstone, 2017-03-06:
unfortunately, the same problem

Tim

jviotti, 2017-03-06:
@timjstone Is it the same code number in the error description? E.g: (code 1)

timjstone, 2017-03-07:
> @timjstone <https://github.com/timjstone> Do you mind trying this little
> program?
> https://drive.google.com/file/d/0B7tkbonGU-RyT0Z0Qm8zOHFuVk0/view?usp=sharing
> You should execute it from cmd.exe or PowerShell by running the run.bat
> script in there.
>
> Its basically Node v6.1.0 (which we use on Etcher) calling drivelist. If
> this fails with the same error message, then we're in good track, since
> we'd have a very small reproducible example. If this doesn't fail, then I
> honestly have no clue about what might be going on, so let's hope it fails
> :)

yes we get the same error

or: Command failed: C:\Program Files
(x86)\Etcher\node_modules\drivelist\scripts\win32.bat
 (code 1)
    at ChildProcess.exithandler (child_process.js:202:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)

pfmscc, 2017-03-09:
C:\Users\xxxxx\tmp>win32.bat
device: "\\\\.\\PHYSICALDRIVE0"
description: "SAMSUNG MZ7TD256HAFV-000 SCSI Disk Device"
size: 256052966400
raw: "\\\\.\\PHYSICALDRIVE0"
system: True
protected: False
mountpoints:
  - path: "C:"

device: "\\\\.\\PHYSICALDRIVE1"
description: "Kingston DataTraveler 3.0 USB Device"
size: 15471751680
raw: "\\\\.\\PHYSICALDRIVE1"
system: False
protected: False
mountpoints:
  - path: "E:"

C:\Users\xxxxx\tmp>echo %ERRORLEVEL%
0

jviotti, 2017-03-09:
Hi @timjstone @pfmscc ,

Do you mind trying my suggestion from https://github.com/resin-io/etcher/issues/1054#issuecomment-284534308? 

The above experiment will validate if this is an issue on Electron or not.

pfmscc, 2017-03-09:
No failure.

C:\Users\xxxxx\tmp\drivelist-test\drivelist-test>run.bat
[
  {
    "device": "\\\\.\\PHYSICALDRIVE0",
    "description": "SAMSUNG MZ7TD256HAFV-000 SCSI Disk Device",
    "size": 256052966400,
    "raw": "\\\\.\\PHYSICALDRIVE0",
    "system": true,
    "protected": false,
    "mountpoints": [
      {
        "path": "C:"
      }
    ]
  },
  {
    "device": "\\\\.\\PHYSICALDRIVE1",
    "description": "Kingston DataTraveler 3.0 USB Device",
    "size": 15471751680,
    "raw": "\\\\.\\PHYSICALDRIVE1",
    "system": false,
    "protected": false,
    "mountpoints": [
      {
        "path": "E:"
      }
    ]
  }
]

d4rk22, 2017-03-17:
Is the *.tmp.bat error the same one as the win32.bat error? I recently recommended Ether to my coworkers only to have a lot of them get stuck with this error.

[image]

jviotti, 2017-03-17:
Hi @d4rk22 @pfmscc ,

They are the same. The `win32.bat` happens if the application is not packaged as an asar (extracting them from the asar for execution purposes makes the file get a temporary name).

> I recently recommended Ether to my coworkers only to have a lot of them get stuck with this error. They are all running Windows 10 x64.

That's very weird. The Etcher team is struggling a lot with this because we can't reproduce in any of our computers (we're also running Windows 10 x64). Is there any special Windows configuration your co-workers have that could be triggering this issue?

Do you mind trying the things I asked on this thread, and confirm that you get the same results? I'm starting to think there is en Electron bug here, but it'd be impossible to debug without getting my hands on a computer that reproduces the bug, building Electron, and playing with it myself.

timjstone, 2017-03-17:
a thought. i am running the insiders preview of windows 10 x64. could this be the issue?

Tim

jviotti, 2017-03-17:
Hi @timjstone ,

Yeah, that would make sense. What insider preview are you running? Could you link me to an image that gives me the closest possible environment to yours? (I'm an insider member already)

jviotti, 2017-03-17:
BTW, can someone clarify if this error happens as soon as you start Etcher, or as soon as you click "Flash"?

pfmscc, 2017-03-17:
I am pretty sure it was on startup.

Once you click to get rid of the error pop-up, the rest of the etcher window comes up but the middle icon does nothing.

timjstone, 2017-03-17:
i am running todays 15060 (i am on the fast ring) - the error still occurs.

Etcher did work in the past (but i can't remember on which build i was on, but it was a long time ago)

jviotti, 2017-03-18:
Cool, I'll install it and see how it goes next week. 

As an alternative, I'm thinking any of you that can reproduce the problem could turn their Windows installation to a virtual machine that the Etcher team can use to debug and fix. I found this article that describes how to do it: https://www.howtogeek.com/213145/how-to%C2%A0convert-a-physical-windows-or-linux-pc-to-a-virtual-machine/. I'm pretty sure these tools should have some way to avoid copying sensitive information (like files, users, etc) but still preserve the Windows installation. How do you feel about that? It might be asking too much, so feel free to ignore me :)

jviotti, 2017-03-18:
@timjstone On what insider level are you?

[image]

timjstone, 2017-03-18:
i am on the fast ring - i like to live dangerously!

lurch, 2017-03-20:
Are these standalone PCs, or are they on some kind of corporate network?

timjstone, 2017-03-20:
standalone (not part of a domain)

tim

lurch, 2017-03-20:
Does the same error occur if you boot the PCs in Safe Mode??

timjstone, 2017-03-24:
just to update, i have installed the win32 version for a test (instead of the x64 version installed previously) and all seems to work. so the problem seems to be with the x64 version .

jviotti, 2017-03-24:
Thanks for the info @timjstone. I managed to setup my Windows 10 x64 machine into the fast insider level, but I still can't reproduce the issue :(

ShalokShalom, 2017-03-31:
Happens to me in Win 7 64 when i open the app

jviotti, 2017-04-17:
Damn, I truly wish I could reproduce this :/ The more I investigate it the more it seems the issue comes from Electron, but I can't know for sure without being able to actively reproduce.

@ShalokShalom Was the Windows 7 installation customized in any way that I could try mimicking here?

Also, does it happen if you open the app while there are no drives plugged in? Could it be that certain drives trigger the issue?

ShalokShalom, 2017-04-18:
Sorry, this happened on a consumer device, about which i already forgot xD
All what i was aware, is that it was a pretty untouched, unmodified version of Windows 7.

benji1000, 2017-04-21:
Hello,

I have the same error as described ([screenshot](http://i.imgur.com/vUnF79C.png)).

Here's what I get when I run the .bat script that @jviotti recommended:

`Erreur CScript: L'accès à Windows Script Host est désactivé sur cette machine. Contactez votre administrateur système pour plus d'informations.`

It basically says that Windows Script Host is disabled (I disabled it some time ago for security reasons).

Is there any way you could avoid using WSH for whatever you're trying to do in the .bat script (and in Etcher, if the two issues are related)?

Thanks!

lurch, 2017-04-21:
There _is_ a plan to replace `drivelist` with a native NodeJS module, but until then I'm afraid re-enabling WSH is your only option.
(Etcher uses VBscript on Windows to gather information about what drives are available on your system, how big they are, etc. etc.)

benji1000, 2017-04-21:
@lurch Maybe you could test the absence of WSH and throw a specific error message in case that happens :) anyway, thank you for your answer.

lurch, 2017-04-22:
Sounds like a good suggestion :+1: 

@jviotti I did a bit of research, and it seems the only way we can tell if WSH is disabled is by reading the registry https://technet.microsoft.com/en-us/library/ee198684.aspx
Also found a native module that allows registry access https://www.npmjs.com/package/windows-registry
Happy to put a PR together for this next week, if you think it's worth it.

P.S. Ooh, just spotted that that NPM module also has "Launching a Process as an Admin" functionality ;-)

walchko, 2017-05-02:
I am getting this error too

[image]

[image]

lurch, 2017-05-03:
In a recent team-meeting, @jhermsmeier suggested that perhaps a virus-checker was responsible for #1108 - perhaps the same is also true for this issue?
I.e. maybe the `XXXX.tmp.bat` is only partially written out to disk (due to delays caused by a virus-scanner), before Etcher tries running it? I wonder if a workaround might be waiting until the `XXXX.tmp.bat` extracted from the `.asar` has the size we expect it to, before we try executing it?

jhermsmeier, 2017-05-03:
@lurch I think antivirus software will just acquire exclusive access to those files, not stall writes, which *might* be the reason Etcher can't "see" it, until the file is released. So `stat`-ing the file (checking it's size) likely won't work either. I think we'd need to repeatedly try to execute it, until it succeeds, or we give up. This theory needs validation though.

timjstone, 2017-05-03:
I only use Windows defender, no other anti virus installed, if that is any help 

Tim

lurch, 2017-05-03:
\*shrug\* I suspect there's so many different virus-scanner programs, who knows how they all work! ;-)

alexandrosm, 2017-05-11:
@lurch "windows defender" is the default microsoft one.

lurch, 2017-05-12:
@alexandrosm My comment was directed at @jhermsmeier , rather than at @timjstone ;)

closed 2017-07-12 by jviotti

lurch, 2017-07-13:
Please see https://github.com/resin-io/etcher/pull/1574#issuecomment-315054168

babi-bibaak, 2018-01-01:
Hello,
i have this error and can't fix that!!!
[image]

jhermsmeier, 2018-01-01:
@babi-bibaak that error doesn't seem to have anything to do with Etcher

lurch, 2018-01-02:
@babi-bibaak Please ensure that you're using the latest version of Etcher v1.2.1

frankgorhamengard, 2018-09-14:
I posted a solution that worked for me in #1793
SD micro card was plugged in to a USB adapter when it was formatted by SD formatter. But after I reformatted in the SD adapter, I was able to install the image successfully with Etcher.
Something about formatting in a USB adapter?

kriekprinsloo, 2018-09-14:
Which USB adapter are you using?  I own a USB cd/DVD adapter, and it's just
not very high quality and doesn't work most times I try to use it.

PR #1574 (2017-07-12): upgrade: drivelist to v5.0.25
This version contains various improvements on the drive detection system
mainly on Windows. The improvements can be summarized as follows:

- Drivelist no longer spawns a .bat script to perform its job. This
  means that a lot of spawning related issues are now fixed

- Drivelist doesn't fetch drive sizes from WMI anymore, which was known
  to incorrectly report a very small size for certain drives, causing
  the famous "0.0 GB" Windows issues (see
  resin-io-modules/drivelist#142)

- Cleanup temporary scripts after execution

This PR also sets the DRIVELIST_DEBUG [... cut]