balena-io/etcher #1064: Error: Path must be a string. Received undefined

daturach, 2017-01-30:
Hello

I am running the version you sent me on Friday last week while debugging an issue with MySQL.
Etcher-1.0.0-beta.18+4f621f6-win32-x64-MYSQL-FIX
Windows 7, French

Today, I am not able to burn an img file. I am getting this error message when clicking on "Flash":
[image]

Can you help?
Thanks

jviotti, 2017-01-30:
Hey there @daturach,

Sorry to hear you're still having issue. The path issue you get is weird. Can you open DevTools (Ctrl+Alt+I), go to the "Console" pane, type `require('electron').remote.process.argv`, press Enter and report back the results?

daturach, 2017-01-30:
Result is:
`["D:\test\Etcher\Etcher.exe"]`

jviotti, 2017-01-30:
OK, cool. Can you try editing the file called `resources/app/lib/child-writer/renderer-utils.js` on line 46 from:

```javascript
return path.join(CONSTANTS.PROJECT_ROOT, electron.remote.process.argv[1]);
```

To:

```javascript
return path.join(CONSTANTS.PROJECT_ROOT, electron.remote.process.argv[0]);
```

? 

Let me know if that's enough to make it work, otherwise I'll upload a new one for you!

daturach, 2017-01-30:
It's enough but I have a new message:

[image]

daturach, 2017-01-30:
Follow by this one:
[image]

jviotti, 2017-01-31:
Hi @daturach ,

Can you try the following new build: https://drive.google.com/open?id=0B7tkbonGU-RyTGVHa1B3QjQ0MW8?

daturach, 2017-01-31:
Hello jviotti

[image]

It's looking pretty good. It's burning at about 10MB/s and just completed the process.
One thing I need to check right now: I have a multi-drive supporting different types of cards but I don't see the drives anymore! I will reboot and report back in a few minutes... 

Thanks again for your great help.

daturach, 2017-01-31:
Tomorrow, I should receive some hardware to test the image.
However, ejecting the card make all drives unavailable. Only a reboot bring them to life again.

lurch, 2017-01-31:
> However, ejecting the card make all drives unavailable. Only a reboot bring them to life again.

There's a discussion about that in #750

jviotti, 2017-02-01:
Awesome, sorry for the custom build mess @daturach. I don't know what happened, all the custom builds usually just work :) 

Regarding the multi-sdcard readers issue, this is something we've heard before as @lurch mentioned. Let's continue the discussion there. @daturach Do you mind linking us to the exact adaptor you're using?

closed 2017-02-01 by jviotti

lurch, 2017-02-01:
Just out of curiosity @jviotti what was the cause of _this_ particular issue (that you fixed with an updated build) ?

jviotti, 2017-02-01:
I released a too custom build for him, almost done by hand, and made various mistakes in the way. The particular issue that this user hit was because I didn't package the code inside the asar, and thus the arguments positions that we expect were different, causing weird stuff in both the child writer execution and elevation stuff.

lurch, 2017-02-01:
Ahhhh. So a bug in @jviotti rather than a bug in Etcher then ;-D

jviotti, 2017-02-01:
Yeah, you're welcome to send a PR! :P