balena-io/etcher #1075: pendrive not detected when started as a normal user, works when started as sudo.

Dark-Passenger, 2017-02-02:
- **Etcher version:**
beta 18
- **Operating system and architecture:**
Centos 7.3 64 bit
- **Do you see any meaningful error information on DevTools?**

doesnt seem to an attached pendrive when starting as a normal user. using sudo it works.

jviotti, 2017-02-02:
Hi @Dark-Passenger ,

Thanks for reporting, that is indeed weird. Can you try running the following bash script with and without `sudo`: https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh?

I'd also love to see the output of `lsblk -d --output NAME` again with and without `sudo`.

Dark-Passenger, 2017-02-13:
Seems a install specific issue. Reinstalled the same distro and same setup, all good.

closed 2017-02-13 by Dark-Passenger