balena-io/etcher #1082: Duplicated error messages

stefan-mihaila, 2017-02-04:
I noticed this while working on https://github.com/resin-io/etcher/issues/1027. It is caused by a `$q` [issue](https://github.com/angular/angular.js/issues/7992) that has been [solved](https://github.com/angular/angular.js/pull/13662) in `angular 1.6`. The issue is that angular's `$exceptionHandler` is called even for `$q` promises that have a rejection handler.

For example, [here we have a handler for the promise](https://github.com/resin-io/etcher/blob/master/lib/gui/pages/main/controllers/image-selection.js#L96), but [the angular exception handler](https://github.com/resin-io/etcher/blob/master/lib/gui/app.js#L195) is triggered also.

[image]

stefan-mihaila, 2017-02-04:
I tried to upgrade `angular` to `1.6.1`, `angular-moment` to `1.0.1` and `angular-ui-router` to `0.4.2`, but I get a very cryptic error message. I think it `angular 1.6` doesn't play well with some other dependency.

```javascript
Uncaught Error: TrackJS Caught: [$injector:modulerr] Failed to instantiate module Etcher due to:
Error: [$injector:modulerr] Failed to instantiate module Etcher.Pages.Main due to:
Error: [$injector:modulerr] Failed to instantiate module {} due to:
Error: [ng:areq] Argument 'module' is not a function, got Object
http://errors.angularjs.org/1.6.1/ng/areq?p0=module&p1=not%20a%20function%2C%20got%20Object
    at /Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:68:12
    at assertArg (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:1972:11)
    at assertArgFn (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:1982:3)
    at /Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4745:11
    at forEach (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:357:20)
    at loadModules (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4720:5)
    at /Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4737:40
    at forEach (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:357:20)
    at loadModules (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4720:5)
    at /Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4737:40
    at forEach (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:357:20)
    at loadModules (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4720:5)
    at createInjector (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4642:19)
    at doBootstrap (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:1838:20)
    at bootstrap (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:1859:12)
    at angularInit (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:1744:5)
    at /Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:32977:5
    at HTMLDocument.trigger (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:3314:5)
    at HTMLDocument.E (file:///Users/stefanmihaila/resin.io/etcher/node_modules/trackjs/tracker.js:5:415)
http://errors.angularjs.org/1.6.1/$injector/modulerr?p0=%7B%7D&p1=Error%3A%20%5Bng%3Aareq%5D%20Argument%20'module'%20is%20not%20a%20function%2C%20got%20Object%0Ahttp%3A%2F%2Ferrors.angularjs.org%2F1.6.1%2Fng%2Fareq%3Fp0%3Dmodule%26p1%3Dnot%2520a%2520function%252C%2520got%2520Object%0A%20%20%20%20at%20%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A68%3A12%0A%20%20%20%20at%20assertArg%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A1972%3A11)%0A%20%20%20%20at%20assertArgFn%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A1982%3A3)%0A%20%20%20%20at%20%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A4745%3A11%0A%20%20%20%20at%20forEach%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A357%3A20)%0A%20%20%20%20at%20loadModules%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A4720%3A5)%0A%20%20%20%20at%20%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A4737%3A40%0A%20%20%20%20at%20forEach%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A357%3A20)%0A%20%20%20%20at%20loadModules%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A4720%3A5)%0A%20%20%20%20at%20%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A4737%3A40%0A%20%20%20%20at%20forEach%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A357%3A20)%0A%20%20%20%20at%20loadModules%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A4720%3A5)%0A%20%20%20%20at%20createInjector%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A4642%3A19)%0A%20%20%20%20at%20doBootstrap%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A1838%3A20)%0A%20%20%20%20at%20bootstrap%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A1859%3A12)%0A%20%20%20%20at%20angularInit%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A1744%3A5)%0A%20%20%20%20at%20%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A32977%3A5%0A%20%20%20%20at%20HTMLDocument.trigger%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A3314%3A5)%0A%20%20%20%20at%20HTMLDocument.E%20(file%3A%2F%2F%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Ftrackjs%2Ftracker.js%3A5%3A415)
    at /Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:68:12
    at /Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4759:15
    at forEach (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:357:20)
    at loadModules (/Users/stefanmihaila/resin.io/etcher/node_modules/angular/angular.js:4720:5)
[... cut ...] it%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A1744%3A5)%0A%20%20%20%20at%20%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A32977%3A5%0A%20%20%20%20at%20HTMLDocument.trigger%20(%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A3314%3A5)%0A%20%20%20%20at%20HTMLDocument.E%20(file%3A%2F%2F%2FUsers%2Fstefanmihaila%2Fresin.io%2Fetcher%2Fnode_modules%2Ftrackjs%2Ftracker.js%3A5%3A415)
```

jviotti, 2017-02-06:
Oh, amazing find @stefan-mihaila . This explain why we get some double errors on TrackJS /cc @alexandrosm.

Regarding the cryptic error, I assume you finally fixed it on the linked PR, right? If so, maybe worth explaining how you resolved it here, just for reference?

stefan-mihaila, 2017-02-06:
@jviotti By try and error I realised that `angular-moment` was causing the error, and this wasn't actually `angular 1.6` related. `angular-moment v1.0.0` works fine, `angular-moment v1.0.1` does not, I think it's related to what `angular-moment v1.0.1` exports. In the PR I just used `angular-moment v1.0.0`. I didn't really investigated further, but now that you mention it, I'll take a look out of curiosity.

jviotti, 2017-02-06:
This seems to be the cause: https://github.com/urish/angular-moment/commit/dccdbbcea6421def5c000c6c88f53affc103d5ad

The module name is not returned anymore if they detect they are running in Electron (for some strange reason).

You could upgrade to v1.0.1 and do `require('angular-moment')` at the top of the file and add the module string to the dependency array yourself.

stefan-mihaila, 2017-02-07:
> You could upgrade to v1.0.1 and do require('angular-moment') at the top of the file and add the module string to the dependency array yourself.

Updated PR to include `angular-moment v1.0.1`: https://github.com/resin-io/etcher/pull/1083/files#diff-c6e5793d29e5153fef626158c52514d3R32

closed 2017-02-07 by jviotti