balena-io/etcher #1090: Question about imaging with .dmgs

pizzapants, 2017-02-10:
I have a question that might be stupid. I just found this tool and it is amazing for flashing from certain images. Im currently backing up my raspian installations by creating a disk image of it, but Etcher doesn't allow me to select a .dmg as my image. Should I be doing something differently? Why would the .dmg not be recognized? 

Secondly, is there a way to essentially do the reverse of that process and create a backup .dmg of an existing SD card?

jhermsmeier, 2017-02-10:
Hey @pizzapants, not a stupid question at all. The issue with `.dmg`s is that Apple's disk image format and some file systems used in it are proprietary, and (as far as I know) not publicly documented by Apple, even though it has been reverse engineered. To fully support it cross-platform, we'd have to implement it, which is fairly complex and I'm not sure we'd work on that soon, but @jviotti would know more.

It's probably best to stick with flat image files that can be read & written by everything for now.

> Secondly, is there a way to essentially do the reverse of that process and create a backup .dmg of an existing SD card?

Etcher currently can't create images from devices, but support for that is planned (see #266)
For now, you can only create `.dmg`s with Mac OS' Disk Utility, for example.

lurch, 2017-02-10:
Yup, see also https://github.com/resin-io/etcher/issues/836 for .dmg support.

Closing this issue, as both questions are already covered by other issues.

closed 2017-02-10 by lurch

pizzapants, 2017-02-10:
@jhermsmeier Thanks for the reply. What is the best way currently to create a flat image of an existing install then, on osx?

jhermsmeier, 2017-02-10:
There are several options, the most popular probably being [`dd`](https://en.wikipedia.org/wiki/Dd_(Unix_software)):
```
dd status=progress if=/dev/rdisk3 of=backup.img
```
Where `/dev/rdisk3` would be the device path to the SD card you want to image (the device name (diskX) can be found in the "Device:" field in Disk Utility)

pizzapants, 2017-02-11:
oh, ive been using that command but just doing of=backup.dmg. Is it as simple as changing the file extension?

lurch, 2017-02-11:
I think if you've just been using `dd`, then what you have is _actually_ an `.img` file, not a `.dmg` file :-)