balena-io/etcher #1093: Support decompressing while streaming

jviotti, 2017-02-10:
This affects the [`image-stream`](https://github.com/resin-io/etcher/tree/master/lib/image-stream) module. Currently, all the decompression handlers in that module rely on the file being present on the disk.

Some decompression methods don't lend themselves to streaming out of the box, therefore we should try to "mimic" this behaviour by probably streaming to the disk (at least partly) until we can decompress.

jviotti, 2017-02-10:
Maybe @jhermsmeier can help here, since this is very related to the writer, and requires hardcore streaming skills :) ?

lurch, 2017-02-12:
> therefore we should try to "mimic" this behaviour by probably streaming to the disk (at least partly) until we can decompress.

For compression methods not designed for streaming, I guess in the pathological case this might end up needing to buffer the _entire_ file on disk before decompressing? (which obviously kinda negates the benefit of streaming in the first place)

jviotti, 2017-02-12:
> I guess in the pathological case this might end up needing to buffer the entire file on disk before decompressing? (which obviously kinda negates the benefit of streaming in the first place)

Its definitely a possibility for some formats, however its something we're willing to accept. The speed advantage of streaming while flashing is lost, but the benefit of downloading the image automatically from the internet is still preserved.

BTW, notice that we will only support streaming from our own catalog (we will not offer a custom URL input, at least for now).

alexandrosm, 2017-02-12:
streaming-while-writing will only be for catalog entries. There, we can
dictate that only suitable compression methods are used. The whole benefit
of the catalog is to have greater predictability in the inputs so we don't
have to jump through so many hoops.

lurch, 2017-02-13:
> There, we can dictate that only suitable compression methods are used.

Yeah, that's what I was thinking / hoping :+1:

jhermsmeier, 2017-02-13:
> Some decompression methods don't lend themselves to streaming out of the box

Are there any others than `.zip` we'd need to consider?

jviotti, 2017-02-13:
Apart from zip:

- gzip is easy to stream, however we need to buffer a bit of it in order to be able to inspect the header, estimate the uncompressed size, and return an object with the uncompressed size + the image stream

- xz has the same problem as gzip (we need to read its uncompressed size on the fly before returning the stream to the user)

jviotti, 2017-02-15:
Some design ideas for this:

Currently, the image stream handlers & archive hooks are very tightly coupled with the concept of "files" (they both accept file names as arguments). Of course this makes little sense in the context of streaming, so I propose the following changes to achieve that:

- At [`lib/image-stream/index.js`](https://github.com/resin-io/etcher/blob/master/lib/image-stream/index.js), we expose a `.getFromFilePath()` function that accepts a file path and calls the handlers with the file path and some metadata for it (like the file size). We can make the handlers accept a readable stream and the metadata, and have `.getFromFilePath()` infer it from the file size, while having a `getFromURL()` function or something like that, that gets the metadata from HTTP headers instead

- Since the entry point functions must know what handler to call, the `.getFromFilePath()` function calculates the MIME type from the file itself. For URL streaming, we should be able to get this information again from HTTP headers

- Since the handlers now get a stream, then it means that the archive hooks will have to accept a stream as well, probably buffering internally if necessary

- We expose a `.getImageMetadata()` function that is tied to a file. We should either create two metadata functions (one for files and one for URLs), or maybe make the same function react differently depending on if it detects that the string is a valid http/https URL

- The `.getImageMetadata()` simply re-uses the `.getFromFilePath()` but ignores the image readable stream. It would be nice if we can decouple the metadata fetching routine from the image stream fetching routine so we don't have to create the stream in vain

lurch, 2017-02-15:
If we're only going to be streaming images that we've added to our catalog, won't the online catalog have all the metadata we need, without having to try and infer any of it from HTTP headers?

jviotti, 2017-02-16:
We don't store the compressed image size anywhere on the metadata. That information is only available from the headers.

lurch, 2017-02-16:
...is there any reason why we couldn't store the compressed image size in the metadata? Surely it makes sense to pre-process as much as possible?

jviotti, 2017-02-16:
The size is automatically calculated by S3 for the header, which is a way more reliable source than property in our metadata (users can get it wrong for example, or forget to update it).

In terms of efficiency, reading the HTTP response headers that are already being returned to us is probably way faster.

lurch, 2017-02-16:
Fair enough, always good to know that other options have at least been considered :-)

(and I was assuming that the metadata in the catalog would be updated/verified by a script, so it would be impossible for it to be incorrect)

jviotti, 2017-02-16:
Yeah, I guess we could have a script or something to check it, but for this particular case, S3 sounds like the way to go.

closed 2018-10-15 by jviotti