balena-io/etcher #1095: Implement a new flashing progress widget

jviotti, 2017-02-10:
Depends on https://github.com/resin-io/etcher/issues/1094

[image]

The color of each dot represents the portion of the drive that has been written, validated, the parts that failed, etc.

Since the amount of dots seem to be static (e.g: it doesn't depend on the image/drive size), I believe this problem can be elegantly solved by a well crafted SVG representing the circle. We can style each dot separately using simply CSS.

jhermsmeier, 2017-10-24:
Connects to https://github.com/resin-io/etcher/issues/778

jviotti, 2022-08-15:
Closing stale issues

closed 2022-08-15 by jviotti