balena-io/etcher #1097: Ability to pass filename to Etcher when launching

oshunluvr, 2017-02-11:
Etcher version: 1.0.0-beta.18-linux-x64
Operating system and architecture: i7-6700K on KDEneon (Kubuntu) 16.04, Plasma 5 DE.
Not an error, but a requested change:

First: I apologize. I am not a coder so I have no idea how to formulate a PR. Rather than muck it up and waste everyone's time, I hope someone who is will read this and agree with my idea and put it forward.

It would immensely enhance the usability of Etcher (at least in Linux) to have it accept a filename when launching the program, both from the GUI and from the CLI. It even seems a natural progression because the first thing Etcher asks for is the name of the file you wish to burn.

In KDE/Plasma, having the ability to right-click from Dolphin, the file manager, and launch Etcher with the chosen file already set to go, or to execute etcher in a terminal along with the filename would be BRILLIANT! 

As a side-note: I'm very active with supporting beginners who are dipping their toes in Linux waters. One of the most common hurdles they face is finding a usable tool to create their first bootable USB or CDROM. Having a single, cross-platform, well-functioning, simple ISO burn tool that everyone can use the first time from any PC would be a huge leap forward.

jviotti, 2017-02-12:
Hi @oshunluvr ,

Thanks a lot for reporting! There was a patch at some point to introduce this feature (https://github.com/resin-io/etcher/pull/333), however it ended up getting abandoned by its submitter and it had some issues that needed to be fixed before.

I'm all for this improvement if we find a nice way to introduce it without disrupting the current behaviour. The main problem is that same application entry point executes the CLI or the GUI mode, depending on certain factors, like the presence of certain environment variables.

lurch, 2017-02-20:
@oshunluvr And just to clarify, Etcher only writes to mass-storage-devices (USB flash drives, HDDs, SD Cards, etc.), I don't think it'll ever support burning CDROMs.

jviotti, 2017-02-20:
I've been looking to try that out for a while now :P If the OS exposes CDROMs as block devices we can write to, then it *should* work (I have no idea about CDROMs really).

lurch, 2017-02-20:
Nah, if it was that easy then there wouldn't be specific CD-burning programs.
https://en.wikipedia.org/wiki/CD-R#Writing_methods

jviotti, 2017-02-20:
Too bad! :(

jhermsmeier, 2017-02-20:
@lurch @jviotti the physical writing methods should be abstracted away by the the kernel, running `dd if=some.iso of=/dev/cdrom1` might actually work (I can't try as I don't have spinning plastic media drives), but if it doesn't, this might be an approach: http://www.ccp14.ac.uk/ccp14admin/linux-server/iso_images_to_dvdram/

lurch, 2017-02-20:
I suspect _very_ few people will be using DVDRAM discs ;-)

(although IIRC being able to use them as block devices was the big appeal of DVDRAM discs when they were introduced - but they were soon eclipsed (and made obsolete) by USB flash drives. Notes from history... ;-)  )

jhermsmeier, 2017-02-20:
Hmm, yes. I wonder if the particular type of disk would actually matter, as long as it's written to in a sequential manner.

oshunluvr, 2017-02-20:
It appears that my use of word "burn" or my follow-up regarding beginners and their trouble with ISOs might have mislead some into thinking I was expecting the ability to write to a CDROM/DVDROM. I apologize for this unfortunate choice of words. 

My only desire was the functionality to pass a filename to Etcher. Thanks for the responses though. Nice to see this project is so active. I will immediately start request more features! ;)

MattDMo, 2017-04-01:
Just my 2¢ (I actually just came across this project via a friend's recommendation, I haven't actually *used* it yet...), but I think having the ability to write to *any* storage device, whether it be any variation of CD-R(W) or DVD-R(W), along with USB sticks and all of the other currently supported devices, would be a great feature. Not all machines in use are recent configurations that don't even include spinny plastic drives anymore, and while USB is obviously the technology of choice for at least the immediate future, there are likely a ton of boxes out there with BIOSes that may not even support booting from USB, so the use of 120mm media isn't going away anytime soon.

That being said, I don't have a lot of (read: any) programatic experience with writing to and "closing" [C|DV]D-Rs, not to mention -RWs, so I have no idea how complex or time-consuming this may end up being.

Like I said before, just my 2¢.

lurch, 2018-02-26:
Closing this as a duplicate of #1386

closed 2018-02-26 by lurch