balena-io/etcher #1098: Error: spawn ENOMEM

keunes, 2017-02-12:
- **Etcher version:** 1.0.0-beta.18
- **Operating system and architecture:** Ubuntu 14.04

Running Etcher to get Ubuntu 16.04 on a 4GB USB key I get the following error:

Error: spawn ENOMEM
    at exports._errnoException (util.js:1026:11)
    at ChildProcess.spawn (internal/child_process.js:313:11)
    at exports.spawn (child_process.js:392:9)
    at Object.exports.execFile (child_process.js:155:15)
    at Object.module.(anonymous function) [as execFile] (ELECTRON_ASAR.js:224:18)
    at Object.exports.run (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/drivelist/lib/scripts.js:84:16)
    at Object.exports.list (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/drivelist/lib/drivelist.js:56:11)
    at Object.tryCatcher (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)
    at Object.ret [as listAsync] (eval at makeNodePromisifiedEval (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/promisify.js:184:12), <anonymous>:13:39)
    at isElevated.then.then (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/lib/cli/etcher.js:70:20)
    at tryCatcher (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)
    at Promise._settlePromise (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Async._drainQueue (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)
    at Async._drainQueues (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)
    at Immediate.Async.drainQueues (/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)
    at runCallback (timers.js:574:20)
    at tryOnImmediate (timers.js:554:5)
    at processImmediate [as _immediateCallback] (timers.js:533:5)

alexandrosm, 2017-02-12:
Thanks for filing this! The maintainers will respond as soon as possible.

jviotti, 2017-02-12:
Hi @keunes,

Thanks for reporting. The error indicates you didn't have enough memory for a certain child process (the drive detection script in this case) to be forked.

How much RAM do you have in that machine, and monitor it while running Etcher? Can you check how much RAM Etcher is consuming? 

We rely on the  http://electron.atom.io framework to build the UI, which is known to be quite memory intensive, so there's not much we could do here anyway.

My suggestions:

- Try increasing the RAM of your machine
- Try stripping down your Ubuntu setup so it doesn't consume that much memory (and thus leaves some space for Etcher), like switching to a lightweight desktop manager, etc
- Try running the Etcher CLI instead

Regarding the Etcher CLI, we're not releasing it officially yet, but here's a guide to run it from source, given you have NodeJS installed: https://github.com/resin-io/etcher/blob/master/docs/CLI.md

lurch, 2017-02-12:
...and of course the easiest suggestion (if you haven't tried it already) is to simply close any other programs that you also have running at the same time, so that Etcher isn't competing for memory with other applications running on your computer ;-)

alexandrosm, 2017-02-12:
@jviotti IMO we should explore how much ram we use on top of electron
before closing. We haven't really measured so it's a bit premature to blame
electron.

jviotti, 2017-02-12:
In my MacBook Pro, Etcher v1.0.0-beta.18 consumes about 138 MB after opening it (without doing anything). The default electron application, which does nothing, consumes about 110 MB here. We're looking at a ~25 MB overhead (at least in my system), which looks too small to try to optimize IMHO. I believe things will get way better with React though.

jviotti, 2017-02-12:
I'm interested in hearing @keunes current specs. Are you already making use of swap? If you do have swap available, the system should have used that, right?

In the worst case, we can compile a list of minimal pre-requisites to run Etcher.

alexandrosm, 2017-02-13:
@jviotti -- the question is what happens while Etcher is working. How high
does the ram usage go?

jviotti, 2017-02-13:
It goes to about ~300 MB for an uncompressed image on my system (mostly due to the internal CLI fork), although that doesn't seem to be the issue here. The user is getting ENOMEM when scanning the drives, which means the writing process is not running (drive scanning stops during write/validation).

It does mean that even if the user optimized his computer to be able to run Etcher without flashing, it will probably still fail while flashing :( 

/cc @jhermsmeier Maybe you can keep an eye on memory usage while re-architecting the writer, and see if there are any low hanging fruits there?

jhermsmeier, 2017-02-13:
@jviotti yeah, will do – we might also want to look at passing some v8 options when forking the process, to optimise the memory consumption a bit.

Regarding Ubuntu, I had it on several occasions that Ubuntu would kill processes, erroring with ENOMEM even though there was still plenty of RAM unused, which was caused by not having allocated any swap space.

keunes, 2017-03-01:
Thanks for your reply, @jviotti. For some reason this whole thread didn't pop-up in my email (or at least, not the way I would notice)

> Try increasing the RAM of your machine
> Try stripping down your Ubuntu setup so it doesn't consume that much memory (and thus leaves some space for Etcher), like switching to a lightweight desktop manager, etc

I wouldn't change my whole desktop environment to 'just' be able to write an image with Etcher as there's other ways available. But now that I understand what's going wrong, I completely understand where it comes from: Etcher was running on quite an old computer (so not worth the investments of a RAM upgrade).

> How much RAM do you have in that machine, and monitor it while running Etcher? Can you check how much RAM Etcher is consuming?

If I'm back home I can check on that, if you're still interested.

---
What I would like to suggest: human-readable error messages. If I would have gotten the message 'not enough RAM/memory to run Etcher' I'd immediately understand and not try again and file this bug report ;)

jviotti, 2017-03-01:
If I'm back home I can check on that, if you're still interested.

Definitely. We have some human friendly translations here: https://github.com/resin-io/etcher/blob/master/lib/cli/errors.js#L27, but we're not addressing the ENOMEM one.

jviotti, 2017-03-01:
@keunes Just to clarify: this error happens as soon as you open the app, right? Or does it happen when you click flash?

keunes, 2017-03-01:
> Just to clarify: this error happens as soon as you open the app, right? Or does it happen when you click flash?

Sorry, I should've been more clear. It does when I'm trying to flash, not immediately upon start.

jviotti, 2017-03-01:
Right. The stack trace says `/tmp/.mount_ZZ2t46/usr/bin/resources/app.asar/lib/cli/etcher.js`, which means it's coming from the latest drive check from the writer process before starting the flash. Thanks a lot @keunes , this should be more than enough for us to clarify the error.

closed 2017-03-10 by jviotti