balena-io/etcher #1108: Error: ENOENT

OhhNooNoob, 2017-02-16:
Sorry, if this is the wrong way of going about reporting an issue, but i have been using Etcher a lot for about 3 - 4 months and never had a problem. But for the past couple of days, this error message has been 
popping up, and i have to admit its beyond my skill level to try and figure out what the cause is.

Again, i apologies if i am going about this in the wrong way, but i use Etcher on a daily basis with my ARM boards and Intel Compute Sticks,  so if this is an actual bug (and not just me doing something wrong), i thought i would send this.

Thanks for an awesome app  and for the effort/hard work you do.

- v1.0.0-beta.18
- Windows 10, 64bit

   
[Error log -1487247693785.txt](https://github.com/resin-io/etcher/files/780034/Error.log.-1487247693785.txt)

[image]

jhermsmeier, 2017-02-16:
Hey @OhhNooNoob, thanks for reporting this! And no, this is not the wrong way to go about reporting issues – quite the opposite :)

Has this only been occurring since you updated to v1.0.0-beta.18, or did it work with this version before?

OhhNooNoob, 2017-02-16:
It has happened before, but at the time i thought it was either the external multicard reader i used (my desktop does not have integrated sd card reader,  unfortunately) or the the .img files had been corrupted during either download or extraction.

I recently got an Acer laptop with built-in card reader, and have not experienced any issues when running etcher on that. And, when i boot into Apricity OS (Linux distro based on Arch) on my desktop there is no problems.

OhhNooNoob, 2017-02-18:
Sorry for the delayed response. 
Here you go: https://www.dropbox.com/s/senwbowqrrbr91z/app.asar?dl=0
If there is anything else that could help, just say the word. I am glad to help, where i can.

jviotti, 2017-02-19:
Hi @OhhNooNoob ,

Thanks for the response! So the `asar` archive does include `win32.bat`, and it seems perfectly valid, so I wonder what might be causing this.

At what stage does this problem appear? Does it happen right after you open Etcher, or right when you click "Flash"?

Based on the stacktrace, it seems that the error is coming from the `drivelist` call coming from the CLI.

lurch, 2017-02-20:
@OhhNooNoob It's not clear from your earlier messages - does this happen every single time (i.e. is it 100% reproducible), or only "sometimes" ?

OhhNooNoob, 2017-02-21:
it happens about 2-3 seconds after i click flash. And, yes. every time i press flash, it gives this error. 
I have installed Etcher as a portable app on a usb stick, and now it works every time. Which is actually more convenient, since i always have it with me now.

jviotti, 2017-02-21:
Interesting. @OhhNooNoob were you using the installer version before? What is the partition format of your C drive? NTFS? Did you make any tweak to your system that you think could be causing this?

I wonder if this is somehow related to https://github.com/resin-io-modules/drivelist/pull/143, even though I got EPERM on that one.

lurch, 2017-02-21:
@jviotti Is it possible that the child-writer-process has different access permissions to the main Etcher GUI, that means it isn't able to read the `app.asar` file?

jviotti, 2017-02-21:
The child writer process is still an Electron process, just with the Chromium stuff disabled, but I'm completely sure it has `.asar` support. There must be something else going on there, but hard to debug without being able to reproduce locally and building a debug Electron release :/

lurch, 2017-02-21:
I meant file-access permissions... (obviously the child writer must have `.asar` support, or everybody would be hitting this issue, not just @OhhNooNoob )
I was wondering if @OhhNooNoob might have some weird file / directory permissions set-up, and that's why he's the only person seeing this. Obviously with the portable version on a USB stick, that's world-readable by everyone, so no file-permissions problems there.

This is just a hunch, I might be barking up completely the wrong tree! ;-)

jviotti, 2017-02-21:
Oh I see, sorry for the confusion. Yeah, that makes sense. Electron temporarily extracts the scripts to a certain place (let me try to find that), so maybe @OhhNooNoob can help us check the permissions of the problematic file when its extracted.

jviotti, 2017-02-21:
OK, Electron seems to store temporary files in `C:\Users\<user>\AppData\Local\Temp\`.

@OhhNooNoob Can you re-install the installer version of Etcher, get the error, **and without closing Etcher** (e.g: leaving the error window open), go to `C:\Users\<user>\AppData\Local\Temp\`, find a file with an extension `.tmp.bat` (for example: `B39C.tmp.bat`), and report back the the permissions of the file? If there are multiple files that match the above characteristics, do you mind reporting the permissions of all of them?

See this guide if you are having troubles inspecting the permissions of Windows files: https://technet.microsoft.com/en-us/library/cc754344(v=ws.11).aspx

In summary:

> - Right-click the file or folder for which you want to set permissions, click Properties, and then click the Security tab.
> - Click Edit to open the Permissions for <Object> dialog box.

I guess some screenshots of that window should do it.

Thanks a lot in advance :)

lurch, 2017-02-21:
> Can you re-install the installer version of Etcher

...or simply **run** the already-installed version of Etcher, if you didn't actually uninstall it yet ;-)

jviotti, 2017-03-03:
I've been investigating the possible causes of this issue and narrowed it down to the following Electron function: https://github.com/electron/electron/blob/v1.4.4/atom/common/asar/archive.cc#L271

```c++
bool Archive::CopyFileOut(const base::FilePath& path, base::FilePath* out) {
  if (external_files_.contains(path)) {
    *out = external_files_.get(path)->path();
    return true;
  }

  FileInfo info;
  if (!GetFileInfo(path, &info))
    return false;

  if (info.unpacked) {
    *out = path_.AddExtension(FILE_PATH_LITERAL("unpacked")).Append(path);
    return true;
  }

  std::unique_ptr<ScopedTemporaryFile> temp_file(new ScopedTemporaryFile);
  base::FilePath::StringType ext = path.Extension();
  if (!temp_file->InitFromFile(&file_, ext, info.offset, info.size))
    return false;

#if defined(OS_POSIX)
  if (info.executable) {
    // chmod a+x temp_file;
    base::SetPosixFilePermissions(temp_file->path(), 0755);
  }
#endif

  *out = temp_file->path();
  external_files_.set(path, std::move(temp_file));
  return true;
}
```

Basically, if that function returns `false`, then this particular error happens. I was considering trying an Electron upgrade to see if that would fix the issue, but there were no changes to Electron in the code we care about since the version we're using (v1.4.4).

Now, the above function tells us that there are two reasons `false` could be returned:

- The `GetFileInfo` function failed (which is defined on this same file)
- We couldn't create a temporary file from the file we're trying to execute

Note that `drivelist` fails for @OhhNooNoob on the writer CLI process, which tells us two things:

- `drivelist` worked fine until he clicked flash (so there is no issue in the asar, etc)
- `drivelist` was executed from the CLI code

The biggest difference between the CLI and the GUI code is that the CLI is running with administration permissions. 

Could it be that we're dealing with a permissions issue that prevented the temporary file from being created?

@OhhNooNoob Can you try starting an elevated command prompt, and try to create some random file at `C:\Users\<user>\AppData\Local\Temp\`, and see if that yields any error? Checking the permissions of that directory would also be cool.

Another way forward to debug this issue is to create a custom build for @OhhNooNoob that contains a custom Electron with augmented logging. I might prepare something next week.

OhhNooNoob, 2017-03-05:
So sorry for the late response. I have not been able to get back to you guys, due to my mother being admitted to the hospital with heart problems. This is just a quick response, to let you all know that i am still here, and will read all the above questions, and see what i can do.
I will get back to you tomorrow (Sunday, marts 5th), with answers.

Again i apologies for the "radio silence".

Best regards, n00b

jviotti, 2017-03-06:
I'm so sorry to hear that @OhhNooNoob , and hope everything goes well! No hurries, I'll be preparing the custom Etcher build with a custom patched Electron in the mid-time :)

jviotti, 2017-03-08:
Hi @OhhNooNoob ,

Check out https://drive.google.com/open?id=0B7tkbonGU-RyOUFHVlEzQmJYM3c. The build is very custom, and thus is unsigned, and the main executable is unbranded and called `electron.exe`.

That build will output lots of stuff to standard output, which means that you should execute it from PowerShell or `cmd.exe`, by `cd`ing into it and executing `electron.exe`.

Let us know how that goes!

closed 2017-07-12 by jviotti

lurch, 2017-07-13:
Please see https://github.com/resin-io/etcher/pull/1574#issuecomment-315054168

PR #1574 (2017-07-12): upgrade: drivelist to v5.0.25
This version contains various improvements on the drive detection system
mainly on Windows. The improvements can be summarized as follows:

- Drivelist no longer spawns a .bat script to perform its job. This
  means that a lot of spawning related issues are now fixed

- Drivelist doesn't fetch drive sizes from WMI anymore, which was known
  to incorrectly report a very small size for certain drives, causing
  the famous "0.0 GB" Windows issues (see
  resin-io-modules/drivelist#142)

- Cleanup temporary scripts after execution

This PR also sets the DRIVELIST_DEBUG [... cut]