balena-io/etcher #1109: Prevent write failure due to block padding

jhermsmeier, 2017-02-16:
Since Etcher pads the last block to either 1 MB or 512 KB, and some devices and images are not divisible by such large block sizes (i.e. the Kingston mSata 60GB SMS200S3/60G with 60022480896 Bytes), padding the last block to such a large size will make the write fail on the last block.

This can be mitigated by detecting the physical block size (see [node-blockdevice#L131-L200](https://github.com/jhermsmeier/node-blockdevice/blob/master/lib/blockdevice.js#L131-L200) for example) of a device prior to flashing, and then padding the last block to that size.

lurch, 2017-02-17:
I think the reason for this (which I've suspected for a while, but never got round to verifying), is that @jviotti once got confused between "512 bytes" and "512 kilobytes".
In a couple of different places he's mentioned that "Windows gives errors when you try to use unusual block sizes" and I _suspect_ that the smallest blocksize that Windows will allow you to use is 512 bytes, but given that _most_ SD cards are an even multiple of 512KB, I suspect that's where the confusion (and the current 512KB block-padding limitation) snuck in.

Ahh, just found up https://github.com/resin-io-modules/etcher-image-write/pull/58 which has more info :-)
So perhaps this issue should be migrated to `etcher-image-write`?

P.S. Good thing I have a great memory ;)

jviotti, 2017-02-17:
Yeah, that might have been the reason. I remember getting EINVAL on Windows (8 I think) for 256 and 128 "something" (not sure if it was bytes or kilobytes anymore :( )

jviotti, 2017-03-03:
@jhermsmeier Can you publish a new version of `etcher-image-write` with this fix, so I can upgrade and close this one?

jhermsmeier, 2017-03-06:
@jviotti yup, published as v9.0.1

closed 2017-03-06 by jviotti