balena-io/etcher #1115: Attempt to detect incomplete images

jviotti, 2017-02-19:
Some images might not be complete (e.g: the download stops halfway through). We could implement the following heuristics to determine if the image is complete or not:

- Inspect the partition table and ensure every position referenced there exists
- If the image contains a GPT, the GPT is also duplicated at the end of the file. Check that it exists at the end as well

 If we detect the image is incomplete, we should present an error/warning to the user.

jviotti, 2017-02-19:
Maybe @jhermsmeier has other ideas as well?

lurch, 2017-02-20:
As mentioned elsewhere, until I see evidence to the contrary, I still suspect all images will be an exact multiple of 512 bytes... ;-)

jviotti, 2017-02-20:
Fair enough, but if the image download stopped in the middle, then the changes are suddenly way higher.

lurch, 2017-02-20:
OTOH, I suspect _most_ people will download images as a compressed file (`.zip`, `.xz`, etc.) in which case I believe the archive won't even open if "the download stopped in the middle" ?

jviotti, 2017-02-20:
Maybe. The module we found to unzip while streaming ignores the zip integrity warnings altogether in order to be able to work. Who knows what we will encounter :)

lurch, 2017-02-20:
I guess streaming images will mean that we won't be able to apply these heuristics until after we've already streamed the whole file.

jviotti, 2022-08-15:
Closing stale issues

closed 2022-08-15 by jviotti